iiab/iiab #3847: Installation failure at Kiwix download

breconides, 2024-11-19:
### Expected behavior and actual behavior:
IIAB won't complete installation and fails when trying to download Kiwix

### Steps to reproduce the problem:
sudo IIAB 

this downloads and installs everything fine up until Kiwix where this happens

------------------
TASK [kiwix : Download https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-x86_64.tar.gz into /opt/iiab/downloads (ACTUAL filename should include kiwix-tools version, or nightly build date)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/opt/iiab/downloads", "elapsed": 0, "gid": 0, "group": "root", "mode": "0755", "msg": "Request failed: <urlopen error [Errno 104] Connection reset by peer>", "owner": "root", "size": 4096, "state": "directory", "uid": 0, "url": "https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-x86_64.tar.gz"}

TASK [kiwix : SEE ERROR ABOVE (skip_role_on_error: False)] *********************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=38   changed=4    unreachable=0    failed=1    skipped=15   rescued=1    ignored=0   
---------------------

### IIAB version, OS version, hardware details:
Version - 8.2
OS - Ubuntu 24.04
Hardware - VirtualBox on Windows 11

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

https://paste.centos.org/view/9481c7fb

holta, 2024-11-19:
Hi @breconides,

This is routine ISP or networking problem as you're trying to download https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-x86_64.tar.gz

Possibly due to an ISP, or any networking setup / proxying (etc) between your VM and kiwix.org's mirrors not being 100% reliable potentially.  FWIW the download works perfectly for me.

Try again running `sudo iiab` later — and/or try with a different ISP and/or from a different location?

OldKnobby, 2024-12-03:
I am having the same issue.  I'm pretty sure it's not caused by an ISP.  Tried with 2 different providers with the same result.  

Do anyone know which .yml to edit to redirect to another mirror?  https://ftp.nluug.nl/kiwix/release/kiwix-tools/ has the files available.

Thanks

Tengu42x, 2024-12-03:
I'm having the same issue, multiple machines, multiple ISP's

OldKnobby, 2024-12-03:
I'm pretty sure there's a broken url somewhere in the tasks.yml's  called by the "1-line installer" script.  [breconides](https://github.com/breconides) output error above is for an x86 install, mines an arm64 but they both fail at the same place, when it has to download an architecture specific package.

holta, 2024-12-03:
> https://paste.centos.org/view/1208bfba

Thanks.  High-level error is:

2024-12-02 17:57:56,062 p=5184 u=root n=ansible INFO| fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/opt/iiab/downloads", "elapsed": 0, "gid": 0, "group": "root", "mode": "0755", "msg": "Request failed: <urlopen error [Errno 111] Connection refused>", "owner": "root", "size": 4096, "state": "directory", "uid": 0, "url": "https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-aarch64.tar.gz"}

Can you debug further using things like...

```
curl https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-aarch64.tar.gz
```

...for example try the `curl` command flag/option to display all http headers?

OldKnobby, 2024-12-03:
Hi [Holta](https://github.com/holta), this is response from the command above:

>  $ curl https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-aarch64.tar.gz
> curl: (7) Failed to connect to download.kiwix.org port 443 after 283 ms: Couldn't connect to server

I also tried "curl -i" command hoping for a better result but still got this response:

>  $ curl -i https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-aarch64.tar.gz
> curl: (28) Failed to connect to download.kiwix.org port 443 after 134107 ms: Couldn't connect to server
> 

I do get a ping response from the kiwix server though:

> $ ping download.kiwix.org
> PING storage.k8s.kiwix.org (135.181.224.247) 56(84) bytes of data.
> 64 bytes from static.247.224.181.135.clients.your-server.de (135.181.224.247): icmp_seq=1 ttl=52 time=112 ms
> 

Let me know what else you'd like me to try.

Thanks

holta, 2024-12-03:
I've confirmed.  The curl commands fails to connect, giving up after about 2 minutes;

```
$ curl https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-aarch64.t
ar.gz
curl: (28) Failed to connect to download.kiwix.org port 443 after 135421 ms: Couldn't connect to server
```

The situation has deteriorated over the past ~2 weeks.  I'm not sure why.

@rgaudin would you know what's happening?

holta, 2024-12-03:
@rgaudin, if you know:

- https://download.kiwix.org appears to be back online at the moment?

- is this an old server that possibly needs replacing or overhaul?

rgaudin, 2024-12-03:
It's back ; our hosting provider killed our server (!!) ; we had to rebuild one from scratch using backups and mirrors.

It's back online ; zimit ZIMs and extra files (archives mostly) are not completely synced yet but will be by tomorrow I think.

holta, 2024-12-03:
Great News, thanks @rgaudin !

OldKnobby, 2024-12-03:
That's great.  I made it past the Kiwix download/install but get another fatal stop after the 'transmission-daemon' needs to restart.

> TASK [transmission : Enable & (Re)Start 'transmission-daemon' systemd service, if transmission_enabled] *****************************************************
> fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service transmission-daemon: Job for transmission-daemon.service failed because the control process exited with error code.\nSee \"systemctl status transmission-daemon.service\" and \"journalctl -xeu transmission-daemon.service\" for details.\n"}
> 
> TASK [transmission : SEE ERROR ABOVE (skip_role_on_error: False)] *******************************************************************************************
> fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}
> 

I'm going to try re-running it.  Any suggestions?

holta, 2024-12-03:
> made it past the Kiwix download/install but get another fatal stop after the 'transmission-daemon' needs to restart.

Sorry about that!

Transmission (torrenting software) has regressed on Ubuntu especially:

So I recommend you opt out of it for now — please set the following line in `/etc/iiab/local_vars.yml`

```
transmission_enabled: False
```

PS We hope Transmission will clean up their Ubuntu build in 2025!

RELATED:

- PR #3845

OldKnobby, 2024-12-03:
Clean install worked this time.

holta, 2024-12-03:
Thanks @OldKnobby:

We'll set up some kind of "more live" community in the new year (2025), to try to help everyone worldwide?!

(e.g. something like Discord or Mastodon, but it's not yet decided exactly!)

OldKnobby, 2024-12-03:
Well, if you're looking for a tester, or feature suggestions let me know.  I mainly focus on Raspberry Pi stuff.