iiab/iiab #3856: hostapd would not start until I ran `rfkill unblock 1` on the latest Raspberry Pi OS (why?!)

holta, 2024-12-10:
FYI hostapd's systemd service would not start until I ran `rfkill unlock 1` to remove the **softblock of wlan** presumably arising from the latest version of Raspberry Pi OS?

In other words, hostapd started working as soon as I got `rfkill` output to appears as:

```
root@box:~# rfkill
ID TYPE      DEVICE      SOFT      HARD
 0 bluetooth hci0   unblocked unblocked
 1 wlan      phy0   unblocked unblocked
```

Suggests the following line(s) may need to be reworked?

https://github.com/iiab/iiab/blob/adfc1cc6962e2d91ab8ecc943fadb6d3dd72b394/roles/network/tasks/rpi_debian.yml#L56-L59

Others (Nov 25, 2024 thread below) would seem to running into very similar?

https://forums.raspberrypi.com/viewtopic.php?t=379975

BACKGROUND:

- I fresh installed MEDIUM-sized IIAB.
- Onto the latest (graphical version of) 64-bit Raspberry Pi OS.
- All apt updates were applied & bootloader upgraded before beginning.
- Raspberry Pi 4 Model B Rev 1.5

```
root@box:~# rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Mon 15 Apr 13:12:14 UTC 2024 (1713186734)
    LATEST: Mon 15 Apr 13:12:14 UTC 2024 (1713186734)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2711/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

tim-moody, 2024-12-10:
I noticed this occasionally with the previous release, but found that it resolved the second time I restarted hostapd.

holta, 2024-12-10:
> resolved the second time I restarted hostapd.

FYI in my case, rebooting and things like `systemctl restart hostapd` did not work.

holta, 2024-12-12:
Either way, we should probably update IIAB's code to run `rfkill unblock wifi` or `rfkill unblock wlan`

(Much as rpi-config does!)

jvonau, 2024-12-17:
Or you should actually set the country code in raspi-config prior to installing IIAB to allow the OS to unblock the wifi interface as intended by upsteam. Just my 5 cents worth due to inflation. Think the real reason is now the bluetooth device is appearing as `device 0`

jvonau, 2025-01-01:
Side note with the mass changing of milestones, that action really muddies the water to able to tell what release dealt with an issue, but with changing release numbers once a year there really is no line in the sand to gauge progress.  
Any good reason this issue is still open with #3867 out in the wild?

holta, 2025-01-01:
This issue can be closed.

Progress is good.  Even if there's no perfect solution given Raspberry Pi Imager is what it is (hopefully improving in 2025!)

closed 2025-01-01 by holta