iiab/iiab #3861: Unable to download maps: Resolved, sharing for others

unsigned8547, 2024-12-14:
### Expected behavior and actual behavior:

(As I was typing this up, the coffee kicked in and I spotted my error. I'm continuing to post here for reference in case anyone else encounters this issue. My solution is in the "steps to reproduce" section.)

Upon a fresh installation, I attempted to install the North American map region. The job appears in the queue and begins processing, but ultimately fails.

### Steps to reproduce the problem:

1) Install IIAB
2) Log into admin console
3) Click Install Content > Get Map Regions
4) Ensure "North America" (along with the required base regions) is checked
5) Click "Install Selected Region"
6) Wait 10 minutes
7) Click Utilities > Display Job Status
8) Click "refresh status"
9) Observe the maps jobs ("rclone -P copy contabo:iiab-maps..." have started, but have downloaded 0 bytes
10) Wait for the job to time out
11) Observe that no map data was downloaded, and all related jobs are in a failed state

I confirmed internet activity by virtue of the fact that I was able to download and run the installation script. I also confirmed connectivity to unrelated domains using wget. Then I ran a simple rclone command as root to confirm network connectivity to contabo and generate some logs. This is the result:

sudo rclone ls contabo: --log-file=/home/(username)/contabo.log --log-level DEBUG

```
2024/12/13 15:54:31 DEBUG : rclone: Version "v1.68.2" starting with parameters ["rclone" "ls" "contabo:" "--log-file=/home/(username)/contabo.log" "--log-level" "DEBUG"]
2024/12/13 15:54:31 DEBUG : Creating backend with remote "contabo:"
2024/12/13 15:54:31 DEBUG : Using config file from "/root/.config/rclone/rclone.conf"
2024/12/13 15:56:18 DEBUG : pacer: low level retry 1/2 (error operation error S3: ListBuckets, exceeded maximum number of attempts, 10, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/?x-id=ListBuckets": dial tcp: lookup usc1.contabostorage.com on 192.168.30.1:53: server misbehaving)
2024/12/13 15:56:18 DEBUG : pacer: Rate limited, increasing sleep to 10ms
2024/12/13 15:58:17 DEBUG : pacer: low level retry 2/2 (error operation error S3: ListBuckets, exceeded maximum number of attempts, 10, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/?x-id=ListBuckets": dial tcp: lookup usc1.contabostorage.com on 192.168.30.1:53: server misbehaving)
2024/12/13 15:58:17 DEBUG : pacer: Rate limited, increasing sleep to 20ms
2024/12/13 15:58:17 DEBUG : 4 go routines active
2024/12/13 15:58:17 NOTICE: Failed to ls: operation error S3: ListBuckets, exceeded maximum number of attempts, 10, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/?x-id=ListBuckets": dial tcp: lookup usc1.contabostorage.com on 192.168.30.1:53: server misbehaving
```

Note the lines:
 (error operation error S3: ListBuckets, exceeded maximum number of attempts, 10, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/?x-id=ListBuckets": dial tcp: lookup usc1.contabostorage.com on 192.168.30.1:53: server misbehaving)

While troubleshooting last night, I only noticed the "exceeded maximum number of attempts" and assumed this was some kind of rate limiting by Contabo. While typing up this post this morning, I noticed "lookup usc1.contabostorage.com" was failing with my DNS server IP. I manually ran nslookup from the server, and lo and behold, DNS resolution was failing. All other tested domains were fine.

**SOLUTION:**

After clearing cached DNS on my local nameserver, I went to my upstream provider NextDNS. I checked their logs, and noticed usc1.contabostorage.com was being blocked by their automated threat analysis. This is plausible, and not a failure on NextDNS' part; it's entirely conceivable that usc1.contabostorage.com was now or at some point in the past being used to host malware or other illicit content, and was consequently blocked.

I manually added usc1.contabostorage.com to the NextDNS allowlist, waited a few minutes for it to kick in, and boom -- success.

Not an IIAB issue, and not even really a NextDNS issue. Just an unexpected late-night issue that I wanted to share in case others experience the same deal.

unsigned8547, 2024-12-14:
Solution TL/DR: usc1.contabostorage.com was blocked as a threat by my upstream DNS provided. Added the domain to the allowlist override, and everything worked.

closed 2024-12-14 by unsigned8547

tim-moody, 2024-12-14:
Glad that someone with your skills was about to work around the problem, but I guess it could be a challenge for others. I have been wondering if we need to find a new storage provider as googling shows others have problems with Contabo.

unsigned8547, 2024-12-14:
I don't have any personal experience with Contabo cloud storage, but my experience with their VPSes has always been poor. Depending on your actual monthly transfers, B2 (by Backblaze) appears to be about half the cost of Contabo. 

If cost is the primary factor, it may be worth looking into bare metal infrastructure at one of the discount datacenters (Wholesale Internet / Nocix, OVH, Hetzner, etc). I've had good experiences with all of them. The big advantage there is that your IP is dedicated to you; there's little risk of disruption.

tim-moody, 2024-12-14:
We are spread across three data stores including Wasabi and Contabo with downloads using rsync and rclone. I would like to consolidate. Apparently Wasabi caps egress at 1x storage per month. Contabo has no cap afaik. Looks like B2 says 3X. I didn't think we had the volume at this point to justify bare metal, and I thought you had to provide your own equipment for colocation, but Wholesale looks like they are renting actual machines. Any chance you'd like to help up move in this direction?

unsigned8547, 2024-12-14:
Sure, I'd be happy to! I'd rather not post my email publicly, but if I reach out to Adam Holt, will he be able to put me in touch with you?

tim-moody, 2024-12-14:
yes, and please do.

tim-moody, 2025-01-04:
@unsigned8547 Still interested in hearing from you.