iiab/iiab #3868: Retest of update_server_time

tim-moody, 2024-12-19:
Tested on W2 installed with none local vars. Neither a connection to the home page by my Android phone nor my Win11 laptop updates the system time.

In order order to be activated the iiab parameter apache_allow_sudo must be True, but it is False in default vars and all localvars except medicine. The menu parameter allow_server_time_update, set via Admin Console, must also be True.

Once I changed apache_allow_sudo and ran a reinstall on ansible and checked allow_server_time_update, both my phone and laptop updated the system time on connection to the home page.

The name apache_allow_sudo is rather stale and could be changed to allow_www_data_sudo.

holta, 2024-12-19:
Valuable insights, thanks @tim-moody !

> The name apache_allow_sudo is rather stale and could be changed to allow_www_data_sudo.

I was tempted to do something just like that ~4 years ago:

- This is not the only situation where our legacy code confusingly uses the word "apache" to mean {nginx, webserver}.

- If some or all of these can be updated to be much more accurate, that'd be very positive (if there's time, this month or soon if possible!)

tim-moody, 2024-12-19:
My question is should this be turned off by default.

tim-moody, 2024-12-19:
>Should be on by default in my opinion

It used to be on by default I think, but was turned off to eliminate the poweroff button. I think it should be on in iiab and then either or both set time and poweroff can be turned on or off in Admin Console

tim-moody, 2024-12-19:
Unless testing was done with medicine local vars it is turned off, so of course does not set the time.

tim-moody, 2024-12-19:
#3869

jvonau, 2025-01-09:
Don't think [020_apache_poweroff ](https://github.com/iiab/iiab/blob/master/roles/www_options/templates/020_apache_poweroff.j2) is needed as there is a function for [poweroff within cmd-srv](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/js/admin_console.js#L2119-L2125) that makes this template file obsolete and maybe was the original reason of the default to False. I see that the conditional to install [021_apache_set_time.](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/templates/021_apache_set_time.j2) is dependent on  allow_www_data_sudo. I don't understand why the php based [set-server-time](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/services/set-server-time.php) is making call directly via a shell when cmd-srv exists and is running as root.

jvonau, 2025-01-09:
#3884

tim-moody, 2025-01-09:
>Don't think 020_apache_poweroff is needed

It is still needed. There are two ways to power off, via admin console and via home page.  In the latter case the user is not logged in, so can't use cmdsrv. This is also the reason that www-data can be enabled to set the time.

The reason poweroff is set to false by default is that some worry about the nuisance of kids powering off the server from the home page, and some don't.

Probably should have renamed these when I renamed allow apache to allow www-data.

jvonau, 2025-01-09:
> > Don't think 020_apache_poweroff is needed
> 
> It is still needed. There are two ways to power off, via admin console and via home page. In the latter case the user is not logged in, so can't use cmdsrv. This is also the reason that www-data can be enabled to set the time.
>

 https://github.com/iiab/iiab-admin-console/pull/594 lets try to keep the fiddling with the OS in one place
> The reason poweroff is set to false by default is that some worry about the nuisance of kids powering off the server from the home page, and some don't.

Valid reason but I feel the default should be disabled.
> Probably should have renamed these when I renamed allow apache to allow www-data.

If you are going to do that hold off till this is PR and one above are pushed.

tim-moody, 2025-01-10:
The whole reason for this retest was that people reported that set time wasn't working. Analysis showed that it was not working because it was not enabled because the default was off, and testing showed that when enabled it works fine. So setting the default to disabled is a step backwards.

holta, 2025-01-10:
Right.  Definitely field-sync'ing of system clock from nearby phones/devices should be enabled.

As schools ask for that repeatedly (even aware that it's a somewhat hacky solution, once their de facto very limited options — but still it's a life-saver that might improve over time!)

tim-moody, 2025-01-10:
>As long as poweroff is disabled on the homepage by default I'm good with that.

agreed, and it is currently disabled. (it is possible that if you install a preset that has it enabled you might get that value, but I'm not sure if such a preset exists.

jvonau, 2025-01-12:
> The whole reason for this retest was that people reported that set time wasn't working. Analysis showed that it was not working because it was not enabled because the default was off, and testing showed that when enabled it works fine. 

Not enabled or [not installed](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/tasks/main.yml#L131-L143)? I can see how the needed sudoers.d file would be missing with the default being False Personally I would of installed unconditionally or maybe with a different variable.

>So setting the default to disabled is a step backwards.

Yes, in the attempt to keep bad actors from shutting a machine down for fun. Visiting http://box.lan/js-menu/menu-files/services/power_off.php directly shuts the machine down without the need to visit any other page since power_off.php is supplied from this [repo](https://github.com/iiab/iiab/blob/master/roles/www_base/files/html/services/power_off.php)

holta, 2025-01-12:
> Visiting http://box.lan/js-menu/menu-files/services/power_off.php directly shuts the machine down without the need to visit any other page

This should really be fixed (especially now that it's being prominently advertised; to students who Google this exact ticket :-)

jvonau, 2025-01-12:
Have at it, you have the power to fix this oversight.

tim-moody, 2025-01-12:
So I think we are making progress. How about this for a design along the lines of @jvonau:

- settime and poweroff go back to being separate sudoer permissions.
- www_data_settime is always installed in sudoer.d. The home page behavior can be turned on or off in the menu. The chance for abuse is small.
- www_data_poweroff is only installed if the renamed parameter allow_www_data_poweroff is True, and its default is False. Implementers can decide whether to accept the risk of a nuisance or not, but to allow will require both an ansible run and a change to the home page json file.

holta, 2025-01-12:
That summary (from @tim-moody, 5 min above) looks great.

jvonau, 2025-01-12:
Seems reasonable, but I would request that admin-console not run the network role unconditionally when ansible [runs if that is still used](https://github.com/iiab/iiab/blob/master/iiab-from-console.yml) given there are functions added for networking within the console now. Lets not have too many cooks spoil the broth.

tim-moody, 2025-01-12:
I guess the other unresolved question is whether all of this sudoer logic really belongs in iiab-admin-console. (other than the default vars part). I think the install and configure of sudo should be in iiab along with password warning ? and the legacy ejabberdmoodle, but I'm OK to move settime, poweroff, and iiab-admin as sudoer to iiab-admin-console if that is the consensus.

tim-moody, 2025-01-12:
>I would request that admin-console not run the network role unconditionally

what condition? would the curl script have the same condition given that there are now quite a few network related one time scripts?

jvonau, 2025-01-12:
I ask specific questions and we can go from there:
Is iiab-from-console.yml used at all? If not please tag as unused. 
I don't think it is but adm-run-roles-tmp.yml does append the network role when Install Configured Options is pressed.
[image]

holta, 2025-01-12:
> I guess the other unresolved question is whether all of this sudoer logic really belongs in iiab-admin-console.

Installing IIAB without Admin Console is very common during testing especially.  And a few other scenarios as well.

So can we keep sudoer logic somewhere within iiab/iiab — e.g. much of this is currently under https://github.com/iiab/iiab/tree/master/roles/iiab-admin — and/or maybe some of this might work better later on, e.g. in www_options or similar?

> legacy ejabberdmoodle

Oops, what's this?  Is that something we overlooked when deprecating ejabberd years ago?!

jvonau, 2025-01-12:
> Installing IIAB without Admin Console is very common during testing especially. And a few other scenarios as well.
>
> So can we keep sudoer logic somewhere within iiab/iiab — e.g. much of this is [...]

The template's presence is already handled in www_options.

tim-moody, 2025-01-13:
>is iiab-from-console.yml used at all?

At the level of cmdsrv the answer would have to be yes. However, in practice it is not used in the sense that there is no front end control to invoke it.

>adm-run-roles-tmp.yml does append the network role when Install Configured Options is pressed

True. It is actually softcoded in that the front end requests it, but there is no logic to say when not to use it.

tim-moody, 2025-01-13:
There is a wrinkle here. The command `RUN-ANSIBLE-ROLES` does not run just any ansible role, only any of those listed in adm_const. The intent is to run roles, generally supporting content, that an admin user is expected to be able to configure. So kiwix, kolibri, etc. can be turned on or off, but the admin is not expected to change the directory structure the web server. 

sudoer functionality is spread across roles www_options and iiab-admin, neither of which will be run by `RUN-ANSIBLE-ROLES`. And I don't think allow_www_data_whatever  can be modified in the front end at present. So if you want to let www-data run poweroff, you will need to run one of the cli ansible commands.

jvonau, 2025-01-13:
> > is iiab-from-console.yml used at all?
> 
> At the level of cmdsrv the answer would have to be yes. However, in practice it is not used in the sense that there is no front end control to invoke it.
>
 
Figured that as the move was to runroles for faster runtime when installing new roles.
> > adm-run-roles-tmp.yml does append the network role when Install Configured Options is pressed
> 
> True. It is actually softcoded in that the front end requests it, but there is no logic to say when not to use it.

Perhaps the functions could be split between ICO to install new roles and new button "Apply Network Settings"? Just to suppress the possibility of some inadvertent user change within the networking section from shooting the end user in the foot. I'm seeing within other issues things like LANCONTROLLER being found and various other complaints now I've poked at some past issues.

jvonau, 2025-01-13:
> There is a wrinkle here. The command `RUN-ANSIBLE-ROLES` does not run just any ansible role, only any of those listed in adm_const. 

That is a bit of a fib, the network role is [not listed in adm_const.py.j2](https://github.com/iiab/iiab-admin-console/blob/master/roles/common/templates/adm_const.py.j2) 
>The intent is to run roles, generally supporting content, that an admin user is expected to be able to configure. 

Why then an admin who wants to only wants add content (kiwix in the above example) is subject to having the network role run?
>So kiwix, kolibri, etc. can be turned on or off, but the admin is not expected to change the directory structure of the web server.

Not really sure what you imply by this statement.
 > sudoer functionality is spread across roles www_options and iiab-admin, neither of which will be run by `RUN-ANSIBLE-ROLES`. 

Granted, but www_options could be in the same way usb_lib is used for usbmount but I'm unsure if the ability to change the settings is used anymore within the admin code base.  I'm sure openvpn went away but that is more of a red herring.
>And I don't think allow_www_data_whatever can be modified in the front end at present. 

Well get that fixed, your change to the defaults has opened up an avenue for bad actors, I just pointed it out.
>So if you want to let www-data run poweroff, you will need to run one of the cli ansible commands.

At present, yes. Should the default change you would need to pass through www_options to enable poweroff. Are you now suggesting hand editing local_vars.yml that you have in the passed suggested is dangerous of the end user to perform?

I get a funny feeling this is going to be dumped on the end user to fix up on their own for the next little while.

holta, 2025-01-14:
Thanks in advance to @EMG70 who has offered to further test in coming days — to help reconfirm the many PRs merged in recent days!

EMG70, 2025-01-14:
SUDO IIAB-DIAGNOSTICS - https://paste.centos.org/view/3b0d210a
Installed on Pi400,iiab size 1 via wifi 2.4GHZ,Country code set to HT(Haiti ) at RPI image building stage.
iiab tested and working OK.

holta, 2025-01-14:
Looks great!?

@EMG70 also mentions:

> After running sudo iiab-network, the ssid was already visible [i.e. IIAB's WiFi hotspot, on the air] even before doing a reboot

jvonau, 2025-01-14:
>DIRECTORY /etc/NetworkManager/system-connections FILES WILL FOLLOW...IF THEY EXIST
-IIAB--------------------------------------------------------------------------
-rw------- 1 root root 308 Nov 19 13:44 /etc/NetworkManager/system-connections/preconfigured.nmconnection
                        ...ITS LAST 100 LINES FOLLOW...
 [connection]
id=preconfigured
uuid=e9ab0002-f065-4d04-9597-122abd3ca1a1
type=wifi
[wifi]
mode=infrastructure
ssid=iiab-haiti
hidden=false
[ipv4]
method=auto
[ipv6]
addr-gen-mode=default
method=auto
[proxy]
[wifi-security]
key-mgmt=wpa-psk
psk=[REDACTED]

What country code is the AP ssid=iiab-haiti set to?
> =IIAB==========================================================================
-rw-r--r-- 1 root root 137601 Jan 14 19:30 /opt/iiab/iiab/iiab-network.log
DEVICE         TYPE      STATE                   CONNECTION
wlan0          wifi      connected               VM0624709  
lo             loopback  connected (externally)  lo        
br0            bridge    connected (externally)  br0        
p2p-dev-wlan0  wifi-p2p  disconnected            --        
eth0           ethernet  unavailable             --        
tailscale0     tun       unmanaged               --        
ap0            wifi      unmanaged               --        
 
>NAME                UUID                                  TYPE      DEVICE
VM0624709           9fb62ebe-a08c-4178-9a4c-bb385d1ec3c9  wifi      wlan0  
lo                  99514dc5-f463-48fc-b262-bf56241847db  loopback  lo    
br0                 cd80e5eb-2dfb-4e94-a4a6-335493c825f1  bridge    br0    
preconfigured       e9ab0002-f065-4d04-9597-122abd3ca1a1  wifi      --    
VM0624709_EXT       acf072b7-6a42-4df8-90f2-6f37e14adb6f  wifi      --    
Wired connection 1  98d65bcb-bf57-359f-aea1-07d61821c25f  ethernet  --    
default via 192.168.0.1 dev wlan0 proto dhcp src 192.168.0.78 metric 600
10.10.10.0/24 dev br0 proto kernel scope link src 10.10.10.10
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.78 metric 600 

There was some messing around, the above was taken from iiab-network.log, that is not the same as below which would be a snapshot of the system at the time of capture. I would expect to see the below.

>=IIAB==========================================================================
COMMAND: /usr/bin/networkctl    # systemd-networkd status
 IDX LINK       TYPE     OPERATIONAL SETUP
  1 lo         loopback carrier     unmanaged
  2 eth0       ether    no-carrier  unmanaged
  3 wlan0      wlan     routable    unmanaged
  4 br0        bridge   routable    configured
  5 ap0        wlan     enslaved    unmanaged
  6 tailscale0 none     carrier     unmanaged
 6 links listed.
 =IIAB==========================================================================
COMMAND: /usr/bin/nmcli d    # NetworkManager status
DEVICE         TYPE      STATE                   CONNECTION
wlan0          wifi      connected               VM0624709  
lo             loopback  connected (externally)  lo        
br0            bridge    connected (externally)  br0        
p2p-dev-wlan0  wifi-p2p  disconnected            --        
eth0           ethernet  unavailable             --        
tailscale0     tun       unmanaged               --        
ap0            wifi      unmanaged               --     

If the machine was rebooted hostapd should of been running without running using iiab-network. Testing environment and other details are scarce so taken with a grain of salt.

jvonau, 2025-01-14:
Can this issue be closed as the subject line seems to be resolved.

holta, 2025-01-14:
@EMG70 can you help clarify, to help us understand better:

- What exact hardware is providing your WiFi SSID iiab-haiti?  (Is this the 2.4 GHz home WiFi you used to install IIAB?)
- What country code is the AP ssid=iiab-haiti set to?  Check inside the WiFi router itself if necessary?
- Is there a reason you ran `iiab-network` e.g. right after installing IIAB, right before rebooting the RPi 400?

EMG70, 2025-01-15:
Sincere apologies for wasting time,testing method was greatly flawed. I should have put my home wifi details at creation of RPI image but instead named it haiti-iiab and selected HT country prefix.I then hooked up to my home wifi via available networks to install iiab,which was wrong and also ran iiab-network after.It's first time attempting iiab install via WiFi as i have been discouraged since i started.

EMG70, 2025-01-15:
I have reinstalled with country code ZW ,hope now better .
SUDO IIAB-DIAGNOSTICS- https.centos.org/view/1af1add7
Wifi hotspot only became visible after a reboot this time.

EMG70, 2025-01-15:
Below is country code details appearing in the RPI after installation:
FYI
```edmag@box:~ $ iw reg get
global
country ZW: DFS-ETSI
(2402 - 2482 @ 40), (N/A, 20), (N/A)
(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS

phy#0
country 99: DFS-UNSET
(2402 - 2482 @ 40), (6, 20), (N/A)
(2474 - 2494 @ 20), (6, 20), (N/A)
(5140 - 5360 @ 160), (6, 20), (N/A)
(5460 - 5860 @ 160), (6, 20), (N/A)
```

jvonau, 2025-01-15:
Having to reboot for hotspot active is the changed behavior when installing over wifi.
Related country code notes https://github.com/iiab/iiab/issues/3890#issuecomment-2591893757

jvonau, 2025-01-15:
The ap0 process predates  https://www.raspberrypi.com/documentation/computers/configuration.html#host-a-wireless-network-on-your-raspberry-pi by a couple of years.

EMG70, 2025-01-15:
I have installed iiab on RPI 3B ,done via wifi with country code set to HT(Haiti).
I entered home router details at image burning stage but i had to re-enter again after initial boot ,not sure why RPI didn't save the home WiFi details.

EMG70, 2025-01-15:
Before reboot:
Sudo iiab-diagnostics- https://paste.centos.org/view/0c051f53

After reboot
sudo iiab-diagnostics - https://paste.centos.org/view/95ff1446

jvonau, 2025-01-16:
@EMG70 Other than `ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting` nothing really jumps out. Seems like a good thing.
>    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=7aa1d893-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=HT

Since you want to play can you edit /boot/cmdline.txt and remove `cfg80211.ieee80211_regdom=HT` from the end of the line, save the file, reboot and post a dump if you want to.

EMG70, 2025-01-16:
SUDO IIAB-DIAGNOSTICS - https://paste.centos.org/view/36dc21ba
removed the line,now found in /boot/firmware/ and rebooted.
[image]
[image]

jvonau, 2025-01-16:
> SUDO IIAB-DIAGNOSTICS - https://paste.centos.org/view/36dc21ba removed the line,now found in /boot/firmware/ and rebooted. 

Ok so the location might of moved, haven't played for while, I recalled /boot/firmware/ as a ubuntu layout. 

Now lets test using US by editing country_code in both /etc/hostapd/hostapd.conf and /etc/hostapd/hostapd.conf.iiab and reboot. Using just `/usr/sbin/iw reg get` should be enough to see what I want but a dump would be ok also.

Now disable hostapd from stating with `sudo systemctl disable hostapd` and reboot. Using just `/usr/sbin/iw reg get` should be enough to see what I want but a dump would be ok also.

Reason for this is Hostapd is being started before NetworkManager so that daemon has the first chance to set the country code. Just seeing where the current behavior is at and validating my observations from years ago, but something else might be setting the country code somewhere else.

jvonau, 2025-01-16:
Thanks for the data point, reviewning