iiab/iiab #3871: Admin Console erroring with "Bad Gateway"

1n5aN1aC, 2024-12-23:
As background, I have this installed on a x86 platform, on Debian Trixie, and it has been working correctly.  
(Yes, I know it is not officially supported.  If you believe that to be the issue, let me know and I can go away.  :)

However, recently I have been unable to log into the admin page.

Troubleshooting:
- I go to box.lan/admin
- press f12 to view what is going on.
- Type a username of iiab-admin, enter password and press login.  

That returns "Invalid Login" on the screen, and the browser console clarifies it got a 502 (Bad Gateway)

Next I try this:
- Change the username to anything other than iiab-admin
- Press enter.

That returns "Invalid Login" on the screen, and the browser console now clarifies it got a 403 (Invalid Login)

`iiab-diagnostics`: https://paste.centos.org/view/21d278b1

1n5aN1aC, 2024-12-23:
Nevermind, upgrading using `iiab-update` worked!

closed 2024-12-23 by 1n5aN1aC

holta, 2024-12-23:
🙏