iiab/iiab #3876: Whenever I'm trying to access etc/iiab/local_vars.yml it says Permission Denied

isuzumac, 2025-01-04:
I am new to IIAB and when I'm trying to access etc/iiab/local_vars.yml through my raspberry pi terminal it says Permission Denied. How do I access this again? The reason I'm trying to get in is so I can change maps_from_internet_archive to true because I can't download any maps not even the base maps.

tim-moody, 2025-01-04:
It's risky to edit local_vars.yml manually, but to do so you must be root or use sudo.
I also doubt that maps_from_internet_archive  is actually used.

tim-moody, 2025-01-04:
Are maps installed and enabled? In Admin Console -> Configure -> Services Enabled do you see  IIAB Maps checked as in

[image]

isuzumac, 2025-01-04:
Yes maps are enabled, this is the message I get when attempting to install 
[image]
[image]

isuzumac, 2025-01-04:
I was reading it might have to have something to do with the DNS settings? I ran systemctl status dnsmasq in terminal and this is the message I got there 
[image]

holta, 2025-01-04:
@isuzumac can you run the following...

```
sudo iiab-diagnostics
```

...and then send us the resulting URL?

isuzumac, 2025-01-04:
Diagnostic Link
https://paste.centos.org/view/df33a110

isuzumac, 2025-01-05:
2025/01/04 15:50:44 ERROR : S3 bucket iiab-maps path osm-planet_z0-z10_2020.mbtiles: error reading source root directory: operation error S3: ListObjects, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/iiab-maps?delimiter=%2F&max-keys=1000&prefix=osm-planet_z0-z10_2020.mbtiles%2F": dial tcp: lookup usc1.contabostorage.com on 127.0.0.53:53: no such host
Transferred: 0 B / 0 B, -, 0 B/s, ETA -
Elapsed time: 0.7s
2025/01/04 15:50:44 ERROR : Attempt 3/3 failed with 1 errors and: operation error S3: ListObjects, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/iiab-maps?delimiter=%2F&max-keys=1000&prefix=osm-planet_z0-z10_2020.mbtiles%2F": dial tcp: lookup usc1.contabostorage.com on 127.0.0.53:53: no such host
Transferred: 0 B / 0 B, -, 0 B/s, ETA -
Errors: 1 (retrying may help)
Elapsed time: 0.7sTransferred: 0 B / 0 B, -, 0 B/s, ETA -
Errors: 1 (retrying may help)
Elapsed time: 0.7s
2025/01/04 15:50:44 NOTICE: Failed to copy: operation error S3: ListObjects, https response error StatusCode: 0, RequestID: , HostID: , request send failed, Get "https://usc1.contabostorage.com/iiab-maps?delimiter=%2F&max-keys=1000&prefix=osm-planet_z0-z10_2020.mbtiles%2F": dial tcp: lookup usc1.contabostorage.com on 127.0.0.53:53: no such host

isuzumac, 2025-01-05:
Above is the entire text IIAB Job Status tells me on why it failed

tim-moody, 2025-01-05:
what is the output of
`rclone tree contabo:`

tim-moody, 2025-01-05:
`rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles .`

downloaded the file for me.

isuzumac, 2025-01-05:
@tim-moody 
2025/01/05 10:56:14 NOTICE: Config file "/home/mac/.config/rclone/rclone.conf" not found using defaults

2025/01/05 10:56:14 CRITICAL: Failed to create file system for "contabo:": didn't find section in config file

Is the output of rclone tree contabo:

isuzumac, 2025-01-05:
@tim-moody  When attempting to download "rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles" I am not met with anything to download, below is the last 4 lines output, there is a bunch of flags above the last 4 lines I can input if needed

Use "rclone [command] --help" for more information about a command.

Use "rclone help flags" for to see the global flags.

Use "rclone help backends" for a list of supported services.

Command copy needs 2 arguments minimum: you provided 1 non flag arguments: ["contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles"]

holta, 2025-01-05:
@isuzumac any chance you can possibly give that rclone test download a quick try with another ISP?

(e g. If your mobile data carrier isn't too outrageously expensive 🫰 !?)

holta, 2025-01-05:
> @tim-moody 2025/01/05 10:56:14 NOTICE: Config file "/home/mac/.config/rclone/rclone.conf" not found using defaults
> 
> 2025/01/05 10:56:14 CRITICAL: Failed to create file system for "contabo:": didn't find section in config file
> 
> Is the output of rclone tree contabo:

Yikes!  Was there possibly a glitch during IIAB's usual process of installing & setting up rclone:

- Can you install IIAB in a quick little Ubuntu 24.04 VM to compare and validate what's happening in your Raspberry Pi 5 (ideally?!)

holta, 2025-01-05:
@isuzumac are you clarify if you're running rclone commands as used `iiab-admin` ?  Or as user `root` ?

isuzumac, 2025-01-05:
> @isuzumac can you clarify if you're running rclone commands as used `iiab-admin` ? Or as user `root` ?

Sorry for the noob question but how do I know if I'm using iiab-admin or user root?

holta, 2025-01-05:
> how do I know if I'm using iiab-admin or user root?

Run this command:

```
whoami
```

isuzumac, 2025-01-05:
It says I am mac which is the name I set my iiab to be I'm fairly certain

isuzumac, 2025-01-05:
So I'm pretty sure I am running as admin

isuzumac, 2025-01-05:
Here is what it says when attempting to run rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles

[image]
[image]

isuzumac, 2025-01-05:
Looks like I forgot to add the last period on the rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles . I just reran it and the output was the same as when I ran rclone tree contabo:

isuzumac, 2025-01-05:
2025/01/05 12:19:59 NOTICE: Config file "/home/mac/.config/rclone/rclone.conf" not found - using defaults

2025/01/05 12:19:59 CRITICAL: Failed to create file system for "contabo:": didn't find section in config file

isuzumac, 2025-01-05:
When running command as sudo I get the same Error word for word as I did in iiab job status

holta, 2025-01-05:
Can you please try running the rclone test command (1) as iiab-admin ...and also try running it (2) as root ?

CLARIF: To become user root, you can run...

```
sudo -i
```

tim-moody, 2025-01-05:
`sudo rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles .`

works for me when I am logged in as iiab-admin. please note the dot at the end which is probably your missing second argument

and `sudo rclone tree contabo:`

should return results

isuzumac, 2025-01-05:
@tim-moody That was indeed my missing argument, I reran it and it gave me the same error Job Status did in IIAB Admin Console, I did as @holta suggested and switched my internet to my phone hotspot and it started to work! I'm super confused as to why it wasn't working on my router, it's hardwired through ethernet, but instantly started to work with my hotspot? And ideas as to why it wouldn't download any maps files specifically? I installed a terabyte worth of other data no problem at all. Thank you both so very much for all of your effort and input! It makes me extremely excited to join the community! Again thank you both tremendously!!!

tim-moody, 2025-01-05:
@isuzumac sounds like your dns hypothesis has merit. Perhaps your router's dhcp doesn't hand out gateway and dns settings?

I regularly plug iiab rpi and other hardware into a switch connected to my router and it works fine. It also works when I have the rpi connect to the router's hotspot.

can you ping usc1.contabostorage.com (209.126.15.85)

holta, 2025-01-05:
Huge thanks @isuzumac for confirming! 🙏

Can you maybe let us know which ISP is blocking our rclone downloads in a private email (holt@unleashkids.org) to help us understand the landscape better if you can?

holta, 2025-01-06:
@isuzumac just FYI as we try to understand this better, here are 2 people who've experienced similar elsewhere on the planet:

- #3789 
- #3861

closed 2025-01-06 by isuzumac