iiab/iiab #3880: Adding a Web Page

isuzumac, 2025-01-08:
I am wanting to add some of my own websites to my IIAB. After scraping the web page and converting it into a json file how would I add it to the server to be accessed in http://box.lan? My plan was to scrape the web page on my desktop and put it on a USB then transfer to my raspberry pi.

holta, 2025-01-08:
Great question!  Quick Summary:

- If you convert the website to a proper ZIM files (one that works fully offline!) then our `FAQ #37` explains what to do next:

  "How do I add ZIM files, like Wikipedia?"
  https://wiki.iiab.io/go/FAQ#How_do_I_add_ZIM_files,_like_Wikipedia%3F

- If you convert the website to a file/folder tree (some people call these RACHEL modules) you'd typically put that within:

  `/library/www/html/modules`

Either way, the final step is generally to set up a menu item, for your IIAB main page: https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/ContentMenus.rst

(Does that help?)

holta, 2025-01-08:
SEE ALSO!

- "Can teachers display their own content?"
  https://wiki.iiab.io/go/FAQ#Can_teachers_display_their_own_content%3F

- "How do I add my own content?"
  https://wiki.iiab.io/go/FAQ#How_do_I_add_my_own_content%3F

isuzumac, 2025-01-08:
@holta This helps TREMENDOUSLY! Thank you!!