iiab/iiab #3881: Bogus host_country_code_applied = 98 applied

deldesir, 2025-01-08:
> host_country_code_applied = 98

That is a [detected value](https://github.com/iiab/iiab/blob/master/roles/network/tasks/hostapd.yml#L12C1-L15C22), the desktop the using NetworkManager, so the question becomes "Where did the bogus country code come from?"

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3875#issuecomment-2576727947_

deldesir, 2025-01-08:
Since my IIAB installation triggered this discussion, I believe adding some context might be helpful for debugging.

Before the installation, I set the country code to US in RPI Imager.
[image]

I installed IIAB via Wi-Fi. Could my router be forcing the detection of country code HT? Here is a relevant screenshot of my router settings:
[image]

jvonau, 2025-01-08:
Short story is what we have here I believe is a user created situation, RPI Imager should be run with the actual country code of the location in which the device will be deployed. Could you try repeating an install with RPI Imager set to HT and confirm all is now well when connected to your local AP please.

I've debugged NM behavior in the past for reference have a look at https://bugs.launchpad.net/netplan/+bug/1951586

tim-moody, 2025-01-08:
It would be nice to have a definitive list of codes (and channels). Some manufacturers have such lists and none I've seen includes a number or HT. e.g. https://www.arubanetworks.com/techdocs/InstantWenger_Mobile/Advanced/Content/Instant%20User%20Guide%20-%20volumes/Country_Codes_List.htm

https://en.wikipedia.org/wiki/List_of_WLAN_channels has channels, but no codes. I see 96 as a 5GHz channel, but not 98.

I'm not sure that https://countrycode.org/ is the same as WLAN country code. But it all probably depends on the libraries used by hostapd, NM,  or other.

I recommend running

`raspi-config nonint do_wifi_country US`

before the iiab install just to get a default value in place.

jvonau, 2025-01-08:
https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2

deldesir, 2025-01-08:
New install with country code set to HT in RPI Imager. The issue persists

https://paste.centos.org/view/10f73a20

jvonau, 2025-01-08:
Ok thanks for that, now can you run: 
`raspi-config nonint do_wifi_country HT` 
then
`sudo iiab-network` 
please paste the contents of iiab-network,log.
I believe there maybe some bug in RasPi OS's first boot scripts, this should help to narrow down if we have an issue in the networking code or not.

deldesir, 2025-01-08:
Done. Here is the iiab-network.log: https://paste.debian.net/hidden/a24a5f80/

jvonau, 2025-01-08:
> no_net_restart = True

Given this install is over WiFi the hotspot functionality is not expected to be operational until till after a reboot as the admin-console still needs to be installed and restarting the networking at this point in the install is pretty risky.

> tailscale_installed: True
discovered_wired_iface = tailscale0

Sidenote with the introduction of #3798 the networking code was not updated to exclude the interface that becomes created with this new reproach [ref](https://github.com/iiab/iiab/blob/1da4d45509868e5fe0fc5c6bf0ed71bf1fd9b8a1/roles/network/defaults/main.yml#L58)

deldesir, 2025-01-08:
New install using Wi-Fi router set to 2.4GHz fails: https://paste.centos.org/view/05c7f178

jvonau, 2025-01-08:
> host_country_code_applied = 98
'host_channel', 'value': '1'})
2025-01-08 14:04:54,170 p=1967 u=root n=ansible INFO| TASK [network : Add 'network' variable 'current_client_channel' value if defined, to /etc/iiab/iiab.ini] ***************
2025-01-08 14:04:54,730 p=1967 u=root n=ansible INFO| changed: [127.0.0.1] => (item={'option': 'client_wifi_channel', 'value': '1'})
2025-01-08 14:04:54,779 p=1967 u=root n=ansible INFO| TASK [network : Restart services] 

>2025-01-08 14:04:57,883 p=1967 u=root n=ansible INFO| TASK [network : Clone wifi if needed] **********************************************************************************
2025-01-08 14:04:58,948 p=1967 u=root n=ansible INFO| changed: [127.0.0.1]

and ap0 is not present, but the channel is correct. 

>[    6.319787] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2233.445488] warning: `cat' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[ 2356.632443] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2359.651995] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[ 2362.097312] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[ 2362.097465] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[ 2362.097545] brcmfmac: dongle trap info: type 0x4 @ epc 0x00021df8
[ 2362.097563] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 2362.097567] ieee80211 phy0: brcmf_c_set_joinpref_default: Set join_pref error (-5)
[ 2362.097573] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 2362.097578] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 2362.097581] ieee80211 phy0: brcmf_cfg80211_connect: BRCMF_C_SET_SSID failed (-5)
[ 2362.097767] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 2362.097772] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
[ 2363.367744] brcmfmac: F1 signature read @0x18000000=0x15294345
[ 2363.370777] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[ 2363.551183] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[ 2363.551572] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: May 14 2020 17:26:08 version 7.84.17.1 (r871554) FWID 01-3d9e1d87
[ 2363.656233] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

Looks like the firmware might be crashing when ap0 is being created, with this being a Pi-400 not sure what the next steps should be other than try the same routine on a non-desktop image as that is what was used testing ap0 on the pi-400 years ago as this is suppose to be a server. However I did dog food the desktop image back then but NetworkManager was not part of the OS then but worked correctly when running Ubuntu with a desktop that used NM at that time.

Good luck, I have better things to look forward to like my grandkids and have no wish to chase others dreams without remuneration.

holta, 2025-01-08:
Thanks @deldesir for really pushing hard on testing, to get to the very bottom here:

- Just for the heck of it, have you upgraded the EEPROM (bootloader) on your Raspberry Pi 400 using `rpi-eeprom-update -a` and then rebooted?  Confirm you have the latest bootloader installed by running: `rpi-eeprom-update`

- FYI IIAB install (using a home's 2.4GHZ Wi-Fi) to a Raspberry Pi 400 worked 2 weeks ago, so please keep trying until you find the root cause hopefully soon.

- If you really can't solve this, we can ask others to compare installing IIAB over WiFi to a different Raspberry Pi 400 (but so far that does not seem necessary?!)

holta, 2025-01-08:
> Sidenote with the introduction of #3798 the networking code was not updated to exclude the interface that becomes created with this new reproach [...]

```
virtual_network_devices: "-e wwlan -e ppp -e ap0 -e lo -e br0 -e tun -e br- -e docker -e bridge0 -e veth"
```

Thanks @jvonau: Tailscale's network interface will rarely be added until after IIAB is installed, but still let's do this to be safe.

Should we add `-e tailscale0` to the end of the string above?  (Or would you prefer something more general like `-e tailscale` in case something changes there in their naming in future?)

jvonau, 2025-01-08:
Shoot just noticed this is a stage2 image with no desktop but has NM active, I'll bet that a wired based install still works fine as does the hotspot functionally works also as cloning for ap0 needs to occur before connecting any WiFi based connections become active.

jvonau, 2025-01-09:
>     * FYI an IIAB install (using a home's 2.4GHZ Wi-Fi) to a Raspberry Pi 400 (desktop version of 64-bit Raspberry Pi OS) definitely worked 2 weeks ago, so please keep trying until you find the root cause hopefully soon.

Good to know, get a funny feeling the wifi firmware might be puking on the HT country code. 
>     * If you really can't solve this, we can ask others to compare installing IIAB over WiFi to a different Raspberry Pi 400 (but so far that does not seem necessary?!)

Was going to fire up some Rpi's but I can't find my hdmi cable so I can't really test anything but prepped a branch but can't open a pull request

jvonau, 2025-01-09:
and now I can #3883

tim-moody, 2025-01-09:
>get a funny feeling the wifi firmware might be puking on the HT country code

why I suggest to start with US, which is known good.  I don't think wifi  country code and regular country code are the same. There has to be regulatory channel data for the former.

As another data point, I did a full 256G content provision over upstream wifi on a W2 while connected to its hotspot as a client. I find wifi up_down works really well.

jvonau, 2025-01-09:
> As another data point, I did a full 256G content provision over upstream wifi on a W2 while connected to its hotspot as a client. I find wifi up_down works really well.

Considering the code was written awhile ago and handled the move to NetworkManager on RasPiOS the upkeep have been pretty minimal sans for the bogus country code issue that just popped up out of nowhere.

jvonau, 2025-01-12:
#3885

jvonau, 2025-01-13:
The above PR might/should correct the country code 98 issue. 
#3886 should deal with hostapd failing during install for the above and perhaps for others that were noted in the issues, might at references later. This one should be pushed as an intimidate bug.
#3887 deals with trimscale standalone

jvonau, 2025-01-13:
When I have time I might write some note and hints on what not to try playing around with and why.

holta, 2025-01-13:
Thank you to everyone who's invested time here over the past 10 days.

@deldesir now that PR #3887 is merged, would you have time to apply both these PRs below — to confirm that they (tested together) offer material improvement — e.g. for one of your original failed IIAB installs over 2.4GHz WiFi?

- PR #3885
- PR #3886

@deldesir conversely if you don't have time, please just let us know!

jvonau, 2025-01-13:
Testing 3886 first standalone should result in the install not failing but hostapd might still be using 98, then move to testing 3885, git pull and iiab-network would be the shortest testing path.

deldesir, 2025-01-14:
About to test on RPI5 and RPI 400. Will report when done.

jvonau, 2025-01-14:
I'm off to work soon, I'll check back in 12 hours.

deldesir, 2025-01-14:
Before reboot: https://paste.centos.org/view/0134f1fc

After reboot: https://paste.centos.org/view/d99c83aa

Hotspot active after reboot

holta, 2025-01-14:
Thank you @jvonau and @deldesir for making this happen:

1) I'll merge both PR's now:
   - #3885
   - #3886

2) @deldesir do let us know if you see any country code quirks in coming days/weeks that need further investigation! 💯

jvonau, 2025-01-14:
Before reboot 0134f1fc
> =IIAB==========================================================================
COMMAND: /usr/bin/sudo journalctl -b 0 -u networkd-dispatcher    # networkd-dispatcher log
 Jan 13 20:54:34 box systemd[1]: networkd-dispatcher.service - Dispatcher daemon for systemd-networkd was skipped because no trigger condition checks were met.

>2025-01-13 20:54:49,963 p=2185 u=root n=ansible INFO| TASK [network : Enable & Restart systemd-networkd.service] *************************************************************
2025-01-13 20:54:50,894 p=2185 u=root n=ansible INFO| changed: [127.0.0.1]
>

Most likely from: https://github.com/iiab/iiab/blob/master/roles/network/tasks/sysd-netd-debian.yml#L106
>2025-01-13 20:55:01,373 p=2185 u=root n=ansible INFO| TASK [network : Restart wpa_supplicant service] ************************************************************************
2025-01-13 20:55:02,000 p=2185 u=root n=ansible INFO| changed: [127.0.0.1] => (item=wpa_supplicant)
>

Now the routine is in https://github.com/iiab/iiab/blob/master/roles/network/tasks/restart.yml if you want to follow along. Next we have the above, recent prior testing suggests that https://github.com/iiab/iiab/commit/b14136adaeb08def15738e4df3aa03bb5346e7b9 might suppress the below traceback. 
>Jan 13 20:55:02 box systemd[1]: Starting networkd-dispatcher.service - Dispatcher daemon for systemd-networkd...
Jan 13 20:55:02 box networkd-dispatcher[37905]: NET-DISP-unmanaged lo carrier
Jan 13 20:55:02 box networkd-dispatcher[37908]: NET-DISP-unmanaged eth0 no-carrier
Jan 13 20:55:02 box networkd-dispatcher[37912]: NET-DISP-unmanaged wlan0 carrier
Jan 13 20:55:02 box networkd-dispatcher[37913]: NET-DISP-WiFi wlan0 carrier
Jan 13 20:55:02 box networkd-dispatcher[37914]: Cannot find device "ap0"
Jan 13 20:55:02 box networkd-dispatcher[37902]: WARNING:Exit status 1 from script '/etc/networkd-dispatcher/carrier.d/iiab-wifi' invoked with environment {'LANG': 'en_GB.UTF-8', 'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'NOTIFY_SOCKET': '/run/systemd/notify', 'INVOCATION_ID': 'b39c585619a749198a3f56343926bb8e', 'JOURNAL_STREAM': '8:81133', 'SYSTEMD_EXEC_PID': '37902', 'networkd_dispatcher_args': '--run-startup-triggers', 'ADDR': '', 'ESSID': 'ff/an', 'IP_ADDRS': '', 'IP6_ADDRS': '', 'IFACE': 'wlan0', 'STATE': 'carrier', 'AdministrativeState': 'unmanaged', 'OperationalState': 'carrier', 'json': '{"AdministrativeState": "unmanaged", "Driver": ["brcmfmac"], "ESSID": "ff/an", "Hardware Address": ["d8:3a:dd:d9:19:a7 (Raspberry Pi Trading Ltd)"], "IPv6 Address Generation Mode": ["none"], "InterfaceName": "wlan0", "Jan 13 20:54:50 box systemd-networkd[37127]": ["wlan0: Gained carrier", "wlan0: Connected WiFi access point: IIAB-Test-2.4GHz (06:c4:ea:27:c6:79)"], "Jan 13 20:54:51 box systemd-resolved[37186]": ["wlan0: Bus client set search domain list to: lan", "wlan0: Bus client set default route setting: yes", "wlan0: Bus client set DNS server list to: 192.168.1.1"], "Jan 13 20:55:01 box systemd-networkd[37127]": ["wlan0: Lost carrier", "wlan0: Gained carrier"], "Jan 13 20:55:01 box systemd-resolved[37186]": ["wlan0: Bus client reset search domain list.", "wlan0: Bus client set default route setting: no", "wlan0: Bus client reset DNS server list."], "Link File": ["n/a"], "MTU": ["1500 (min: 68, max: 1500)"], "Network File": ["n/a"], "Number of Queues (Tx/Rx)": ["1/1"], "Online state": ["unknown"], "OperationalState": "carrier", "Path": ["platform-1001100000.mmc"], "QDisc": ["pfifo_fast"], "State": "carrier (unmanaged)", "Type": "wlan", "WiFi access point": ["(null) (00:00:00:00:00:00)"]}'}
Jan 13 20:55:02 box networkd-dispatcher[37918]: NET-DISP-unmanaged tailscale0 carrier
Jan 13 20:55:02 box networkd-dispatcher[37921]: NET-DISP-configured br0 no-carrier
Jan 13 20:55:02 box systemd[1]: Started networkd-dispatcher.service - Dispatcher daemon for systemd-networkd.
Jan 13 20:55:03 box networkd-dispatcher[37902]: WARNING:Unknown index 6 seen, reloading interface list
Jan 13 20:55:03 box networkd-dispatcher[37987]: NET-DISP-pending ap0 off
Jan 13 20:55:03 box networkd-dispatcher[37990]: NET-DISP-pending ap0 no-carrier
Jan 13 20:55:03 box networkd-dispatcher[37995]: NET-DISP-unmanaged ap0 off
Jan 13 20:55:03 box networkd-dispatcher[38002]: NET-DISP-unmanaged ap0 no-carrier

>2025-01-13 20:55:02,024 p=2185 u=root n=ansible INFO| TASK [network : Enable & Restart networkd-dispatcher.service] **********************************************************
2025-01-13 20:55:02,684 p=2185 u=root n=ansible INFO| changed: [127.0.0.1]

Timestamps match up. Scary but harmless, the cloning for ap0 is the next step
>2025-01-13 20:55:02,706 p=2185 u=root n=ansible INFO| TASK [network : Clone wifi if needed] **********************************************************************************
2025-01-13 20:55:03,179 p=2185 u=root n=ansible INFO| changed: [127.0.0.1]

>Jan 13 20:55:13 box networkctl[38034]: Interface "ap0" not found.
Jan 13 20:55:13 box systemd[1]: networkd-dispatcher.service: Got notification message from PID 38034, but reception only permitted for main PID 37902
Jan 13 20:55:13 box networkd-dispatcher[37902]: ERROR:Failed to get interface "ap0" status: Command '['/usr/bin/networkctl', 'status', '--no-pager', '--no-legend', '--', 'ap0']' returned non-zero exit status 1.
Jan 13 20:55:13 box networkd-dispatcher[38035]: ap0       No such device
[... cut ...]
tcp6       0      0 fdc4:ea37:c679:10::7:53 :::*                    LISTEN      28763/dnsmasq      
tcp6       0      0 2803:9810:a0fa:ed10::53 :::*                    LISTEN      28763/dnsmasq      
tcp6       0      0 :::8083                 :::*                    LISTEN      33230/python3      
tcp6       0      0 fe80::af9:ac38:f10c::53 :::*                    LISTEN      28763/dnsmasq      
 
Now I have to wonder if having tailscale active is somehow interfearing with the cloning process as this never occured with the older vpn that used tun0 as the tunnel interface.

tim-moody, 2025-01-14:
I twice built a 'none' server on rpi 5 and then moved the card to W2. In both cases hostapd did not come up the first time after the move. I think I had experienced this before and it was the reason that I had cmdsrv restart hostapd. After power cycling hostapd comes up.

I also connected to an upstream router from Admin Console. I notice that ap0 is reported as the hostapd device and the upstream device is wlan0. Since this is now the gateway will that affect no_net_restart and any restarting of hostapd.

jvonau, 2025-01-14:
>Since this is now the gateway will that affect no_net_restart and any restarting of hostapd. 

Yes during the running of the network role auto starting of hostapd is now suppressed when there is a WiFi gateway present as that is what was blocking the install routine from getting to install admin-console. A reboot does bring up ap0 and hostapd with the older firmware in place so I think this is a change with the shipped firmware.

holta, 2025-01-14:
> The firmware is being rolled back [after installing IIAB onto an RPi 5]

Thanks @jvonau for catching that:

- In 2020, Raspberry Pi 400 introduced the 43456 WiFi chip, which I tested (around the very end of 2020) to be able to support fully 32 simultaneous "student" WiFi client devices, right out of the box with the stock WiFi firmware that comes with RasPiOS.
- I (incorrectly) assumed that newer Raspberry Pis (e.g. RPi 5 from Sept/Oct 2023 and RPi 500 from Dec 2024) would do likewise, using a more modern WiFi chip, e.g. something like the 43456 chip or similar, "as progress marches forward."
- But that's not what happened in the end.  In fact, recent Raspberry Pi hardware (e.g. RPi 5 and 500) both appear to have reverted to the original/older 43455 WiFi chip.

Really good to know!

ASIDE: RPi 5 and 500 are much too power-hungry for low-income communities needing a low electrical power footprint, low environmental footprint, and low finanicial footprint — e.g. developing world schools depending on portable battery banks is an extremely common real-world use case.  (Just FYI RPi 4 or even RPi 3 are much better solutions than than RPi 5, in those cases.)

REGARDLESS: We should definitely try to arrange for some real-world WiFi capacity testing this year (e.g. maximum number of student WiFi devices that can connect simultaneously) to truly confirm RPi 5 and 500 WiFi hotspot behavior, one way or another.  One Possibility: IIAB might already be doing just what schools need — i.e. ensuring that their 43455 WiFi chips (just like inside RPi 4) are continuing to work well serving entire classrooms with large numbers of student WiFi devices — conscientiously avoiding very problematic Raspberry Pi OS default WiFi firmware.  Either way, our documentation definitely will need to be clarified, tightened up, and made more friendly wherever we can. ✅

jvonau, 2025-01-15:
Can we dispense with the grandstanding and get back to engineering?

- first issue with some out of the box firmware cloning ap0 appears to be failing for whatever reason during installation knocking out hostapd till older firmware is booted. I'd like someone to try an install with `pi3bplus_rpi4_wifi_firmware: os` just get a better understanding of what the newest firmwhere can and can't do as a reference point.
>#rpi3bplus_rpi4_wifi_firmware: 24    # REQUIRES "wifi_up_down: False" BELOW!
#rpi3bplus_rpi4_wifi_firmware: 32    # UNRELIABLE (INTERMITTENT) with 2021+ OS's

Please explain what the comments are about? 
- second `ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting` 
Speaks for itself, like to get to the bottom but not going to spend my time with that.

- third https://github.com/iiab/iiab/commit/b14136adaeb08def15738e4df3aa03bb5346e7b9 might be needed with the current needs of the os

holta, 2025-01-15:
> documentation definitely will need to be clarified, tightened up, and be made more friendly

Just FYI with so many new Raspberry Pi hardware choices appearing almost every single year, in coming months I want to consider renaming these 2 Ansible variables:

- Rename `rpizerow_rpi3_wifi_firmware` to `wifi_firmware_43430`
  (e.g. so that we're also welcoming the RPi Zero 2 W; which also uses the 43430 WiFi chip!)
- Rename `rpi3bplus_rpi4_wifi_firmware` to `wifi_firmware_43455`
  (e.g. so that we're also welcoming the RPi 5 and RPi 500; which also use the 43455 WiFi chip!)

RECAP: An equivalent variable should not be needed for RPi 400 etc (e.g. `wifi_firmware_43456` is not currently needed) as these already "Just Work" in large classrooms, when hostapd etc are used with the out-of-the-box WiFi firmware included with Raspberry Pi OS.