iiab/iiab #3882: Uanble to access Admin Console

isuzumac, 2025-01-09:
I am all of a sudden experiencing being locked out of admin console, when I first set up iiab I changed the password and have been able to login with no issues but now it's telling me "invalid login"

holta, 2025-01-09:
@isuzumac, Does changing your password at the Linux command-line help?

(In other words, use the `passwd` command while logged in as the user in question!)

- iiab/iiab-admin-console#593

isuzumac, 2025-01-09:
> @isuzumac, Does changing your password at the Linux command-line help?
> 
> (In other words, use the `passwd` command while logged in as the user in question!)
> 
> * [Invalid Login Admin Console  iiab-admin-console#593](https://github.com/iiab/iiab-admin-console/issues/593)

It does not unfortunately.

holta, 2025-01-09:
What username are you using?  Screenshot(s) of failure please?

isuzumac, 2025-01-09:
Using user name iiab-admin
[image]

holta, 2025-01-09:
How exactly are you setting the password for `iiab-admin` ?

Screenshot if possible, including output of the `whoami` command?

isuzumac, 2025-01-09:
Screenshot shows how I am setting the password 
[image]

holta, 2025-01-09:
@tim-moody any idea what's happening here?

holta, 2025-01-09:
@isuzumac if you don't figure this out on your own after rebooting etc — we can arrange a VPN-like connection to log into your IIAB directly, if you want us to diagnose in coming days?

isuzumac, 2025-01-09:
That would be awesome!! Please let me know what time works best for you guys!! 

I ran sudo iiab-update to try and resolve my issues and was met with this.
[image]

holta, 2025-01-09:
@isuzumac are you running `iiab-update` as root?  (e.g. did you run `sudo iiab-update` ?)

isuzumac, 2025-01-09:
Yes I ran `sudo iiab-update`

holta, 2025-01-09:
@isuzumac

Something seriously damaged your `/etc/default/locale` (did you install any old software possible?!)
In any case, I recovered as follows:

1. I logged into your Raspberry Pi 5 and fixed your `/etc/default/locale` file to be:

```
#  File generated by update-locale
LANG=en_US.UTF-8
```

Whereas your highly problematic file had been:

```
#  File generated by update-locale
#LANG=en_US
LC_ALL=en_US
LANGUAGE=en_US
```

2. Then I logged back in (completely new session) to confirm a proper locale — e.g. run commands like `locale` and `localectl` to check.

3. I reinstalled Ansible:

```
cd /opt/iiab/iiab
sudo scripts/ansible
```

4. I reinstalled Admin Console

```
cd /opt/iiab/iiab-admin-console
sudo ./install
```

**All set?  If so, please now RUN THESE COMMANDS:** (to turn off the VPN, and to set your iiab-admin password!)

```
sudo tailscale down
sudo passwd iiab-admin
```

closed 2025-01-09 by holta

holta, 2025-03-01:
Very useful tips from 2021 and 2023, from apt-listchanges: (I updated [FAQ.IIAB.IO NUMBER 12](https://wiki.iiab.io/go/FAQ#What_to_do_if_Raspberry_Pi_OS_messes_up_language_locale%3F) accordingly)

```
locales (2.37-6) unstable; urgency=low

  Starting with locales 2.37-6, locale variables are now stored in
  /etc/locale.conf and /etc/default/locale will be converted to a symlink
  to the former. This is to harmonize locale support configuration across
  distributions.

 -- Aurelien Jarno <aurel32@debian.org>  Fri, 14 Jul 2023 17:04:23 +0200

locales (2.31-14) unstable; urgency=low

  Starting with locales 2.31-14, non UTF-8 locales are deprecated and not
  offered anymore in the debconf dialog, except for the ones already
  configured. Nevertheless users of non UTF-8 locales are encouraged to
  switch their system to an UTF-8 locale.

  Please note that iconv still supports conversion to and from non UTF-8
  charset. For instance reading a file using an ISO-8859-15 charset can be
  done with: iconv --from-code=ISO-8859-15 foobar.txt

 -- Aurelien Jarno <aurel32@debian.org>  Tue, 17 Aug 2021 16:27:59 +0200
```

Related:

- #3917