iiab/iiab #3890: Getting long winded - Summary of issues

jvonau, 2025-01-15:
Can we dispense with the grandstanding and get back to engineering?

- first issue with some out of the box firmware cloning ap0 appears to be failing for whatever reason during installation knocking out hostapd till older firmware is booted. I'd like someone to try an install with `pi3bplus_rpi4_wifi_firmware: os` just get a better understanding of what the newest firmwhere can and can't do as a reference point.
>#rpi3bplus_rpi4_wifi_firmware: 24    # REQUIRES "wifi_up_down: False" BELOW!
#rpi3bplus_rpi4_wifi_firmware: 32    # UNRELIABLE (INTERMITTENT) with 2021+ OS's

Please explain what the comments are about? 
- second `ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting` 
Speaks for itself, like to get to the bottom but not going to spend my time with that.

- third https://github.com/iiab/iiab/commit/b14136adaeb08def15738e4df3aa03bb5346e7b9 might be needed with the current needs of the os
- forth https://github.com/iiab/iiab/blob/0c4df6c42cab74019034161e652852c7635fe969/roles/network/tasks/restart.yml#L17-L29
Maybe swapping the order might settle some of the extra noise in the logging.

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3881#issuecomment-2591346549_

holta, 2025-01-15:
> > #rpi3bplus_rpi4_wifi_firmware: 24    # REQUIRES "wifi_up_down: False" BELOW!
> > #rpi3bplus_rpi4_wifi_firmware: 32    # UNRELIABLE (INTERMITTENT) with 2021+ OS's
> 
> Please explain what the comments are about?

The comments above really do say almost everything there is to say.  But if long form phrasing helps:

- `rpi3bplus_rpi4_wifi_firmware: 24` is a WiFi firmware (for the 43455 WiFi chip) that requires `wifi_up_down: False` — in other words this firmware does not support AP+STA.

- `rpi3bplus_rpi4_wifi_firmware: 32` is a WiFi firmware (for the 43455 WiFi chip) that proved to be unreliable, when tested with 2021 Raspberry Pi OS's.

jvonau, 2025-01-15:
> The comments above really do say almost everything there is to say. But if long form phrasing helps:
>
>     * `rpi3bplus_rpi4_wifi_firmware: 24` is a WiFi firmware (for the 43455 WiFi chip) that [...]

That would make this firmware probably unsuitable in a desktop environment where wlan0 is expected to usable. Would make a testing point of using `wifi_up_down: False` during an install. The install should work but hotspot-on would need to be ran and **the upstream wifi connection info removed** for a working hotspot.

holta, 2025-01-15:
Yes, that sounds exactly right.

jvonau, 2025-01-15:
I'm going suggest having the firmware replacement be the last action so there is a matched set if firmware and blobs at network runtime. The next action would be to have wifi_up_down forced false when selected firmware demands it.

jvonau, 2025-01-15:
Related: #2853

jvonau, 2025-01-15:
The use of ap0 predates https://www.raspberrypi.com/documentation/computers/configuration.html#host-a-wireless-network-on-your-raspberry-pi by a couple of years.

jvonau, 2025-01-15:
Think both pre-reboot and post-reboot diagnostic dumps would be useful to see if there is an issue with 
- ap0 being created 
- a hotspot without ap0 being created with the stock firmware.

jvonau, 2025-01-15:
https://github.com/iiab/iiab/issues/3868#issuecomment-2590983056 3b0d210a shows 3 distinct connections tried:
>-rw------- 1 root root 308 Nov 19 13:44 /etc/NetworkManager/system-connections/preconfigured.nmconnection

This one would be from raspi-imager
>-rw------- 1 root root 269 Jan 14 18:28 /etc/NetworkManager/system-connections/VM0624709_EXT.nmconnection
 -rw------- 1 root root 282 Jan 14 18:26 /etc/NetworkManager/system-connections/VM0624709.nmconnection

This two I would assume came from messing around within the desktop.
> isplay-manager? active   Arch1: arm64   Arch2: armhf

jvonau, 2025-01-15:
Day job commitments puts constraints on the amount of time to toss at this project.

holta, 2025-01-15:
> I'm going suggest having the firmware replacement be the last action so there is a matched set if firmware and blobs at network runtime.

Is the idea to increase debugging feedback via iiab-diagnostics, allowing for diagnostic details from both firmwares?

(One iiab-diagnostics before reboot and another after reboot.)

jvonau, 2025-01-16:
> > I'm going suggest having the firmware replacement be the last action so there is a matched set if firmware and blobs at network runtime.
> 
>     * Is the idea to increase debugging feedback via iiab-diagnostics, allowing for diagnostic details from both firmwares?
>       (One iiab-diagnostics before reboot and another after reboot.)
> 
>     * I can't remember, does trying a different WiFi firmware effectively require a reboot at the moment?  e.g. if so, is this a bit like kernel option `cfg80211.ieee80211_regdom=HT` mentioned just above? ([Getting long winded - Summary of issues [4 WiFi & firmware questions] #3890 (comment)](https://github.com/iiab/iiab/issues/3890#issuecomment-2591893757))

More or less yes on all accounts. Firmware is loaded on boot and kept in memory same with country code via the cmdline Could try https://github.com/iiab/iiab/issues/3868#issuecomment-2594315300 as the rfkill block is effectively disabled.

Side note: Might want to look at raspi-config, some of the other first boots scripts or pi-gen to get an idea why you can't just swap an image between various different models, think I know the answer but I'm to lazy to do other's homework for them. Is it really that hard to re-run raspi-config after swapping between machines?

holta, 2025-01-16:
Just FYI, Nzola sent these:

Before reboot...
https://paste.centos.org/view/02aa0527

After reboot...
https://paste.centos.org/view/4670163c

(It shows `host_country_code_applied = US` but WiFi country code ends up as CA.  I don't know exactly how he proceeded.)

holta, 2025-01-16:
Nzola wonders why WiFi country code "CD" (for Democratic Republic of Congo) did not appear in the end.

As he set "WiFi LAN country: CD" in Raspberry Pi Imager, prior to flashing the SD card:

[image]

jvonau, 2025-01-16:
> Just FYI, Nzola sent these:
> (It shows `host_country_code_applied = US` but WiFi country code ends up as CA. I don't know exactly how he proceeded.)

Given he is Canada that would follow my past observed behavior in 
https://bugs.launchpad.net/netplan/+bug/1951586/comments/16
https://bugs.launchpad.net/netplan/+bug/1951586/comments/18

jvonau, 2025-01-16:
#3892 replacement for https://github.com/iiab/iiab/commit/b14136adaeb08def15738e4df3aa03bb5346e7b9

jvonau, 2025-01-16:
> Nzola wonders why WiFi country code "CD" (for Democratic Republic of Congo) did not appear in the end.
> 

I'm assuming that is a drop down pick-list to choose from but the cmdline was altered as noted above. Ask upstream about HT and CD and add the noted "Firmware rejected country setting" to the bug report.

jvonau, 2025-01-16:
https://github.com/iiab/iiab/pull/3892#issuecomment-2596211157

jvonau, 2025-01-16:
https://github.com/iiab/iiab/issues/3895#issuecomment-2596396356

holta, 2025-01-17:
ASIDE: RasPiOS 13 (based on Debian 13 / Trixie) is not too many months away; approaching the end of its 2-year release cycle.

We might need to keep an eye on that by mid-2025?  (e.g. if WifI country code / ETC pragmatic realities... are still sifting out.)

jvonau, 2025-01-19:
https://github.com/iiab/iiab/pull/3892#issuecomment-2600724458 Would be a good idea to be running the latest rpi-imager

jvonau, 2025-01-19:
https://wireless.docs.kernel.org/en/latest/

holta, 2025-01-19:
> [#3892 (comment)](https://github.com/iiab/iiab/pull/3892#issuecomment-2600724458) Would be a good idea to be running the latest rpi-imager

Definitely.

- Even if there's some ambiguity there as rpi-imager 1.9.0 was released 2024-09-11 but not fully mainlined (RasPiOS ships 1.9.0, but 1.8.5 is offered from their download site for Windows & macOS).  

- Back in September, they talked about 1.9.1 and 1.9.2 being imminent, but that hasn't quite happened yet.

jvonau, 2025-01-19:
Quick down and dirty:
`grep country /home/jvonau/Downloads/db.txt | sed s"|country||" | sed s"|:||" | awk '{print $1}' > db.clean`
`cat /usr/share/zoneinfo/iso3166.tab | awk '{print $1}' > iso.clean`
`diff -u db.clean iso.clean > clean.diff.txt`
Anything with a + is in iso but not in db

[clean.diff.txt](https://github.com/user-attachments/files/18469367/clean.diff.txt)

jvonau, 2025-01-19:
`cat ~/clean.diff.txt | grep + | awk '!/@/'| sed s'|+||' | awk '!/#/' > real_clean.txt`
[real_clean.txt](https://github.com/user-attachments/files/18470294/real_clean.txt)

One line at the top, good enough. Work now.

jvonau, 2025-01-22:
https://github.com/iiab/iiab/issues/3900#issuecomment-2600849496
`wget https://raw.githubusercontent.com/raspberrypi/rpi-imager/refs/heads/qml/src/countries.txt`
`diff -u db.clean countries.txt | awk '!/@/' > imager-diff.txt` db.clean from above.

[imager-diff.txt](https://github.com/user-attachments/files/18511346/imager-diff.txt) + are in raspi-imager's countries.txt while - are absent.

jvonau, 2025-01-23:
@holta When are you opening issues around the bogus countries in raspi-imager and raspi-config that are not part of [wireless-regdb](https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/)'s [db.txt](https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/plain/db.txt)?

https://github.com/iiab/iiab/issues/3890#issuecomment-2601007839
[real_clean.txt](https://github.com/user-attachments/files/18470294/real_clean.txt) Anything with a + are in iso  but not in db.txt.

https://github.com/iiab/iiab/issues/3890#issuecomment-2608227256 
[imager-diff.txt](https://github.com/user-attachments/files/18511346/imager-diff.txt) Anything with a + are in raspi-imager's countries.txt but not in db.txt.

These are a bit of a problem as wpa_supplicant logging is getting spammed really bad and leaves 00 as an unset set country code.

closed 2025-01-23 by jvonau