iiab/iiab #3894: Manage Raspberry Pi  Wi-Fi Capabilities

tim-moody, 2025-01-16:
### Overview

IIAB has pushed the envelop on increasing and optimizing the capacity of Raspberry Pi devices for number of connections and duplicate use of the built-in Wi-Fi chip. As of January, 2025 there are a number of changes that need to be understood and integrated into IIAB implementations.

### Analysis and Documentation
- [ ] Rpi Wi-Fi Tools and Commands
- [ ] Rpi 5  Wi-Fi Firmware under raspios November 19th 2024
- [ ] Rpi W2 Wi-Fi Firmware under raspios November 19th 2024
- [ ] Reconcile IIAB supported firmware to currently installed

### List of Current Approaches
- [ ] #3890 
- [ ] #3892 
- [ ] #3891 
- [ ] #3881

jvonau, 2025-01-16:
Related: https://github.com/iiab/iiab/pull/3893

holta, 2025-01-16:
The central question (so far, quick high-level recap over 10+ days) would appear to be:

- WiFi country codes have been going seriously haywire[*], during WiFi installs of IIAB.

- Ethernet cable installs of IIAB do not seem to be affected.

[*] Here are a few examples of WiFi country code going haywire, in a few different ways... !

- Bogus WiFi country code 98 (#3881)
- Incorrect WiFi country code — e.g. CA instead of CD as requested, or US as "applied" (#3890)
- Valid WiFi country codes are _sometimes_ being rejected entirely — e.g. HT, CD (#3868, #3890)

tim-moody, 2025-01-16:
>Valid WiFi country codes are sometimes being rejected entirely — e.g. HT, CD

* with current OS is it not always instead of sometimes?
* what makes you think they are valid?

jvonau, 2025-01-16:
Think the quick take away is the cmdline works to set the `iw reg` value thus that is what is displayed when queried. Having the firmware spitting out rejected country is a different issue and I suspect might be the ground zero of most of these issues.