iiab/iiab #3895: Rpi Wi-Fi Tools and Commands

tim-moody, 2025-01-16:


tim-moody, 2025-01-16:
* List wireless interfaces
`/usr/sbin/iw dev`
* Detected WiFi country code / legal frequencies
`/usr/sbin/iw reg get`

* Get firmware version	
`dmesg | grep Firmware`
`strings <firmware file> |grep Version`
`cat /sys/firmware/devicetree/base/model`

jvonau, 2025-01-16:
Think its time to grab some more info around wpa_supplicant, think there is a log file for that daemon and broader system logging from whatever is doing the logging nowadays. Need get down to the level of detail I used in the past with the upstream issue noted in https://github.com/iiab/iiab/issues/3890#issuecomment-2595417428

holta, 2025-01-16:
Yes.  Three different country codes appearing in [Nzola's iiab-diagnostics overnight](https://github.com/iiab/iiab/issues/3890#issuecomment-2594637863) would appear to be a very useful "smoking gun" / canary-in-coalmine:

 - `CA` was definitely not what he wanted
 - when he set `CD` using rpi-imager (for country DRC)
 - and then he observed `host_country_code_applied = US`
   - ...which was not the truth in the end: reboot turned on hostapd with `CA` — presumably as a result on his home router in Canada.

jvonau, 2025-01-16:
https://github.com/iiab/iiab/issues/3868#issuecomment-2596100972
Useful data point with:
00 country code when booted without the cmdline present and hostapd disabled. 
US country code when booted without the cmdline present and hostapd enabled.

jvonau, 2025-01-17:
https://github.com/iiab/iiab/issues/3897#issuecomment-2597235685 00 country code when booted without the cmdline present

jvonau, 2025-01-17:
https://github.com/iiab/iiab/issues/3897#issuecomment-2598171506 suggested to be added to dumps.

holta, 2025-01-23:
> * List connected devices

_hostapd_cli_ is a lifesaver too, as we discovered[1][2] five years ago, so maybe add this to the list too:

```
hostapd_cli all_sta | grep : | wc
hostapd_cli -i ap0 list_sta | wc
```

...so long as /etc/hostapd/hostapd.conf contains: (which IIAB now does automatically!)

```
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
```

PS Looking into arp's cache is similar, but usually not as accurate or as useful:

```
arp -na | grep br0 | wc
```

[1] https://github.com/iiab/iiab/issues/823#issuecomment-590668335
[2] https://github.com/iiab/iiab/issues/823#issuecomment-662285202