iiab/iiab #3896: Rpi 5  Wi-Fi Firmware under raspios November 19th 2024

tim-moody, 2025-01-16:


tim-moody, 2025-01-16:
* Raspberry Pi: 5
* OS: 2024-11-19-raspios-bookworm-arm64-lite.img
* apt upgrade: Yes
* IIAB Installed: No

tim-moody, 2025-01-16:
```
cat /sys/firmware/devicetree/base/model
Raspberry Pi 5 Model B Rev 1.0

root@raspberrypi:~# dmesg | grep Firmware
[    0.808540] raspberrypi-firmware soc:firmware: Firmware hash is 5be4f30400000000000000000000000000000000
[    5.383094] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b

```

tim-moody, 2025-01-16:
```
root@raspberrypi:~# ll /lib/firmware/cypress/
total 6948
drwxr-xr-x  2 root root   4096 Nov 19 13:33 .
drwxr-xr-x 47 root root   4096 Nov 19 13:33 ..
-rw-r--r--  1 root root 574096 Aug 10 06:08 cyfmac4339-sdio.bin
-rw-r--r--  1 root root 601704 Aug 10 06:08 cyfmac4354-sdio.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac4354-sdio.clm_blob
-rw-r--r--  1 root root 616174 Aug 10 06:08 cyfmac4356-pcie.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac4356-pcie.clm_blob
-rw-r--r--  1 root root 580258 Aug 10 06:08 cyfmac4356-sdio.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac4356-sdio.clm_blob
-rw-r--r--  1 root root 563834 Aug 10 06:08 cyfmac4373-sdio.bin
-rw-r--r--  1 root root   4684 Aug 10 06:08 cyfmac4373-sdio.clm_blob
-rw-r--r--  1 root root 490686 Aug 10 06:08 cyfmac43012-sdio.bin
-rw-r--r--  1 root root   4692 Aug 10 06:08 cyfmac43012-sdio.clm_blob
-rw-r--r--  1 root root 409717 Aug 10 06:08 cyfmac43340-sdio.bin
-rw-r--r--  1 root root 202623 Aug 10 06:08 cyfmac43362-sdio.bin
-rw-r--r--  1 root root 399344 Aug 10 06:08 cyfmac43430-sdio.bin
-rw-r--r--  1 root root   4733 Aug 10 06:08 cyfmac43430-sdio.clm_blob
-rw-r--r--  1 root root 229520 Aug 10 06:08 cyfmac43439-sdio.bin
-rw-r--r--  1 root root   1438 Aug 10 06:08 cyfmac43439-sdio.txt
-rw-r--r--  1 root root   3208 Aug 10 06:08 cyfmac43439-sdio.clm_blob
lrwxrwxrwx  1 root root     38 Aug 10 06:08 cyfmac43455-sdio.bin -> /etc/alternatives/cyfmac43455-sdio.bin
-rw-r--r--  1 root root 548728 Aug 10 06:08 cyfmac43455-sdio-minimal.bin
-rw-r--r--  1 root root 609309 Aug 10 06:08 cyfmac43455-sdio-standard.bin
-rw-r--r--  1 root root   2676 Aug 10 06:08 cyfmac43455-sdio.clm_blob
-rw-r--r--  1 root root 558388 Aug 10 06:08 cyfmac43570-pcie.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac43570-pcie.clm_blob
-rw-r--r--  1 root root 615882 Aug 10 06:08 cyfmac54591-pcie.bin
-rw-r--r--  1 root root   5764 Aug 10 06:08 cyfmac54591-pcie.clm_blob
```

tim-moody, 2025-01-16:
```
root@raspberrypi:~# ll /lib/firmware/brcm/
total 13316
drwxr-xr-x  2 root root   12288 Nov 19 13:33 .
drwxr-xr-x 47 root root    4096 Nov 19 13:33 ..
-rw-r--r--  1 root root    6746 Feb 26  2024 BCM4343A2.hcd
-rw-r--r--  1 root root   63806 Feb 26  2024 BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,3-model-a-plus.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,3-model-b-plus.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,4-compute-module.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,4-model-b.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,5-model-b.hcd -> BCM4345C0.hcd
-rw-r--r--  1 root root   55676 Feb 26  2024 BCM4345C5.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C5.raspberrypi,4-compute-module.hcd -> BCM4345C5.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C5.raspberrypi,400.hcd -> BCM4345C5.hcd
-rw-r--r--  1 root root   30049 Feb 26  2024 BCM43430A1.hcd
lrwxrwxrwx  1 root root      14 Feb 26  2024 BCM43430A1.raspberrypi,3-model-b.hcd -> BCM43430A1.hcd
lrwxrwxrwx  1 root root      27 Feb 26  2024 BCM43430A1.raspberrypi,model-zero-2-w.hcd -> ../synaptics/SYN43430A1.hcd
lrwxrwxrwx  1 root root      14 Feb 26  2024 BCM43430A1.raspberrypi,model-zero-w.hcd -> BCM43430A1.hcd
-rw-r--r--  1 root root   44376 Feb 26  2024 BCM43430B0.hcd
lrwxrwxrwx  1 root root      27 Feb 26  2024 BCM43430B0.raspberrypi,model-zero-2-w.hcd -> ../synaptics/SYN43430B0.hcd
lrwxrwxrwx  1 root root      17 Aug 10 06:08 BCM-0a5c-6410.hcd -> BCM-0bb4-0306.hcd
-rw-r--r--  1 root root   54734 Aug 10 06:08 BCM-0bb4-0306.hcd
-rw-r--r--  1 root root   96224 Aug 10 06:08 bcm43xx-0.fw
-rw-r--r--  1 root root     180 Aug 10 06:08 bcm43xx_hdr-0.fw
-rw-r--r--  1 root root  253748 Aug 10 06:08 brcmfmac4329-sdio.bin
-rw-r--r--  1 root root  223581 Aug 10 06:08 brcmfmac4330-sdio.bin
-rw-r--r--  1 root root    1509 Aug 10 06:08 brcmfmac4330-sdio.Prowise-PT301.txt
-rw-r--r--  1 root root  451566 Aug 10 06:08 brcmfmac4334-sdio.bin
-rw-r--r--  1 root root  569291 Aug 10 06:08 brcmfmac4335-sdio.bin
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4339-sdio.bin -> ../cypress/cyfmac4339-sdio.bin
-rw-r--r--  1 root root  623304 Aug 10 06:08 brcmfmac4350c2-pcie.bin
-rw-r--r--  1 root root  626140 Aug 10 06:08 brcmfmac4350-pcie.bin
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4354-sdio.bin -> ../cypress/cyfmac4354-sdio.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4354-sdio.clm_blob -> ../cypress/cyfmac4354-sdio.clm_blob
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4356-pcie.bin -> ../cypress/cyfmac4356-pcie.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4356-pcie.clm_blob -> ../cypress/cyfmac4356-pcie.clm_blob
-rw-r--r--  1 root root    2515 Aug 10 06:08 brcmfmac4356-pcie.gpd-win-pocket.txt
-rw-r--r--  1 root root    2610 Aug 10 06:08 brcmfmac4356-sdio.AP6356S.txt
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4356-sdio.bin -> ../cypress/cyfmac4356-sdio.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4356-sdio.clm_blob -> ../cypress/cyfmac4356-sdio.clm_blob
lrwxrwxrwx  1 root root      29 Aug 10 06:08 brcmfmac4356-sdio.firefly,firefly-rk3399.txt -> brcmfmac4356-sdio.AP6356S.txt
lrwxrwxrwx  1 root root      29 Aug 10 06:08 brcmfmac4356-sdio.khadas,vim2.txt -> brcmfmac4356-sdio.AP6356S.txt
lrwxrwxrwx  1 root root      29 Aug 10 06:08 brcmfmac4356-sdio.vamrs,rock960.txt -> brcmfmac4356-sdio.AP6356S.txt
-rw-r--r--  1 root root  634361 Aug 10 06:08 brcmfmac4358-pcie.bin
-rw-r--r--  1 root root 1105361 Aug 10 06:08 brcmfmac4366b-pcie.bin
-rw-r--r--  1 root root 1120971 Aug 10 06:08 brcmfmac4366c-pcie.bin
-rw-r--r--  1 root root  623448 Aug 10 06:08 brcmfmac4371-pcie.bin
-rw-r--r--  1 root root  479232 Aug 10 06:08 brcmfmac4373.bin
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4373-sdio.bin -> ../cypress/cyfmac4373-sdio.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4373-sdio.clm_blob -> ../cypress/cyfmac4373-sdio.clm_blob
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43012-sdio.bin -> ../cypress/cyfmac43012-sdio.bin
lrwxrwxrwx  1 root root      36 Aug 10 06:08 brcmfmac43012-sdio.clm_blob -> ../cypress/cyfmac43012-sdio.clm_blob
-rw-r--r--  1 root root  397312 Aug 10 06:08 brcmfmac43143.bin
-rw-r--r--  1 root root  385067 Aug 10 06:08 brcmfmac43143-sdio.bin
-rw-r--r--  1 root root  348160 Aug 10 06:08 brcmfmac43236b.bin
-rw-r--r--  1 root root  459988 Aug 10 06:08 brcmfmac43241b0-sdio.bin
-rw-r--r--  1 root root  403855 Aug 10 06:08 brcmfmac43241b4-sdio.bin
-rw-r--r--  1 root root  408682 Aug 10 06:08 brcmfmac43241b5-sdio.bin
-rw-r--r--  1 root root  479232 Aug 10 06:08 brcmfmac43242a.bin
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43340-sdio.bin -> ../cypress/cyfmac43340-sdio.bin
-rw-r--r--  1 root root    1693 Aug 10 06:08 brcmfmac43340-sdio.meegopad-t08.txt
-rw-r--r--  1 root root    1724 Aug 10 06:08 brcmfmac43340-sdio.pov-tab-p1006w-data.txt
-rw-r--r--  1 root root    1689 Aug 10 06:08 brcmfmac43340-sdio.predia-basic.txt
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43362-sdio.bin -> ../cypress/cyfmac43362-sdio.bin
-rw-r--r--  1 root root     803 Aug 10 06:08 brcmfmac43362-sdio.cubietech,cubietruck.txt
lrwxrwxrwx  1 root root      43 Aug 10 06:08 brcmfmac43362-sdio.lemaker,bananapro.txt -> brcmfmac43362-sdio.cubietech,cubietruck.txt
[... cut ...]
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43570-pcie.bin -> ../cypress/cyfmac43570-pcie.bin
lrwxrwxrwx  1 root root      36 Aug 10 06:08 brcmfmac43570-pcie.clm_blob -> ../cypress/cyfmac43570-pcie.clm_blob
-rw-r--r--  1 root root  595472 Aug 10 06:08 brcmfmac43602-pcie.ap.bin
-rw-r--r--  1 root root  635449 Aug 10 06:08 brcmfmac43602-pcie.bin
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac54591-pcie.bin -> ../cypress/cyfmac54591-pcie.bin
lrwxrwxrwx  1 root root      36 Aug 10 06:08 brcmfmac54591-pcie.clm_blob -> ../cypress/cyfmac54591-pcie.clm_blob

```

tim-moody, 2025-01-16:
@jvonau please review what other info should be captured. (this is raw OS, not IIAB)

jvonau, 2025-01-16:
Can you confirm raw disk write, using wired connection, and with no wifi present please.
`cat cmdline.txt` just to show what is passed to the kernel or the first line of `dmesg` and 
 grep for other related `brcmf` messages in dmesg.
`/usr/sbin/iw reg get` with just 'on the wire' is a good start. 
What logging is present? wpa_supplicant logging would be of interest as would be NetworkManager.

jvonau, 2025-01-16:
I see a bunch of symlinks point from specific raspi models to more generalized names to cut down on duplicate usage.

tim-moody, 2025-01-16:
>Can you confirm raw disk write, using wired connection, and with no wifi present please.
cat cmdline.txt just to show what is passed to the kernel or the first line of dmesg and
grep for other related brcmf messages in dmesg.
/usr/sbin/iw reg get with just 'on the wire' is a good start.
What logging is present? wpa_supplicant logging would be of interest as would be NetworkManager.

This is OS only, no IIAB. And it has now been overwritten with an IIAB install.

Still

```
root@box:~# cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=c1090cca-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US

```

```
root@box:~# dmesg |grep brcmf
[    5.095215] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.097123] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.102163] usbcore: registered new interface driver brcmfmac
[    5.322916] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    5.323361] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    6.114740] brcmfmac mmc1:0001:1 ap0: entered allmulticast mode
[    6.114820] brcmfmac mmc1:0001:1 ap0: entered promiscuous mode
[    6.890319] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

```

tim-moody, 2025-01-16:
>I see a bunch of symlinks point from specific raspi models to more generalized names to cut down on duplicate usage.

But note that the brcm bin and blob files mostly point at cypress.

jvonau, 2025-01-16:
> But note that the brcm bin and blob files mostly point at cypress.

Yup, iiab can exploit that to point them to iiab supplied firmware.

tim-moody, 2025-01-18:
Time to start fresh.

closed 2025-01-18 by tim-moody