iiab/iiab #3897: Rpi W2 Wi-Fi Firmware under raspios November 19th 2024

tim-moody, 2025-01-16:


tim-moody, 2025-01-16:
* Raspberry Pi: W2
* OS: 2024-11-19-raspios-bookworm-arm64-lite.img
* apt full upgrade: Yes
* IIAB Installed: No
* raspi-config nonint do_wifi_country: No

tim-moody, 2025-01-16:
```
dmesg | grep Firmware
[    0.076405] raspberrypi-firmware soc:firmware: Firmware hash is 2ae30f53898ae2f1ba77ff570a92991bedfb0398
[    9.355713] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7

cat /sys/firmware/devicetree/base/model
Raspberry Pi Zero 2 W Rev 1.0root@raspberrypi:~#
```

tim-moody, 2025-01-16:
```
root@raspberrypi:~# ll /lib/firmware/cypress/
total 6948
drwxr-xr-x  2 root root   4096 Nov 19 13:33 .
drwxr-xr-x 47 root root   4096 Nov 19 13:33 ..
-rw-r--r--  1 root root 574096 Aug 10 06:08 cyfmac4339-sdio.bin
-rw-r--r--  1 root root 601704 Aug 10 06:08 cyfmac4354-sdio.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac4354-sdio.clm_blob
-rw-r--r--  1 root root 616174 Aug 10 06:08 cyfmac4356-pcie.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac4356-pcie.clm_blob
-rw-r--r--  1 root root 580258 Aug 10 06:08 cyfmac4356-sdio.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac4356-sdio.clm_blob
-rw-r--r--  1 root root 563834 Aug 10 06:08 cyfmac4373-sdio.bin
-rw-r--r--  1 root root   4684 Aug 10 06:08 cyfmac4373-sdio.clm_blob
-rw-r--r--  1 root root 490686 Aug 10 06:08 cyfmac43012-sdio.bin
-rw-r--r--  1 root root   4692 Aug 10 06:08 cyfmac43012-sdio.clm_blob
-rw-r--r--  1 root root 409717 Aug 10 06:08 cyfmac43340-sdio.bin
-rw-r--r--  1 root root 202623 Aug 10 06:08 cyfmac43362-sdio.bin
-rw-r--r--  1 root root 399344 Aug 10 06:08 cyfmac43430-sdio.bin
-rw-r--r--  1 root root   4733 Aug 10 06:08 cyfmac43430-sdio.clm_blob
-rw-r--r--  1 root root 229520 Aug 10 06:08 cyfmac43439-sdio.bin
-rw-r--r--  1 root root   1438 Aug 10 06:08 cyfmac43439-sdio.txt
-rw-r--r--  1 root root   3208 Aug 10 06:08 cyfmac43439-sdio.clm_blob
lrwxrwxrwx  1 root root     38 Aug 10 06:08 cyfmac43455-sdio.bin -> /etc/alternatives/cyfmac43455-sdio.bin
-rw-r--r--  1 root root 548728 Aug 10 06:08 cyfmac43455-sdio-minimal.bin
-rw-r--r--  1 root root 609309 Aug 10 06:08 cyfmac43455-sdio-standard.bin
-rw-r--r--  1 root root   2676 Aug 10 06:08 cyfmac43455-sdio.clm_blob
-rw-r--r--  1 root root 558388 Aug 10 06:08 cyfmac43570-pcie.bin
-rw-r--r--  1 root root   5750 Aug 10 06:08 cyfmac43570-pcie.clm_blob
-rw-r--r--  1 root root 615882 Aug 10 06:08 cyfmac54591-pcie.bin
-rw-r--r--  1 root root   5764 Aug 10 06:08 cyfmac54591-pcie.clm_blob

```

tim-moody, 2025-01-16:
```
root@raspberrypi:~# ll /lib/firmware/brcm
total 13316
drwxr-xr-x  2 root root   12288 Nov 19 13:33 .
drwxr-xr-x 47 root root    4096 Nov 19 13:33 ..
-rw-r--r--  1 root root    6746 Feb 26  2024 BCM4343A2.hcd
-rw-r--r--  1 root root   63806 Feb 26  2024 BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,3-model-a-plus.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,3-model-b-plus.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,4-compute-module.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,4-model-b.hcd -> BCM4345C0.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C0.raspberrypi,5-model-b.hcd -> BCM4345C0.hcd
-rw-r--r--  1 root root   55676 Feb 26  2024 BCM4345C5.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C5.raspberrypi,4-compute-module.hcd -> BCM4345C5.hcd
lrwxrwxrwx  1 root root      13 Feb 26  2024 BCM4345C5.raspberrypi,400.hcd -> BCM4345C5.hcd
-rw-r--r--  1 root root   30049 Feb 26  2024 BCM43430A1.hcd
lrwxrwxrwx  1 root root      14 Feb 26  2024 BCM43430A1.raspberrypi,3-model-b.hcd -> BCM43430A1.hcd
lrwxrwxrwx  1 root root      27 Feb 26  2024 BCM43430A1.raspberrypi,model-zero-2-w.hcd -> ../synaptics/SYN43430A1.hcd
lrwxrwxrwx  1 root root      14 Feb 26  2024 BCM43430A1.raspberrypi,model-zero-w.hcd -> BCM43430A1.hcd
-rw-r--r--  1 root root   44376 Feb 26  2024 BCM43430B0.hcd
lrwxrwxrwx  1 root root      27 Feb 26  2024 BCM43430B0.raspberrypi,model-zero-2-w.hcd -> ../synaptics/SYN43430B0.hcd
lrwxrwxrwx  1 root root      17 Aug 10 06:08 BCM-0a5c-6410.hcd -> BCM-0bb4-0306.hcd
-rw-r--r--  1 root root   54734 Aug 10 06:08 BCM-0bb4-0306.hcd
-rw-r--r--  1 root root   96224 Aug 10 06:08 bcm43xx-0.fw
-rw-r--r--  1 root root     180 Aug 10 06:08 bcm43xx_hdr-0.fw
-rw-r--r--  1 root root  253748 Aug 10 06:08 brcmfmac4329-sdio.bin
-rw-r--r--  1 root root  223581 Aug 10 06:08 brcmfmac4330-sdio.bin
-rw-r--r--  1 root root    1509 Aug 10 06:08 brcmfmac4330-sdio.Prowise-PT301.txt
-rw-r--r--  1 root root  451566 Aug 10 06:08 brcmfmac4334-sdio.bin
-rw-r--r--  1 root root  569291 Aug 10 06:08 brcmfmac4335-sdio.bin
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4339-sdio.bin -> ../cypress/cyfmac4339-sdio.bin
-rw-r--r--  1 root root  623304 Aug 10 06:08 brcmfmac4350c2-pcie.bin
-rw-r--r--  1 root root  626140 Aug 10 06:08 brcmfmac4350-pcie.bin
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4354-sdio.bin -> ../cypress/cyfmac4354-sdio.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4354-sdio.clm_blob -> ../cypress/cyfmac4354-sdio.clm_blob
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4356-pcie.bin -> ../cypress/cyfmac4356-pcie.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4356-pcie.clm_blob -> ../cypress/cyfmac4356-pcie.clm_blob
-rw-r--r--  1 root root    2515 Aug 10 06:08 brcmfmac4356-pcie.gpd-win-pocket.txt
-rw-r--r--  1 root root    2610 Aug 10 06:08 brcmfmac4356-sdio.AP6356S.txt
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4356-sdio.bin -> ../cypress/cyfmac4356-sdio.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4356-sdio.clm_blob -> ../cypress/cyfmac4356-sdio.clm_blob
lrwxrwxrwx  1 root root      29 Aug 10 06:08 brcmfmac4356-sdio.firefly,firefly-rk3399.txt -> brcmfmac4356-sdio.AP6356S.txt
lrwxrwxrwx  1 root root      29 Aug 10 06:08 brcmfmac4356-sdio.khadas,vim2.txt -> brcmfmac4356-sdio.AP6356S.txt
lrwxrwxrwx  1 root root      29 Aug 10 06:08 brcmfmac4356-sdio.vamrs,rock960.txt -> brcmfmac4356-sdio.AP6356S.txt
-rw-r--r--  1 root root  634361 Aug 10 06:08 brcmfmac4358-pcie.bin
-rw-r--r--  1 root root 1105361 Aug 10 06:08 brcmfmac4366b-pcie.bin
-rw-r--r--  1 root root 1120971 Aug 10 06:08 brcmfmac4366c-pcie.bin
-rw-r--r--  1 root root  623448 Aug 10 06:08 brcmfmac4371-pcie.bin
-rw-r--r--  1 root root  479232 Aug 10 06:08 brcmfmac4373.bin
lrwxrwxrwx  1 root root      30 Aug 10 06:08 brcmfmac4373-sdio.bin -> ../cypress/cyfmac4373-sdio.bin
lrwxrwxrwx  1 root root      35 Aug 10 06:08 brcmfmac4373-sdio.clm_blob -> ../cypress/cyfmac4373-sdio.clm_blob
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43012-sdio.bin -> ../cypress/cyfmac43012-sdio.bin
lrwxrwxrwx  1 root root      36 Aug 10 06:08 brcmfmac43012-sdio.clm_blob -> ../cypress/cyfmac43012-sdio.clm_blob
-rw-r--r--  1 root root  397312 Aug 10 06:08 brcmfmac43143.bin
-rw-r--r--  1 root root  385067 Aug 10 06:08 brcmfmac43143-sdio.bin
-rw-r--r--  1 root root  348160 Aug 10 06:08 brcmfmac43236b.bin
-rw-r--r--  1 root root  459988 Aug 10 06:08 brcmfmac43241b0-sdio.bin
-rw-r--r--  1 root root  403855 Aug 10 06:08 brcmfmac43241b4-sdio.bin
-rw-r--r--  1 root root  408682 Aug 10 06:08 brcmfmac43241b5-sdio.bin
-rw-r--r--  1 root root  479232 Aug 10 06:08 brcmfmac43242a.bin
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43340-sdio.bin -> ../cypress/cyfmac43340-sdio.bin
-rw-r--r--  1 root root    1693 Aug 10 06:08 brcmfmac43340-sdio.meegopad-t08.txt
-rw-r--r--  1 root root    1724 Aug 10 06:08 brcmfmac43340-sdio.pov-tab-p1006w-data.txt
-rw-r--r--  1 root root    1689 Aug 10 06:08 brcmfmac43340-sdio.predia-basic.txt
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43362-sdio.bin -> ../cypress/cyfmac43362-sdio.bin
-rw-r--r--  1 root root     803 Aug 10 06:08 brcmfmac43362-sdio.cubietech,cubietruck.txt
lrwxrwxrwx  1 root root      43 Aug 10 06:08 brcmfmac43362-sdio.lemaker,bananapro.txt -> brcmfmac43362-sdio.cubietech,cubietruck.txt
[... cut ...]
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac43570-pcie.bin -> ../cypress/cyfmac43570-pcie.bin
lrwxrwxrwx  1 root root      36 Aug 10 06:08 brcmfmac43570-pcie.clm_blob -> ../cypress/cyfmac43570-pcie.clm_blob
-rw-r--r--  1 root root  595472 Aug 10 06:08 brcmfmac43602-pcie.ap.bin
-rw-r--r--  1 root root  635449 Aug 10 06:08 brcmfmac43602-pcie.bin
lrwxrwxrwx  1 root root      31 Aug 10 06:08 brcmfmac54591-pcie.bin -> ../cypress/cyfmac54591-pcie.bin
lrwxrwxrwx  1 root root      36 Aug 10 06:08 brcmfmac54591-pcie.clm_blob -> ../cypress/cyfmac54591-pcie.clm_blob

```

tim-moody, 2025-01-16:
>Can you confirm raw disk write, using wired connection, and with no wifi present please.
cat cmdline.txt just to show what is passed to the kernel or the first line of dmesg and
grep for other related brcmf messages in dmesg.
/usr/sbin/iw reg get with just 'on the wire' is a good start.
What logging is present? wpa_supplicant logging would be of interest as would be NetworkManager.

This has not yet been overwritten, but it is OS only and only uses wired connection.

I don't know where to look for raw disk write (? to the device, not the file system)

```
root@raspberrypi:~# cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=400f5002-02 rootfstype=ext4 fsck.repair=yes rootwait

root@raspberrypi:~# dmesg |grep brcmf
[    9.025056] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    9.030417] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    9.031577] usbcore: registered new interface driver brcmfmac
[    9.355169] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    9.355713] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7

```

```
root@raspberrypi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:cf:67:8b:d0:dd brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 3c:8c:f8:f6:4c:07 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.76/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 82178sec preferred_lft 82178sec
    inet6 fe80::d76f:4323:9c17:a336/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
```

```
root@raspberrypi:~# /usr/sbin/iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)
```

```
root@raspberrypi:~# journalctl -b |grep wpa_supplicant
Jan 16 20:18:56 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 16 20:18:57 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 16 20:18:57 raspberrypi wpa_supplicant[421]: Successfully initialized wpa_supplicant

```

tim-moody, 2025-01-16:
```
root@raspberrypi:~# journalctl -b |grep NetworkManager
Jan 16 20:18:56 raspberrypi systemd[1]: Starting NetworkManager.service - Network Manager...
Jan 16 20:18:57 raspberrypi NetworkManager[416]: <info>  [1737058737.9515] NetworkManager (version 1.42.4) is starting... (boot:be3dbb75-014b-4568-86a3-7d04ee96c01a)
Jan 16 20:18:57 raspberrypi NetworkManager[416]: <info>  [1737058737.9518] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf)
Jan 16 20:18:57 raspberrypi systemd[1]: Started NetworkManager.service - Network Manager.
Jan 16 20:18:57 raspberrypi NetworkManager[416]: <info>  [1737058737.9685] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Jan 16 20:18:57 raspberrypi systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0387] manager[0x557e62a000]: monitoring kernel firmware directory '/lib/firmware'.
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0389] monitoring ifupdown state file '/run/network/ifstate'.
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0747] hostname: hostname: using hostnamed
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0748] hostname: static hostname changed from (none) to "raspberrypi"
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0793] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0867] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0881] manager[0x557e62a000]: rfkill: Wi-Fi hardware radio set disabled
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.0883] manager[0x557e62a000]: rfkill: WWAN hardware radio set enabled
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.3979] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.4850] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.4938] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5223] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5582] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5600] manager: rfkill: Wi-Fi disabled by radio killswitch; disabled by state file
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5604] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5609] manager: Networking is enabled by state file
Jan 16 20:18:58 raspberrypi dbus-daemon[346]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=416 comm="/usr/sbin/NetworkManager --no-daemon")
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5696] settings: Loaded settings plugin: ifupdown ("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5698] settings: Loaded settings plugin: keyfile (internal)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5699] ifupdown: management mode: unmanaged
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5721] ifupdown: interfaces file /etc/network/interfaces doesn't exist
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5886] dhcp: init: Using DHCP client 'internal'
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.5903] manager: (lo): new Loopback device (/org/freedesktop/NetworkManager/Devices/1)
Jan 16 20:18:58 raspberrypi systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.6218] device (lo): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.6307] device (lo): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.6411] device (lo): Activation: starting connection 'lo' (061c4670-deee-4896-b3a8-25914d311a42)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.6531] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
Jan 16 20:18:58 raspberrypi NetworkManager[416]: <info>  [1737058738.6768] settings: (eth0): created default wired connection 'Wired connection 1'
[... cut ...]
Jan 16 20:19:00 raspberrypi NetworkManager[416]: <info>  [1737058740.9236] device (eth0): Activation: successful, device activated.
Jan 16 20:19:00 raspberrypi NetworkManager[416]: <info>  [1737058740.9259] manager: NetworkManager state is now CONNECTED_GLOBAL
Jan 16 20:19:00 raspberrypi NetworkManager[416]: <info>  [1737058740.9279] manager: startup complete
Jan 16 20:19:00 raspberrypi systemd[1]: Finished NetworkManager-wait-online.service - Network Manager Wait Online.
Jan 16 20:19:11 raspberrypi systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.

```

jvonau, 2025-01-17:
brcmfmac43455-sdio.txt 
brcmfmac43436-sdio.txt
brcmfmac43430-sdio.txt

So what are in these files?

jvonau, 2025-01-17:
>I don't know where to look for raw disk write (? to the device, not the file system)

With "raw disk write" I meant using dd and no other changes, WiFi should be still blocked by rfkill, 
https://github.com/iiab/iiab/pull/3865#issuecomment-2555004727
> rfkill unblock wifi
> for filename in /var/lib/systemd/rfkill/*:wlan ; do
>        echo 0 > $filename

Run the rfkill command as above, check the status of those files and adjust as needed. Might need to reboot.

holta, 2025-01-17:
> brcmfmac43436-sdio.txt

https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/brcm/brcmfmac43436-sdio.txt

Background/summary on 43436:

- Zero 2 W shipped with 1 of 2 different 43436 chips, either one from Synaptics, or one from Infineon.
- Both are basically clones of the 43430, as included in Zero W and RPi 3 (whereas 3 B+ uses 43455).
- Details: https://github.com/raspberrypi/linux/issues/4723

jvonau, 2025-01-17:
https://github.com/RPi-Distro/firmware-nonfree/tree/bookworm/debian/config/brcm80211/brcm
The other two are more important, looks like some config setting that might be loaded on demand when the firmware loads or maybe just notes.

jvonau, 2025-01-17:
`journalctl -b |grep wpa_supplicant` 
`journalctl -b |grep NetworkManager`

Would be useful in the dumps

tim-moody, 2025-01-17:
@holta how does 2 W in 64 bit mode use a driver from W? Wasn't 43430 32 bit?

tim-moody, 2025-01-17:
Still no IIAB, but has been rebooted several times (but was copied with dd and no other changes except apt update/upgrade)

```
root@raspberrypi:~# rfkill
ID TYPE      DEVICE      SOFT      HARD
 0 bluetooth hci0   unblocked unblocked
 1 wlan      phy0     blocked unblocked

```

tim-moody, 2025-01-17:
```
root@raspberrypi:~# journalctl -b |grep wpa_supplicant
Jan 16 23:09:23 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 16 23:09:23 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 16 23:09:23 raspberrypi wpa_supplicant[401]: Successfully initialized wpa_supplicant

root@raspberrypi:~# journalctl -b |grep wpa_supplicant
Jan 16 23:09:23 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 16 23:09:23 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 16 23:09:23 raspberrypi wpa_supplicant[401]: Successfully initialized wpa_supplicant
root@raspberrypi:~# journalctl -b |grep NetworkManager
Jan 16 23:09:23 raspberrypi systemd[1]: Starting NetworkManager.service - Network Manager...
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.7290] NetworkManager (version 1.42.4) is starting... (boot:08fcead6-8ff4-4331-ba4a-98f928f5cfcd)
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.7292] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf)
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.7541] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Jan 16 23:09:24 raspberrypi systemd[1]: Started NetworkManager.service - Network Manager.
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.8037] manager[0x558f0f9000]: monitoring kernel firmware directory '/lib/firmware'.
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.8039] monitoring ifupdown state file '/run/network/ifstate'.
Jan 16 23:09:24 raspberrypi systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.9148] hostname: hostname: using hostnamed
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.9150] hostname: static hostname changed from (none) to "raspberrypi"
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.9171] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.9241] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.9252] manager[0x558f0f9000]: rfkill: Wi-Fi hardware radio set disabled
Jan 16 23:09:24 raspberrypi NetworkManager[400]: <info>  [1737068964.9253] manager[0x558f0f9000]: rfkill: WWAN hardware radio set enabled
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.1265] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2051] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2131] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2436] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2670] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2683] manager: rfkill: Wi-Fi disabled by radio killswitch; disabled by state file
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2687] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2690] manager: Networking is enabled by state file
Jan 16 23:09:25 raspberrypi dbus-daemon[343]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.8' (uid=0 pid=400 comm="/usr/sbin/NetworkManager --no-daemon")
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2762] settings: Loaded settings plugin: ifupdown ("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2766] settings: Loaded settings plugin: keyfile (internal)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2767] ifupdown: management mode: unmanaged
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2801] ifupdown: interfaces file /etc/network/interfaces doesn't exist
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.2994] dhcp: init: Using DHCP client 'internal'
Jan 16 23:09:25 raspberrypi systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.3093] manager: (lo): new Loopback device (/org/freedesktop/NetworkManager/Devices/1)
Jan 16 23:09:25 raspberrypi NetworkManager[400]: <info>  [1737068965.3510] device (lo): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
[... cut ...]
Jan 16 23:09:27 raspberrypi NetworkManager[400]: <info>  [1737068967.6402] device (eth0): Activation: successful, device activated.
Jan 16 23:09:27 raspberrypi NetworkManager[400]: <info>  [1737068967.6425] manager: NetworkManager state is now CONNECTED_GLOBAL
Jan 16 23:09:27 raspberrypi NetworkManager[400]: <info>  [1737068967.6445] manager: startup complete
Jan 16 23:09:27 raspberrypi systemd[1]: Finished NetworkManager-wait-online.service - Network Manager Wait Online.
Jan 16 23:09:37 raspberrypi systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.

```

holta, 2025-01-17:
> how does 2 W in 64 bit mode use a driver from W? Wasn't 43430 32 bit?

True that RasPiOS focus was almost entirely 32-bits until ~5 years ago.  I don't know for sure.  But so far it seems very plausible (?) that RPi WiFi chip (and its WiFi firmware blobs, supplied by RasPiOS or Ubuntu-on-RPi etc) is entirely unaffected by the 32-bit &/or 64-bit personality of the OS?

tim-moody, 2025-01-17:
```
lrwxrwxrwx 1 root root 27 Feb 26  2024 /lib/firmware/brcm/BCM43430A1.raspberrypi,model-zero-2-w.hcd -> ../synaptics/SYN43430A1.hcd
lrwxrwxrwx 1 root root 27 Feb 26  2024 /lib/firmware/brcm/BCM43430B0.raspberrypi,model-zero-2-w.hcd -> ../synaptics/SYN43430B0.hcd
lrwxrwxrwx 1 root root 22 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin
lrwxrwxrwx 1 root root 22 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436-sdio.txt
lrwxrwxrwx 1 root root 27 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.clm_blob -> brcmfmac43436-sdio.clm_blob
lrwxrwxrwx 1 root root 23 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx 1 root root 23 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
lrwxrwxrwx 1 root root 23 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
lrwxrwxrwx 1 root root 23 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436s-sdio.txt
lrwxrwxrwx 1 root root 22 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin
lrwxrwxrwx 1 root root 22 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt -> brcmfmac43436-sdio.txt
lrwxrwxrwx 1 root root 27 Aug 10 06:08 /lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.clm_blob -> brcmfmac43436-sdio.clm_blob

```

tim-moody, 2025-01-18:
>rfkill unblock wifi
for filename in /var/lib/systemd/rfkill/*:wlan ; do
echo 0 > $filename

I'm not clear on the purpose for `echo 0` to each wlan file as I believe the rfkill command will write 0 to these files.

jvonau, 2025-01-18:
>Run the rfkill command as above, check the status of those files and adjust as needed. Might need to reboot.

>I'm not clear on the purpose for echo 0 to each wlan file as I believe the rfkill command will write 0 to these files.

You found the code in upstream's usage, I asked to check the files present to ensure the behavior is to write as you describe. 

https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L863-L915 is the upstream process of raspi-config

tim-moody, 2025-01-18:
OK

```
root@raspberrypi:~# rfkill
ID TYPE      DEVICE      SOFT      HARD
 0 bluetooth hci0   unblocked unblocked
 1 wlan      phy0     blocked unblocked

root@raspberrypi:~# ll /var/lib/systemd/rfkill/*:wlan
-rw-r--r-- 1 root root 2 Nov 19 13:39 /var/lib/systemd/rfkill/platform-3f300000.mmcnr:wlan

root@raspberrypi:~# cat /var/lib/systemd/rfkill/platform-3f300000.mmcnr:wlan
1

root@raspberrypi:~# rfkill unblock wifi
root@raspberrypi:~# ll /var/lib/systemd/rfkill/*:wlan
-rw-r--r-- 1 root root 2 Nov 19 13:39 /var/lib/systemd/rfkill/platform-3f300000.mmcnr:wlan

root@raspberrypi:~# cat /var/lib/systemd/rfkill/platform-3f300000.mmcnr:wlan
0
```

I don't think the write of 0 to the file is necessary if rfkill unblock is available.

jvonau, 2025-01-18:
Ok from the upstream link above think you need `nmcli radio wifi on` and check for WirelessEnabled=true/' /var/lib/NetworkManager/NetworkManager.state if it is present. 
Now connect up to WiFi with `nmcli`'s routine and note the NM logging and iw get status.

tim-moody, 2025-01-18:
Test for router country code without IIAB

```
root@raspberrypi:~# ll /etc/NetworkManager/system-connections/
total 8
drwxr-xr-x 2 root root 4096 Jul 24  2023 .
drwxr-xr-x 7 root root 4096 Nov 19 13:33 ..

raspi-config nonint do_wifi_country DE

root@raspberrypi:~# ll /etc/NetworkManager/system-connections/
total 8
drwxr-xr-x 2 root root 4096 Jul 24  2023 .
drwxr-xr-x 7 root root 4096 Nov 19 13:33 ..

root@raspberrypi:~# cat /var/lib/NetworkManager/NetworkManager.state
[main]
NetworkingEnabled=true
WirelessEnabled=true
WWANEnabled=true

root@raspberrypi:~# nmcli device wifi connect SunRoom password xxxxx
Device 'wlan0' successfully activated with '8b1ca8ac-3e33-465f-9a6d-b5d47f662b57'.

root@raspberrypi:~# ll /etc/NetworkManager/system-connections/
total 12
drwxr-xr-x 2 root root 4096 Jan 18 16:07 .
drwxr-xr-x 7 root root 4096 Nov 19 13:33 ..
-rw------- 1 root root  270 Jan 18 16:07 SunRoom.nmconnection

root@raspberrypi:~# journalctl -b |grep NetworkManager
Jan 18 15:00:57 raspberrypi systemd[1]: Starting NetworkManager.service - Network Manager...
Jan 18 15:00:58 raspberrypi NetworkManager[418]: <info>  [1737212458.9276] NetworkManager (version 1.42.4) is starting... (boot:79e803cd-b001-4f1a-9333-884ced2961e2)
Jan 18 15:00:58 raspberrypi NetworkManager[418]: <info>  [1737212458.9278] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf)
Jan 18 15:00:58 raspberrypi systemd[1]: Started NetworkManager.service - Network Manager.
Jan 18 15:00:58 raspberrypi NetworkManager[418]: <info>  [1737212458.9430] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Jan 18 15:00:58 raspberrypi systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Jan 18 15:00:58 raspberrypi NetworkManager[418]: <info>  [1737212458.9951] manager[0x559ca3e000]: monitoring kernel firmware directory '/lib/firmware'.
Jan 18 15:00:58 raspberrypi NetworkManager[418]: <info>  [1737212458.9953] monitoring ifupdown state file '/run/network/ifstate'.
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.0213] hostname: hostname: using hostnamed
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.0214] hostname: static hostname changed from (none) to "raspberrypi"
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.0269] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.0345] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.0358] manager[0x559ca3e000]: rfkill: Wi-Fi hardware radio set disabled
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.0361] manager[0x559ca3e000]: rfkill: WWAN hardware radio set enabled
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.2868] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.3709] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.3792] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4094] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4451] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4467] manager: rfkill: Wi-Fi disabled by radio killswitch; disabled by state file
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4474] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4480] manager: Networking is enabled by state file
Jan 18 15:00:59 raspberrypi dbus-daemon[347]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=418 comm="/usr/sbin/NetworkManager --no-daemon")
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4576] settings: Loaded settings plugin: ifupdown ("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4578] settings: Loaded settings plugin: keyfile (internal)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4579] ifupdown: management mode: unmanaged
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4590] ifupdown: interfaces file /etc/network/interfaces doesn't exist
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4852] dhcp: init: Using DHCP client 'internal'
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.4871] manager: (lo): new Loopback device (/org/freedesktop/NetworkManager/Devices/1)
[... cut ...]
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)
```

tim-moody, 2025-01-18:
Cut down the noise

```
root@raspberrypi:~# journalctl -b |grep wlan0
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.5897] device (wlan0): driver supports Access Point (AP) mode
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.5975] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/3)
Jan 18 15:00:59 raspberrypi NetworkManager[418]: <info>  [1737212459.5992] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Jan 18 16:06:23 raspberrypi NetworkManager[418]: <info>  [1737216383.1641] device (wlan0): supplicant interface state: internal-starting -> disconnected
Jan 18 16:06:23 raspberrypi NetworkManager[418]: <info>  [1737216383.1645] Wi-Fi P2P device controlled by interface wlan0 created
Jan 18 16:06:23 raspberrypi NetworkManager[418]: <info>  [1737216383.1659] manager: (p2p-dev-wlan0): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/4)
Jan 18 16:06:23 raspberrypi NetworkManager[418]: <info>  [1737216383.1670] device (p2p-dev-wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Jan 18 16:06:23 raspberrypi NetworkManager[418]: <info>  [1737216383.1691] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Jan 18 16:06:23 raspberrypi NetworkManager[418]: <info>  [1737216383.1715] device (p2p-dev-wlan0): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9406] device (wlan0): Activation: starting connection 'SunRoom' (8b1ca8ac-3e33-465f-9a6d-b5d47f662b57)
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9428] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9449] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9464] device (wlan0): Activation: (wifi) access point 'SunRoom' has security, but secrets are required.
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9466] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9475] sup-iface[b7ea40a425e4ed23,0,wlan0]: wps: type pbc start...
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9540] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9559] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:58 raspberrypi NetworkManager[418]: <info>  [1737216478.9575] device (wlan0): Activation: (wifi) connection 'SunRoom' has security, and secrets exist.  No new secrets needed.
Jan 18 16:07:58 raspberrypi wpa_supplicant[425]: wlan0: WPS-CANCEL
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.0446] device (wlan0): supplicant interface state: disconnected -> scanning
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.0447] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Jan 18 16:07:59 raspberrypi wpa_supplicant[425]: wlan0: Trying to associate with SSID 'SunRoom'
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.6647] device (wlan0): supplicant interface state: scanning -> associating
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.6648] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Jan 18 16:07:59 raspberrypi wpa_supplicant[425]: wlan0: Associated with c0:4a:00:e5:ec:04
Jan 18 16:07:59 raspberrypi wpa_supplicant[425]: wlan0: CTRL-EVENT-CONNECTED - Connection to c0:4a:00:e5:ec:04 completed [id=0 id_str=]
Jan 18 16:07:59 raspberrypi wpa_supplicant[425]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.8029] device (wlan0): supplicant interface state: associating -> completed
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.8030] device (wlan0): Activation: (wifi) Stage 2 of 5 (Device Configure) successful. Connected to wireless network "SunRoom"
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.8031] device (p2p-dev-wlan0): supplicant management interface state: associating -> completed
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.8037] device (wlan0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Jan 18 16:07:59 raspberrypi NetworkManager[418]: <info>  [1737216479.8052] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Jan 18 16:07:59 raspberrypi avahi-daemon[344]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::aa1e:766a:5dc5:cc20.
Jan 18 16:07:59 raspberrypi avahi-daemon[344]: New relevant interface wlan0.IPv6 for mDNS.
Jan 18 16:07:59 raspberrypi avahi-daemon[344]: Registering new address record for fe80::aa1e:766a:5dc5:cc20 on wlan0.*.
Jan 18 16:08:03 raspberrypi NetworkManager[418]: <info>  [1737216483.1714] dhcp4 (wlan0): state changed new lease, address=192.168.3.94
[... cut ...]
Jan 18 16:08:03 raspberrypi NetworkManager[418]: <info>  [1737216483.2077] device (wlan0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
Jan 18 16:08:03 raspberrypi NetworkManager[418]: <info>  [1737216483.2864] device (wlan0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Jan 18 16:08:03 raspberrypi NetworkManager[418]: <info>  [1737216483.2876] device (wlan0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Jan 18 16:08:03 raspberrypi NetworkManager[418]: <info>  [1737216483.3004] device (wlan0): Activation: successful, device activated.

```

holta, 2025-01-18:
@tim-moody: maybe add some of your above commands to #3895 (Rpi Wi-Fi Tools and Commands) if they can help others?

tim-moody, 2025-01-18:
- raspi-config nonint do_wifi_country HT - no complaints
- ssh in over wifi
- disconnected ethernet cable - Now connected only via wifi

root@raspberrypi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:8b:d0:dd brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.94/24 brd 192.168.3.255 scope global dynamic noprefixroute wlan0
       valid_lft 85227sec preferred_lft 85227sec
    inet6 fe80::aa1e:766a:5dc5:cc20/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 3c:8c:f8:f6:4c:07 brd ff:ff:ff:ff:ff:ff

- N.B. this is with reg code HT

tim-moody, 2025-01-18:
I would like to now install hostapd hotspot without IIAB to test max connections. Does this seem like a viable cookbook?

https://trainedmonkey.com/2024/05/20/using_a_raspberry_pi_as_a_wifi_hotspot

jvonau, 2025-01-18:
`nmcli d` would be a useful data point as I'm seeing "p2p-dev-wlan0" in the logging. I don't think being ssh'd on the wifi in is going to allow you to disconnect wlan0 and continue to play around. Test as you wish, I might try to decrypt what is going on.

tim-moody, 2025-01-18:
```
root@raspberrypi:~# nmcli d
DEVICE         TYPE      STATE                   CONNECTION
eth0           ethernet  connected               Wired connection 1
lo             loopback  connected (externally)  lo
wlan0          wifi      connected               SunRoom
p2p-dev-wlan0  wifi-p2p  disconnected            --
```

back on wired connection

jvonau, 2025-01-18:
So much for doing step by step analyzes of NM can do when the country code is not set in advance. Running `raspi-config nonint do_wifi_country HT` has muddied the environment to be able to capture what NM does when not preset, be able to demonstrate that NM can set the country code when not already set in an effort to explain why CA is showing up for nzola.

jvonau, 2025-01-18:
Whatever, I'd edit cmdline.txt ensuring there is no country code present and reboot to get back to the 00 country code as noted above. If the country code is not 00 then the assumption is NM set the value on the fly.

tim-moody, 2025-01-18:
Time to start fresh.

closed 2025-01-18 by tim-moody

jvonau, 2025-01-18:
> Time to start fresh.

Not really:
https://github.com/iiab/iiab/issues/3868#issuecomment-2595350758
https://github.com/iiab/iiab/issues/3868#issuecomment-2596100972
NM interacting with country code is highly dependent upon the AP being connected to.