iiab/iiab #3898: Install IIAB with Firmware OS on above Rpi 5

tim-moody, 2025-01-16:


tim-moody, 2025-01-16:
* Raspberry Pi: 5
* OS: 2024-11-19-raspios-bookworm-arm64-lite.img
* apt full upgrade: Yes
* IIAB Installed: Yes with Wi-Fi Firmware: OS
* raspi-config nonint do_wifi_country: No

tim-moody, 2025-01-16:
Immediately after install

https://paste.centos.org/view/7ac2a3be

After reboot

https://paste.centos.org/view/b5fd51f0

tim-moody, 2025-01-16:
```
root@box:~# cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=c1090cca-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US
root@box:~# dmesg |grep brcmf
[    5.095215] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.097123] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.102163] usbcore: registered new interface driver brcmfmac
[    5.322916] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    5.323361] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    6.114740] brcmfmac mmc1:0001:1 ap0: entered allmulticast mode
[    6.114820] brcmfmac mmc1:0001:1 ap0: entered promiscuous mode
[    6.890319] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

```

jvonau, 2025-01-16:
See that cfg80211.ieee80211_regdom=US was propagated to `iw reg get` but iiab acted on default_vars as the testing for country_code is dependent upon having a wifi gateway present.

jvonau, 2025-01-16:
Quick test just edit hostapd.conf changing to CA and reboot.. Want to look at order of authority over the value applied to iw reg. Should follow hostapd's over to CA when rebooted or we have rejected country code present? If it remains US then the cmdline has the greater power over the setting.

tim-moody, 2025-01-16:
root@box:~# /usr/sbin/iw reg get
global
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

tim-moody, 2025-01-16:
But why does cmdline.txt not have regdom=US for the W2?

jvonau, 2025-01-16:
Nice, come up an upstream AP that has different country from what is used within hostapd locally.   
Give admin-console at test by supplying upstream wifi info. Lets see where 'reg' ends up.

tim-moody, 2025-01-16:
looking at /boot on raspios prior to firstboot:

```
root@iiab-devel:/Files/Images# ll /mnt/boot/cmdline.txt
-rwxr-xr-x 1 root root 154 Nov 19 13:39 /mnt/boot/cmdline.txt*
root@iiab-devel:/Files/Images# cat  /mnt/boot/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=8a438930-02 rootfstype=ext4 fsck.repair=yes rootwait quiet init=/usr/lib/raspberrypi-sys-mods/firstboot

```

jvonau, 2025-01-16:
> But why does cmdline.txt not have regdom=US for the W2?

No clue, that is an upstream question, seems like a bug but might be the source of the swapping issue. Is the newest rpi-imager being used?

tim-moody, 2025-01-16:
no. I write with dd after I download and unxz the file

tim-moody, 2025-01-16:
After connecting over wifi to local router

root@box:/opt/iiab/iiab-admin-console# /usr/sbin/iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

tim-moody, 2025-01-16:
root@box:/opt/iiab/iiab-admin-console# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.60/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 84768sec preferred_lft 84768sec
    inet6 fe80::2205:d4ce:ba05:2ec8/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:cb brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.62/24 brd 192.168.3.255 scope global dynamic noprefixroute wlan0
       valid_lft 86312sec preferred_lft 86312sec
    inet6 fe80::36d:8dd8:741e:1e92/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 26:bf:71:3c:0f:6d brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.10/24 brd 10.10.10.255 scope global br0
       valid_lft forever preferred_lft forever
5: ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 02:a6:23:5c:3c:bb brd ff:ff:ff:ff:ff:ff permaddr 2e:cf:67:aa:aa:cb

jvonau, 2025-01-16:
Just confirming hostapd.conf still has CA? Might of been rewritten when the wifi was connected to get the channel right.

tim-moody, 2025-01-16:
root@box:~# cat /etc/hostapd/hostapd.conf
# Basic configuration

interface=ap0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=Internet in a Box
channel=6
bridge=br0

# Hardware configuration
driver=nl80211
country_code=CA
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g

tim-moody, 2025-01-16:
Think something went wrong, as channel should be 11. Also it did not shut down cleanly.

tim-moody, 2025-01-16:
Think most of this happens in firstboot, but I haven't found where yet.

jvonau, 2025-01-16:
`sudo journalctl -u networkd-dispatcher` lets have a look at the channel changing scripts.

jvonau, 2025-01-16:
Just edit hostapd.conf correcting channel for now and reboot, I'll try to figure that out when I can review it.

tim-moody, 2025-01-16:
Think there is a problem with the router I chose, so tried a different one:

root@box:~# /usr/sbin/iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

```
root@box:~# cat /etc/hostapd/hostapd.conf
# Basic configuration

interface=ap0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=Internet in a Box
channel=3
bridge=br0

# Hardware configuration
driver=nl80211
country_code=US
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g
```

```
root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.60/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 86218sec preferred_lft 86218sec
    inet6 fe80::2205:d4ce:ba05:2ec8/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:cb brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.62/24 brd 192.168.3.255 scope global dynamic noprefixroute wlan0
       valid_lft 86263sec preferred_lft 86263sec
    inet6 fe80::909d:d40a:ed3d:4a14/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 26:bf:71:3c:0f:6d brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.10/24 brd 10.10.10.255 scope global br0
       valid_lft forever preferred_lft forever
5: ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 02:a6:23:5c:3c:bb brd ff:ff:ff:ff:ff:ff permaddr 2e:cf:67:aa:aa:cb
```

jvonau, 2025-01-17:
Now with cmdline being US and the CA demo above, can you ensure hostapd.conf(.iiab) both have CA and reboot please. The channel changing on the fly might of been US as that is what was written to both files when installed.
What is the upstream AP point, can the country be set and if so what is the setting?

tim-moody, 2025-01-18:
>What is the upstream AP point, can the country be set and if so what is the setting?

I couldn't find any place to change this on my TPLink router. I found a posting about another TPLink that says it is hard coded and can't be changed. (set to DE for the poster)

```
root@box:~# journalctl |grep ountry
Jan 16 21:00:31 box ansible-ini_file[22659]: Invoked with dest=/etc/iiab/iiab.ini section=detected_network option=host_country_code_found value=US path=/etc/iiab/iiab.ini backup=False state=present exclusive=True no_extra_spaces=False ignore_spaces=False allow_no_value=False modify_inactive_option=True create=True follow=False unsafe_writes=False section_has_values=None values=None mode=None owner=None group=None seuser=None serole=None selevel=None setype=None attributes=None
Jan 16 21:00:39 box ansible-ansible.legacy.command[23396]: Invoked with _raw_params=raspi-config nonint do_wifi_country US _uses_shell=False expand_argument_vars=True stdin_add_newline=True strip_empty_ends=True argv=None chdir=None executable=None creates=None removes=None stdin=None
Jan 16 21:01:11 box ansible-ini_file[25142]: Invoked with dest=/etc/iiab/iiab.ini section=network option=host_country_code_applied value=US path=/etc/iiab/iiab.ini backup=False state=present exclusive=True no_extra_spaces=False ignore_spaces=False allow_no_value=False modify_inactive_option=True create=True follow=False unsafe_writes=False section_has_values=None values=None mode=None owner=None group=None seuser=None serole=None selevel=None setype=None attributes=None
root@box:~#

```

Any suggestions on finding the router country code?

jvonau, 2025-01-18:
Take the 2 variables under indirect user control, disable hostapd and remove the country code from cmdline.txt. Should be left with only NetworkManager and the AP in the picture. If the AP has the beacon set to a country code then NetworkManager should set `iw reg` to that code and if 00 is displayed I would suspect the AP doesn't do country code. The NM logging should capture the country code exchange.