iiab/iiab #3900: Copy of Wifi Regulatory Database

tim-moody, 2025-01-17:
https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git (2024-10-07)

[db.txt](https://github.com/user-attachments/files/18459730/db.txt)

holta, 2025-01-18:
Similar view (though not quite as complete) here: http://kernel.org/pub/software/network/wireless-regdb/

- Anybody how to make `/usr/lib/firmware/regulatory.db` human-readable?
- It's used to be so easy (using command `regdbdump`) but that seems to have disappeared very recently, in the past couple years.

```
root@box:~# file /usr/lib/firmware/regulatory.db
/usr/lib/firmware/regulatory.db: CRDA wireless regulatory database file
```

ASIDE: How long until IIAB is used on the moon?  FT.com says "commercial radio spectrum filings shows bet on future lunar economy" https://archive.ph/zzU57 (2025-01-09).  More immediately, IIAB is no doubt already used in international waters, where the UN and [others](https://www.pfetten.com/logistiklexikon/locode/locode-xz.html) use "country code" XZ (e.g. for "installations" etc).  Even if that's not yet taken seriously by the cruise ship industry, CRDA (Central Regulatory Domain Agent), etc.  No doubt because the ITU (which generally governs RF spectrum in international waters) is only really concerned with long-range RF like Marine VHF Radio, AIS (automatic identification system), not to mention military frequencies.

tim-moody, 2025-01-18:
>Anybody how to make /usr/lib/firmware/regulatory.db human-readable?

The link is literally at the top of the ticket, db.txt is human readable.

holta, 2025-01-18:
I fully understand that db.txt is human readable.

The question is how to make amy specific `/etc/lib/firmware/regulatory.db` human-readable, as used to be a walk in the park.

holta, 2025-01-18:
In any case, rather than programmatically probing `/usr/lib/firmware/regulatory.db` using `iw reg set XX` to poll for all possible [26*26 = 676 country codes](https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2), I'm personally hoping for something like `regdbdump`, that worked so incredibly well just ~2 years ago.

tim-moody, 2025-01-18:
OK. Sorry. This all seems to be in flux as tools that used to exist are gone, and it is very hard to find much in the way of documentation.

The git repo I cited has a dbparse.py among other things that may be the answer.

The README, which quite old, says:

```
root@raspberrypi:~/wireless-regdb-master-2024-10-07# cat README
 wireless-regdb
================
<https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb>

This repository contains the plain text version of the regulatory
database file I maintain for use with Central Regulatory Database
Agent daemon.  Also included is the compiled binary version of this
file signed with my RSA key.  This represents a good faith attempt
to capture regulatory information that is correct at the time of its last
modification.  This information is provided to you with no warranty
either expressed or implied.

Also included are the tools used to compile and sign the regulatory.bin
file as well as a MoinMoin macro used for viewing the database.

 TECHNICAL INFORMATION
=======================

The regulatory information in `db.txt' is stored in a human-readable
format which can be read using the `dbparse.py' python module. This
python module is used by the web viewer (Regulatory.py) which is
implemented as a MoinMoin macro (and used on http://wireless.kernel.org)
to allow viewing the database for verification.

The dbparse module is also used by db2bin.py and db2fw.py, the `compilers'
that compile the database to its binary formats.

For more information, please see the CRDA page:

        https://wireless.wiki.kernel.org/en/developers/regulatory/crda

and the full description of the Linux regulatory stack:

        https://wireless.wiki.kernel.org/en/developers/regulatory

John W. Linville
17 November 2008

```

I take this to mean that the db.txt file is definitive, though our firmware could be using an earlier version.

jvonau, 2025-01-18:
[raspi-config](https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L863-L915)  is using /usr/share/zoneinfo/iso3166.tab

tim-moody, 2025-01-18:
Probably this is a mistake as that table contains CD, but db.txt does not. A valid country code is not necessarily a valid regulatory domain code. All reg codes are country codes, but not the reverse.

jvonau, 2025-01-18:
Sure I agree, it is what is presented by raspi-confg, that would be the cause of the "CD" issue with country code rejected in dmesg.

tim-moody, 2025-01-18:
Interestingly in https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/log/?ofs=200 the most recent update of rules for Haiti is 10 years ago, so I don't think our problem with HT is a stale db, or am I confused that it didn't work.

https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/commit/?id=6e3e2a9272e2d8e1419d57a45339458b88729eef

holta, 2025-01-18:
Absolute worst case we can look into compiling regdbdump is someone upstream has dropped the ball?

The "Technocracy" really should pay all of us: Especially if we're successful in documenting a bit of very concrete progress here over coming weeks ~ to help enable the entire developing world (low-income educators/parents and offline folks of all kind in all countries, and non-countries too, i.e. stateless people leading even more excruciating lives often having been chased away from many different countries including where they were born!)

jvonau, 2025-01-18:
> Interestingly in https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/log/?ofs=200 the most recent update of rules for Haiti is 10 years ago, so I don't think our problem with HT is a stale db,

Don't think so either, seems to be pretty consistent now with the recent code changes, I'm not seeing a rejection of the country code in dmesg now. 

>or am I confused that it didn't work.

Think the restarting of wpa_supplicant was aggravating the situation in the recent past.

jvonau, 2025-01-19:
> Probably this is a mistake as that table contains CD, but db.txt does not. A valid country code is not necessarily a valid regulatory domain code. All reg codes are country codes, but not the reverse.

That would be worthy of an issue being opened at https://github.com/RPi-Distro/raspi-config and/or https://github.com/raspberrypi/rpi-imager/issues

Country code list from [rpi-imager](https://github.com/raspberrypi/rpi-imager/blob/master/src/countries.txt)

jvonau, 2025-01-19:
Can a list be established for Country Codes that are in iso3166.tab but are not in db.txt? Be helpful in opening an issue upstream and the knowledge base here for documentation in the wiki notes.

tim-moody, 2025-01-19:
>Can a list be established for Country Codes that are in iso3166.tab but are not in db.txt?

Think python is the only way if someone wants to write it.