iiab/iiab #3903: Fresh install of OS without IIAB on Raspberry Pi 5

tim-moody, 2025-01-18:


tim-moody, 2025-01-18:
Raspberry Pi: 5
OS: 2024-11-19-raspios-bookworm-arm64-lite.img
apt full upgrade: Yes
IIAB Installed: No
raspi-config nonint do_wifi_country: Yes - HT

tim-moody, 2025-01-19:
### Tests Needed

- cmdline and regulatory domain (see below)
- Please add others including enough details to conduct the tests

tim-moody, 2025-01-19:
### cmdline and regulatory domain

#### On image before boot

- image written with dd
- my keys copied to $ROOTFS_PART_MNT/root/.ssh/authorized_keys
- /boot/firmware does not exist on boot partition
- it does exist on file system partition

On boot partition

```
root@iiab-devel:/Files/Images/2024-11-19-raspios-bookworm-arm64-lite.img.Files/boot# cat cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=8a438930-02 rootfstype=ext4 fsck.repair=yes rootwait quiet init=/usr/lib/raspberrypi-sys-mods/firstboot

root@iiab-devel:/Files/Images/2024-11-19-raspios-bookworm-arm64-lite.img.Files/boot# ll ssh
-rwxr-xr-x 1 root root 0 Jan 16 22:33 ssh*

root@iiab-devel:/Files/Images/2024-11-19-raspios-bookworm-arm64-lite.img.Files/boot# cat userconf.txt
pi:$5$wYllupbvp8$tdUzKq4AngIPY0oXKydNT4mWoi0PMVp7/DEB9295oPD:19086:0:99999:7:::
```

On filesystem partition

```
root@iiab-devel:/Files/Images/2024-11-19-raspios-bookworm-arm64-lite.img.Files/sdcard/boot# cat cmdline.txt
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/cmdline.txt

root@iiab-devel:/Files/Images/2024-11-19-raspios-bookworm-arm64-lite.img.Files/sdcard/boot# ll firmware/
total 8
drwxr-xr-x 2 root root 4096 Nov 19 13:38 ./
drwxr-xr-x 3 root root 4096 Nov 19 13:39 ../
```

jvonau, 2025-01-19:
OK now the first script to run on the fresh image is [firstboot](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/usr/lib/raspberrypi-sys-mods/firstboot). Within that file there is a check for "custom.toml" for some future functionality, next is a call to [get_fw_loc](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/usr/lib/raspberrypi-sys-mods/get_fw_loc)  to figure out if /boot or /boot/firmware is the mount point for the first partition. Next [init_config](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/usr/lib/raspberrypi-sys-mods/init_config) is used to apply the userconf.txt customization.

Boot the rpi, should auto reboot after the resizing is done.  After that you should be able to ssh in over the wire, don't run raspi-config just let, lets see if we can unblock the wifi without it. Back in a bit as I want be concise with the next part and might need you to poke at some files.

tim-moody, 2025-01-19:
#### After first boot

root@raspberrypi:/boot# cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=b2ddbb49-02 rootfstype=ext4 fsck.repair=yes rootwait

root@raspberrypi:/boot# ll /boot
total 41724
drwxr-xr-x  3 root root     4096 Nov 19 13:39 .
drwxr-xr-x 18 root root     4096 Nov 19 13:39 ..
-rw-r--r--  1 root root       83 Oct  8 19:49 System.map-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root       83 Oct  8 19:49 System.map-6.6.51+rpt-rpi-v8
-rw-r--r--  1 root root       92 Nov 19 13:32 cmdline.txt
-rw-r--r--  1 root root       91 Nov 19 13:32 config.txt
-rw-r--r--  1 root root   238336 Oct  8 19:49 config-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root   238325 Oct  8 19:49 config-6.6.51+rpt-rpi-v8
drwxr-xr-x  3 root root     4096 Jan  1  1970 firmware
-rw-r--r--  1 root root 11813294 Nov 19 13:39 initrd.img-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root 11809677 Nov 19 13:39 initrd.img-6.6.51+rpt-rpi-v8
lrwxrwxrwx  1 root root       18 Nov 19 13:39 issue.txt -> firmware/issue.txt
lrwxrwxrwx  1 root root       17 Nov 19 13:32 overlays -> firmware/overlays
-rw-r--r--  1 root root  9292371 Oct  8 19:49 vmlinuz-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root  9284374 Oct  8 19:49 vmlinuz-6.6.51+rpt-rpi-v8

root@raspberrypi:/boot# /usr/sbin/iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

>OK now the first script to run on the fresh image is [firstboot](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/usr/lib/raspberrypi-sys-mods/firstboot)

No way to avoid that as it happens when the rpi powered up with the image on the sdcard.

tim-moody, 2025-01-19:
- after 10 minutes the led on the rpi 5 turned orange and my ssh session was dropped and fails to reconnect.
- I powered off with button and powered back on and am able to ssh in

jvonau, 2025-01-19:
>No way to avoid that as it happens when the rpi powered up with the image on the sdcard.

Yea, no problem can't change that. 
>after 10 minutes the led on the rpi 5 turned orange and my ssh session was dropped and fails to reconnect.

Think that was the filesystem resizing, how big is the SDcard?

Now based on [do_wifi_country](https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L863-L924) think the next steps would be

- may not exist `sudo rm -f /etc/default/crda`
- `sudo nmcli radio wifi on`
- `sudo rfkill unblock wifi`
- should be present `sudo sed -i 's/^WirelessEnabled=.*/WirelessEnabled=true/' /var/lib/NetworkManager/NetworkManager.state` 
- check "/var/lib/systemd/rfkill/*:wlan" should contain 0
- reboot or restart NetworkManager

tim-moody, 2025-01-19:
256G. also I forgot to apt update/upgrade

jvonau, 2025-01-19:
Do the apt stuff now, might get a newer version of raspi-config

tim-moody, 2025-01-19:
crda is not installed

root@raspberrypi:~# cat /var/lib/NetworkManager/NetworkManager.state
[main]
NetworkingEnabled=true
WirelessEnabled=true
WWANEnabled=true

after rfkill unblock wifi and before sed -i 's/^WirelessEnabled=.*/WirelessEnabled=true/' (which I accordingly didn't run)

root@raspberrypi:/boot# ll /var/lib/systemd/rfkill/*:wlan
-rw-r--r-- 1 root root 2 Nov 19 13:39 /var/lib/systemd/rfkill/platform-1001100000.mmc:wlan

root@raspberrypi:~# cat /var/lib/systemd/rfkill/platform-1001100000.mmc:wlan
0

reboot

tim-moody, 2025-01-19:
root@raspberrypi:~# /usr/sbin/iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

tim-moody, 2025-01-19:
apt update
apt -y full-upgrade
reboot

jvonau, 2025-01-19:
That 00 is expected, so just `rfkill` is needed. I'm going to assume /etc/NetworkManager/system-connections is empty?

tim-moody, 2025-01-19:
yes. empty

jvonau, 2025-01-19:
OK then connect up to an AP, might create files in /etc/NetworkManager/system-connections that would be used upon reboot.

tim-moody, 2025-01-19:
```
root@raspberrypi:~# nmcli device wifi connect SunRoom password xxxx
Device 'wlan0' successfully activated with 'c735385f-3d4f-4da4-88f6-5e4e2be5846e'.
root@raspberrypi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.60/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 65184sec preferred_lft 65184sec
    inet6 fe80::4a06:aa0d:d278:5ad0/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:cb brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.64/24 brd 192.168.3.255 scope global dynamic noprefixroute wlan0
       valid_lft 86394sec preferred_lft 86394sec
    inet6 fe80::2bfe:d8af:8b33:c000/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
```

I am able to ssh in to wlan0 address as well as eth0.

jvonau, 2025-01-20:
- NetworkManager & wpa_supplicant logging for the active connection
- Contents of /etc/NetworkManager/system-connection with the connection active? Think save is the default.
- Tear down connection, how did you do that? 
- Try `sudo iw reg set CD` check dmesg for rejected country code and `sudo iw reg get`
- Try `sudo iw reg set HT` check dmesg for rejected country code and `sudo iw reg get`
- reboot to clear country code 
- try making a hotspot NM, document the commands, `sudo iw dev` NetworkManager & wpa_supplicant logging for the AP and if a file is created in /etc/NetworkManager/system-connection/ Was country code needed?

These would be very useful.

tim-moody, 2025-01-20:
```
root@raspberrypi:~# cat  /etc/NetworkManager/system-connections/SunRoom.nmconnection
[connection]
id=SunRoom
uuid=c735385f-3d4f-4da4-88f6-5e4e2be5846e
type=wifi
interface-name=wlan0

[wifi]
mode=infrastructure
ssid=SunRoom

[wifi-security]
auth-alg=open
key-mgmt=wpa-psk
psk=xxxxx

[ipv4]
method=auto

[ipv6]
addr-gen-mode=default
method=auto

[proxy]

root@raspberrypi:~# cat wpa.log
Nov 19 13:39:32 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Nov 19 13:39:32 raspberrypi wpa_supplicant[866]: Successfully initialized wpa_supplicant
Nov 19 13:39:33 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 19 17:10:47 raspberrypi systemd[1]: Stopping wpa_supplicant.service - WPA supplicant...
Jan 19 17:10:47 raspberrypi systemd[1]: wpa_supplicant.service: Deactivated successfully.
Jan 19 17:10:47 raspberrypi systemd[1]: Stopped wpa_supplicant.service - WPA supplicant.
Jan 19 17:11:11 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 19 17:11:11 raspberrypi wpa_supplicant[639]: Successfully initialized wpa_supplicant
Jan 19 17:11:11 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 19 17:18:51 raspberrypi wpa_supplicant[639]: nl80211: kernel reports: Registration to specific type not supported
Jan 19 17:22:31 raspberrypi systemd[1]: Stopping wpa_supplicant.service - WPA supplicant...
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: p2p-dev-wlan0: CTRL-EVENT-TERMINATING
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Jan 19 17:22:31 raspberrypi wpa_supplicant[639]: wlan0: CTRL-EVENT-TERMINATING
Jan 19 17:22:31 raspberrypi systemd[1]: wpa_supplicant.service: Deactivated successfully.
Jan 19 17:22:31 raspberrypi systemd[1]: Stopped wpa_supplicant.service - WPA supplicant.
Jan 19 17:22:47 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 19 17:22:47 raspberrypi wpa_supplicant[627]: Successfully initialized wpa_supplicant
Jan 19 17:22:47 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 19 17:22:48 raspberrypi wpa_supplicant[627]: nl80211: kernel reports: Registration to specific type not supported
Jan 19 17:27:04 raspberrypi systemd[1]: Stopping wpa_supplicant.service - WPA supplicant...
Jan 19 17:27:04 raspberrypi wpa_supplicant[627]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: p2p-dev-wlan0: CTRL-EVENT-TERMINATING
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Jan 19 17:27:05 raspberrypi wpa_supplicant[627]: wlan0: CTRL-EVENT-TERMINATING
Jan 19 17:27:05 raspberrypi systemd[1]: wpa_supplicant.service: Deactivated successfully.
Jan 19 17:27:05 raspberrypi systemd[1]: Stopped wpa_supplicant.service - WPA supplicant.
Jan 19 17:27:21 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 19 17:27:21 raspberrypi wpa_supplicant[631]: Successfully initialized wpa_supplicant
Jan 19 17:27:21 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 19 17:27:21 raspberrypi wpa_supplicant[631]: nl80211: kernel reports: Registration to specific type not supported
Jan 19 23:20:50 raspberrypi wpa_supplicant[631]: wlan0: WPS-PBC-ACTIVE
Jan 19 23:20:50 raspberrypi wpa_supplicant[631]: wlan0: WPS-CANCEL
Jan 19 23:20:54 raspberrypi wpa_supplicant[631]: wlan0: Trying to associate with SSID 'SunRoom'
Jan 19 23:20:54 raspberrypi wpa_supplicant[631]: wlan0: Associated with c0:4a:00:e5:ec:04
Jan 19 23:20:54 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 19 23:20:54 raspberrypi wpa_supplicant[631]: wlan0: WPA: Key negotiation completed with c0:4a:00:e5:ec:04 [PTK=CCMP GTK=TKIP]
Jan 19 23:20:54 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-CONNECTED - Connection to c0:4a:00:e5:ec:04 completed [id=0 id_str=]

```

tim-moody, 2025-01-20:
```
journalctl |grep NetworkManager

Nov 19 13:39:32 raspberrypi systemd[1]: Starting NetworkManager.service - Network Manager...
Nov 19 13:39:32 raspberrypi NetworkManager[863]: <info>  [1732023572.9164] NetworkManager (version 1.42.4) is starting... (boot:b293f370-3bae-4bdf-a851-b2b3a071f8b3)
Nov 19 13:39:32 raspberrypi NetworkManager[863]: <info>  [1732023572.9165] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf)
Nov 19 13:39:32 raspberrypi NetworkManager[863]: <info>  [1732023572.9199] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Nov 19 13:39:32 raspberrypi NetworkManager[863]: <info>  [1732023572.9304] manager[0x55565ca0a000]: monitoring kernel firmware directory '/lib/firmware'.
Nov 19 13:39:32 raspberrypi NetworkManager[863]: <info>  [1732023572.9307] monitoring ifupdown state file '/run/network/ifstate'.
Nov 19 13:39:33 raspberrypi systemd[1]: Started NetworkManager.service - Network Manager.
Nov 19 13:39:33 raspberrypi systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.4695] hostname: hostname: using hostnamed
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.4696] hostname: static hostname changed from (none) to "raspberrypi"
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.4745] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.4755] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.4757] manager[0x55565ca0a000]: rfkill: Wi-Fi hardware radio set disabled
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.4757] manager[0x55565ca0a000]: rfkill: WWAN hardware radio set enabled
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.5401] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.5632] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.5767] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.5892] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6071] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6074] manager: rfkill: Wi-Fi disabled by radio killswitch; disabled by state file
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6075] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6075] manager: Networking is enabled by state file
Nov 19 13:39:33 raspberrypi dbus-daemon[546]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.8' (uid=0 pid=863 comm="/usr/sbin/NetworkManager --no-daemon")
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6124] settings: Loaded settings plugin: ifupdown ("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6124] settings: Loaded settings plugin: keyfile (internal)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6125] ifupdown: management mode: unmanaged
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6148] ifupdown: interfaces file /etc/network/interfaces doesn't exist
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6254] dhcp: init: Using DHCP client 'internal'
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6256] manager: (lo): new Loopback device (/org/freedesktop/NetworkManager/Devices/1)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6410] device (lo): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6414] device (lo): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6420] device (lo): Activation: starting connection 'lo' (c8e58002-f8a4-4a09-85f6-52de4c1bd67b)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6432] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
Nov 19 13:39:33 raspberrypi NetworkManager[863]: <info>  [1732023573.6462] settings: (eth0): created default wired connection 'Wired connection 1'
[... cut ...]
Jan 19 23:20:57 raspberrypi NetworkManager[613]: <info>  [1737328857.2571] device (wlan0): Activation: successful, device activated.
Jan 19 23:21:07 raspberrypi systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Jan 20 05:27:25 raspberrypi NetworkManager[613]: <info>  [1737350845.9210] dhcp4 (eth0): state changed new lease, address=192.168.3.60
Jan 20 11:20:56 raspberrypi NetworkManager[613]: <info>  [1737372056.3816] dhcp4 (wlan0): state changed new lease, address=192.168.3.64

```

tim-moody, 2025-01-20:
```
root@raspberrypi:~# nmcli -t -f type,name connection show
802-3-ethernet:Wired connection 1
loopback:lo
802-11-wireless:SunRoom

root@raspberrypi:~# nmcli con delete SunRoom
Connection 'SunRoom' (c735385f-3d4f-4da4-88f6-5e4e2be5846e) successfully deleted.

root@raspberrypi:~# nmcli -t -f type,name connection show
802-3-ethernet:Wired connection 1
loopback:lo

root@raspberrypi:~# ll   /etc/NetworkManager/system-connections
total 8
drwxr-xr-x 2 root root 4096 Jan 20 14:21 .
drwxr-xr-x 7 root root 4096 Nov 19 13:33 ..

```

jvonau, 2025-01-20:
Nice thanks will review later after work 
have a look at #3908

tim-moody, 2025-01-20:
Try another more recent router

```
root@raspberrypi:~# nmcli device wifi connect rrrr password pppp
Device 'wlan0' successfully activated with 'a71e4363-1b9d-49b2-a187-54f5b699fdd6'.
```

AND now we are in CA

root@raspberrypi:~# /usr/sbin/iw reg get
global
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

tim-moody, 2025-01-20:
```
Jan 19 17:27:21 raspberrypi systemd[1]: Starting NetworkManager.service - Network Manager...
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6353] NetworkManager (version 1.42.4) is starting... (boot:40e43715-af62-4325-9e66-a7e2b7e3105c)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6354] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf)
Jan 19 17:27:21 raspberrypi systemd[1]: Started NetworkManager.service - Network Manager.
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6374] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6491] manager[0x555618da0000]: monitoring kernel firmware directory '/lib/firmware'.
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6491] monitoring ifupdown state file '/run/network/ifstate'.
Jan 19 17:27:21 raspberrypi systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6532] hostname: hostname: using hostnamed
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6532] hostname: static hostname changed from (none) to "raspberrypi"
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6556] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6573] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6578] manager[0x555618da0000]: rfkill: Wi-Fi hardware radio set enabled
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.6578] manager[0x555618da0000]: rfkill: WWAN hardware radio set enabled
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7496] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7658] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7718] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7783] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7849] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7852] manager: rfkill: Wi-Fi enabled by radio killswitch; enabled by state file
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7853] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7853] manager: Networking is enabled by state file
Jan 19 17:27:21 raspberrypi dbus-daemon[527]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=613 comm="/usr/sbin/NetworkManager --no-daemon")
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7870] settings: Loaded settings plugin: ifupdown ("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7870] settings: Loaded settings plugin: keyfile (internal)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7870] ifupdown: management mode: unmanaged
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7873] ifupdown: interfaces file /etc/network/interfaces doesn't exist
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7896] dhcp: init: Using DHCP client 'internal'
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7898] manager: (lo): new Loopback device (/org/freedesktop/NetworkManager/Devices/1)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7949] device (lo): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7952] device (lo): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7957] device (lo): Activation: starting connection 'lo' (b39339ec-da24-43ae-8823-5fce8d9e3cbc)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7966] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7975] settings: (eth0): created default wired connection 'Wired connection 1'
Jan 19 17:27:21 raspberrypi NetworkManager[613]: <info>  [1737307641.7975] device (eth0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
[... cut ...]
Jan 20 14:46:36 raspberrypi NetworkManager[613]: <info>  [1737384396.5978] device (wlan0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Jan 20 14:46:36 raspberrypi NetworkManager[613]: <info>  [1737384396.5984] device (wlan0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Jan 20 14:46:36 raspberrypi NetworkManager[613]: <info>  [1737384396.6018] device (wlan0): Activation: successful, device activated.
Jan 20 14:46:46 raspberrypi systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
```

jvonau, 2025-01-20:
> Jan 20 14:46:29 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=CA

Bingo

tim-moody, 2025-01-20:
Now reconnect to CA router and try to set code to US

Device 'wlan0' successfully activated with '8d6111b2-acf4-4075-a129-fce6d8899899'.
root@raspberrypi:~# iw reg get
global
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

root@raspberrypi:~# iw reg set US
root@raspberrypi:~# iw reg get
global
country 98: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 100), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 100), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

AND 98 raises its head

```
root@raspberrypi:~# journalctl -b |grep wpa_supplicant
Jan 20 15:28:32 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 20 15:28:32 raspberrypi wpa_supplicant[631]: Successfully initialized wpa_supplicant
Jan 20 15:28:32 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 20 15:28:33 raspberrypi wpa_supplicant[631]: nl80211: kernel reports: Registration to specific type not supported
Jan 20 15:28:39 raspberrypi wpa_supplicant[631]: wlan0: WPS-PBC-ACTIVE
Jan 20 15:28:39 raspberrypi wpa_supplicant[631]: wlan0: WPS-CANCEL
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: MBO: Disable MBO/OCE due to misbehaving AP not having enabled PMF
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: wlan0: Trying to associate with SSID 'TP-Link_8CC4'
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: wlan0: Associated with 98:25:4a:c7:8c:c3
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: MBO: Disable MBO/OCE due to misbehaving AP not having enabled PMF
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: wlan0: WPA: Key negotiation completed with 98:25:4a:c7:8c:c3 [PTK=CCMP GTK=CCMP]
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-CONNECTED - Connection to 98:25:4a:c7:8c:c3 completed [id=0 id_str=]
Jan 20 15:28:43 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=CA
Jan 20 15:31:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-DISCONNECTED bssid=98:25:4a:c7:8c:c3 reason=3 locally_generated=1
Jan 20 15:31:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 20 15:31:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 20 15:37:49 raspberrypi wpa_supplicant[631]: wlan0: WPS-CANCEL
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: MBO: Disable MBO/OCE due to misbehaving AP not having enabled PMF
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: wlan0: Trying to associate with SSID 'TP-Link_8CC4'
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: wlan0: Associated with 98:25:4a:c7:8c:c3
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: MBO: Disable MBO/OCE due to misbehaving AP not having enabled PMF
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=CA
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: wlan0: WPA: Key negotiation completed with 98:25:4a:c7:8c:c3 [PTK=CCMP GTK=CCMP]
Jan 20 15:37:55 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-CONNECTED - Connection to 98:25:4a:c7:8c:c3 completed [id=0 id_str=]
Jan 20 15:38:55 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=INTERSECTION

```

So I think the rules state that if more than one code is active the lowest common denominator is used, which seems to be  type=INTERSECTION, which I take to be 98

tim-moody, 2025-01-20:
Now if I disconnect from CA router it reverts to US

root@raspberrypi:~# nmcli con delete TP-Link_8CC4
Connection 'TP-Link_8CC4' (8d6111b2-acf4-4075-a129-fce6d8899899) successfully deleted.
root@raspberrypi:~# iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

Jan 20 15:43:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-DISCONNECTED bssid=98:25:4a:c7:8c:c3 reason=3 locally_generated=1
Jan 20 15:43:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 20 15:43:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 20 15:43:56 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US

tim-moody, 2025-01-20:
```
root@raspberrypi:~# iw reg set CD
root@raspberrypi:~# dmesg | grep ieee80211
root@raspberrypi:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

```

And nothing more in wpa_supplicant

tim-moody, 2025-01-20:
reconnect to CA router

```
iw reg set CD

root@raspberrypi:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

root@raspberrypi:~# dmesg | grep ieee80211
```

nothing

tim-moody, 2025-01-20:
Another reboot

```
root@raspberrypi:~# iw reg get
global
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

root@raspberrypi:~# iw reg set CD

root@raspberrypi:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

root@raspberrypi:~# journalctl -b |grep wpa_supplicant
Jan 20 16:38:28 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 20 16:38:28 raspberrypi wpa_supplicant[631]: Successfully initialized wpa_supplicant
Jan 20 16:38:28 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 20 16:38:29 raspberrypi wpa_supplicant[631]: nl80211: kernel reports: Registration to specific type not supported
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: WPS-CANCEL
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: Trying to associate with SSID 'TP-Link_8CC4'
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: Associated with 98:25:4a:c7:8c:c3
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=CA
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: WPA: Key negotiation completed with 98:25:4a:c7:8c:c3 [PTK=CCMP GTK=CCMP]
Jan 20 16:38:35 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-CONNECTED - Connection to 98:25:4a:c7:8c:c3 completed [id=0 id_str=]
Jan 20 16:39:57 raspberrypi wpa_supplicant[631]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

root@raspberrypi:~# journalctl -b |grep ieee80211
Jan 20 16:38:28 raspberrypi NetworkManager[611]: <info>  [1737391108.7539] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)

```

>try making a hotspot NM, document the commands

Think this needs both hostapd and dnsmasq, but I'm not sure how to do it. But I also found:

`nmcli device wifi hotspot ssid <hotspot name> password <hotspot password> ifname wlan0`

tim-moody, 2025-01-20:
disconnect from CA router and reboot

```
root@raspberrypi:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)
```

>try making a hotspot NM, document the commands

https://www.baeldung.com/linux/nmcli-wap-sharing-internet

```
root@raspberrypi:~# sudo nmcli connection add type wifi ifname wlan0 con-name testhotspot autoconnect yes ssid testhotspot
Connection 'testhotspot' (7615090c-492e-482b-bfb0-a6fcdc70448a) successfully added.
root@raspberrypi:~# sudo nmcli connection modify testhotspot 802-11-wireless.mode ap 802-11-wireless.band bg ipv4.method shared
root@raspberrypi:~# sudo nmcli connection modify testhotspot wifi-sec.key-mgmt wpa-psk
sudo nmcli connection modify testhotspot wifi-sec.psk 12345678
root@raspberrypi:~# sudo nmcli connection up testhotspot
Connection successfully activated (D-Bus active path: /org/freedesktop/NetworkManager/ActiveConnection/4)

```

testhotspot shows up on my laptop

tim-moody, 2025-01-20:
```
root@raspberrypi:~# journalctl -b |grep wpa_supplicant
Jan 20 16:46:14 raspberrypi systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 20 16:46:14 raspberrypi wpa_supplicant[621]: Successfully initialized wpa_supplicant
Jan 20 16:46:14 raspberrypi systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 20 16:46:14 raspberrypi wpa_supplicant[621]: nl80211: kernel reports: Registration to specific type not supported
Jan 20 16:50:48 raspberrypi wpa_supplicant[621]: Note: nl80211 driver interface is not designed to be used with ap_scan=2; this can result in connection failures
Jan 20 16:50:48 raspberrypi wpa_supplicant[621]: wlan0: interface state UNINITIALIZED->ENABLED
Jan 20 16:50:48 raspberrypi wpa_supplicant[621]: wlan0: AP-ENABLED
Jan 20 16:50:48 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-CONNECTED - Connection to 2c:cf:67:0a:aa:cb completed [id=0 id_str=]
Jan 20 16:52:18 raspberrypi wpa_supplicant[621]: wlan0: interface state ENABLED->DISABLED
Jan 20 16:52:18 raspberrypi wpa_supplicant[621]: wlan0: AP-DISABLED
Jan 20 16:52:18 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-DISCONNECTED bssid=2c:cf:67:0a:aa:cb reason=3 locally_generated=1
Jan 20 16:52:18 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jan 20 16:52:18 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 20 16:52:18 raspberrypi wpa_supplicant[621]: Note: nl80211 driver interface is not designed to be used with ap_scan=2; this can result in connection failures
Jan 20 16:52:25 raspberrypi wpa_supplicant[621]: wlan0: interface state UNINITIALIZED->ENABLED
Jan 20 16:52:25 raspberrypi wpa_supplicant[621]: wlan0: AP-ENABLED
Jan 20 16:52:25 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-CONNECTED - Connection to 2c:cf:67:0a:aa:cb completed [id=0 id_str=]
Jan 20 16:59:31 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 20 16:59:31 raspberrypi wpa_supplicant[621]: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 96:c1:02:d8:2e:42
Jan 20 16:59:32 raspberrypi wpa_supplicant[621]: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 96:c1:02:d8:2e:42
Jan 20 16:59:33 raspberrypi wpa_supplicant[621]: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 96:c1:02:d8:2e:42
Jan 20 16:59:34 raspberrypi wpa_supplicant[621]: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 96:c1:02:d8:2e:42
Jan 20 17:00:03 raspberrypi wpa_supplicant[621]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 20 17:00:03 raspberrypi wpa_supplicant[621]: wlan0: AP-STA-CONNECTED 96:c1:02:d8:2e:42
Jan 20 17:00:03 raspberrypi wpa_supplicant[621]: wlan0: EAPOL-4WAY-HS-COMPLETED 96:c1:02:d8:2e:42

root@raspberrypi:~# iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

root@raspberrypi:~# journalctl -b |grep ieee80211
Jan 20 16:46:14 raspberrypi NetworkManager[620]: <info>  [1737391574.6684] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Jan 20 16:52:18 raspberrypi kernel: ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
Jan 20 16:52:18 raspberrypi kernel: ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
Jan 20 17:00:04 raspberrypi kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0x8a, action 0x6
Jan 20 17:02:38 raspberrypi kernel: ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0x8a, action 0x6

```

tim-moody, 2025-01-20:
```
root@raspberrypi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.60/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 85188sec preferred_lft 85188sec
    inet6 fe80::4a06:aa0d:d278:5ad0/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:cb brd ff:ff:ff:ff:ff:ff
    inet 10.42.0.1/24 brd 10.42.0.255 scope global noprefixroute wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::5d0c:b8dd:eba4:6cb2/64 scope link noprefixroute
       valid_lft forever preferred_lft forever

root@raspberrypi:~# cat  /etc/NetworkManager/system-connections/testhotspot.nmconnection
[connection]
id=testhotspot
uuid=7615090c-492e-482b-bfb0-a6fcdc70448a
type=wifi
interface-name=wlan0
timestamp=1737391849

[wifi]
band=bg
mode=ap
ssid=testhotspot

[wifi-security]
key-mgmt=wpa-psk
psk=12345678

[ipv4]
method=shared

[ipv6]
addr-gen-mode=default
method=auto

[proxy]

```

my laptop got an ip from 10.42.0.1

jvonau, 2025-01-21:
I'll review after work tomorrow, there is a lot of material to digest. What is your take away from all this?

jvonau, 2025-01-21:
> So it seems that HT, though valid in db.txt triggers 'Firmware rejected country setting' in our specific firmware, but CD does not.

CD would not have any mapping in db.txt, seems to be essentially ignored.

jvonau, 2025-01-22:
With not changing cmdline.txt so no raspi-config can you disable all wifi connections (delete  /etc/NetworkManager/system-connections/*), ssh in over wired and run `iw reg set HT` `dmesg | grep ieee80211`.
Looking to see in the absents of wifi if the firmware is still rejecting the country code.

tim-moody, 2025-01-22:
```
root@raspberrypi:~# dmesg | grep ieee80211
root@raspberrypi:~# iw reg set HT
root@raspberrypi:~# dmesg | grep ieee80211
root@raspberrypi:~# ll /etc/NetworkManager/system-connections
total 8
drwxr-xr-x 2 root root 4096 Jul 24  2023 .
drwxr-xr-x 7 root root 4096 Nov 19 13:33 ..
root@raspberrypi:~# cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=c5da4552-02 rootfstype=ext4 fsck.repair=yes rootwait

```

jvonau, 2025-01-22:
I'm going to assume `iw reg get` show HT, and connecting up to the CA access point results in CA or 98?