iiab/iiab #3906: Bug: On RPi4, USB stick is not mounted after reboot

avni, 2025-01-19:
To reproduce: 
1. Plug in a USB stick into an RPi4 that has IIAB installed and is on. 
2. lsblk should show it mounted at a point such as: `/media/usb2` a là: 

```
sdb      8:16   1  58.6G  0 disk 
└─sdb1   8:17   1  58.6G  0 part /media/usb2
```

3. Reboot your IIAB
4. Stick is no longer mounted. If you remove and replug in the stick, it will be mounted. Running `mount` manually also works.

---

I can test these cases next week:
NB: Check if behavior is different on an RPi4 that does NOT have IIAB installed.
NB: Check if behavior is different when you plug in stick prior to booting RPi4.

holta, 2025-01-19:
FYI Even though this appears to be longstanding behavior, many view this as a bug, as they still expect it to work otherwise! 🧘‍♀️

Related:

- PR #3875

jvonau, 2025-01-23:
Some required reading #654  #682 #2282 #2607 and the linked issues and PRs
Seems like a timing issue, been years since looking at this.

avni, 2025-01-23:
This was seen before: #329

jvonau, 2025-01-23:
The issue is not mounted when booting with the key inserted, on a hunch try editing
 "/etc/systemd/system/usbmount@.service" 
change 
>After=rc-local.service

to 
>#After=rc-local.service

and reboot.

jvonau, 2025-01-23:
Much better docs around now: https://www.flatcar.org/docs/latest/setup/systemd/udev-rules/ might need to revisit the udev rules used seems SUBSYSTEM=="block" might be a better fit.

avni, 2025-01-24:
Just tested this and it works! Tested before (where it unmounts the USB stick and doesn't remount it after reboot) and after (where the stick is remounted after reboot)! This will help teachers so much! 

One note is that if the USB is not mounted and you reboot, it will NOT mount after reboot even with this change. 

Thanks @jvonau! 🎉

closed 2025-01-24 by avni

holta, 2025-01-24:
> if the USB is not mounted and you reboot, it will NOT mount after reboot even with this change.

Inserting USB stick while IIAB is off (and then booting) presumably also fails?

jvonau, 2025-01-24:
> > if the USB is not mounted and you reboot, it will NOT mount after reboot even with this change.
> 
> Inserting USB stick while IIAB is off (and then booting) presumably also fails?

I'd be looking at some corruption on the usbkey.
`dmesg` and look for some thing like 
''FAT-fs (sda2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

The safest way would to power down the machine and remove the key, might get away with pulling out after the halt phase but before the boot. Something should "properly un-mount and eject the key" like on a desktop, not just yank the drive out. Might look at having a password protected admin page for the un-mounting utility.

`sudo systemctl status usbmount@dev-sdX0.service` to have look if the mounting worked is mounted or not 
"X being the drive letter 0 being the partition number"
>usbmount@dev-sda2.service
     Loaded: loaded (/etc/systemd/system/usbmount@.service; static)
     Active: active (exited) since Fri 2025-01-24 14:39:28 CST; 43s ago
    Process: 546 ExecStart=/usr/share/usbmount/usbmount add (code=exited, status=0/SUCCESS)
   Main PID: 546 (code=exited, status=0/SUCCESS)
        CPU: 169ms
Jan 24 14:39:28 box systemd[1]: Started usbmount@dev-sda2.service.
Jan 24 14:39:33 box usbmount[916]: executing command: mount -tvfat -osync,noexec,nodev,noatime,nodiratime,umask=0000 dev/sda2 >
Jan 24 14:39:33 box usbmount[924]: executing command: run-parts /etc/usbmount/mount.d
Jan 24 14:39:33 box 70-usb-library[928]: Display entire USB drive is True. Checking for rootfs or /library on /media/usb0.
Jan 24 14:39:33 box 70-usb-library[944]: Found Share Directory /media/usb0.
Jan 24 14:39:33 box 70-usb-library[948]: Creating link to /library/www/html/local_content/USB0.

Hope that helps.

jvonau, 2025-01-24:
I could not duplicate having the key unmounted then rebooting and not have the key mounted automatically. Power issue perhaps?

jvonau, 2025-01-24:
The dev-sdX0 is passed from udev, a X for each key and a 0 for each partition on the key

avni, 2025-01-24:
I am not sure if rebooting with the prior code (`After=rc-local.service`) and then changing the code and rebooting played a part. To be clear on what works and what doesn't, I'm testing the following use cases: 

**- Test Case 1 (w/ Old code)** 
    - USB stick plugged in, reboot. Result Expected: USB stick is unmounted. 
    - USB stick unmounted, Run fsck, and check status of volume, if it was umounted properly or not etc. 
**-  Test Case 2 (w/ New code)** 
    -  USB stick is plugged in but is unmounted properly. 
    - Reboot. 
    - Result expected: USB stick is mounted. 
**- Test Case 3 (w/ New code)** 
    - No USB stick plugged in. 
    - Turn IIAB server off.
    - Plug in USB stick
    - Turn IIAB server on
    - Result expected: USB stick is mounted. 
**- Test Case 4 (w/ New code)** 
    - USB stick is plugged in and mounted.
    - Turn IIAB server off.
    -  Turn IIAB server on
    - Result expected: USB stick is mounted. 
**- Test Case 5 (w/ New code)**  ✅ (confirmed and working per above).
    - USB stick is plugged in and mounted.
    - Reboot IIAB
    - Result expected: USB stick is mounted.

avni, 2025-01-25:
**Test Case 1:** Result: USB is not mounted as expected upon reboot with old code. I don't see anything from `dmesg` or `systemctl status` to indicate the volume was unmounted improperly. Regardless, I will run umount before Test Case 2. 

```
root@box:~# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda      8:0    0 953.9G  0 disk 
├─sda1   8:1    0   512M  0 part /media/usb0
│                                /boot/firmware
└─sda2   8:2    0 953.4G  0 part /media/usb1
                                 /
sdb      8:16   1  58.6G  0 disk 
└─sdb1   8:17   1  58.6G  0 part 
```

```
root@box:~# dmesg |grep sdb
[    3.712915] sd 1:0:0:0: [sdb] 122880000 512-byte logical blocks: (62.9 GB/58.6 GiB)
[    3.713088] sd 1:0:0:0: [sdb] Write Protect is off
[    3.713094] sd 1:0:0:0: [sdb] Mode Sense: 0b 00 00 08
[    3.713226] sd 1:0:0:0: [sdb] No Caching mode page found
[    3.713232] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[    3.741427]  sdb: sdb1
[    3.741722] sd 1:0:0:0: [sdb] Attached SCSI removable disk
```

```
root@box:~# sudo systemctl status usbmount@dev-sdb1.service
Warning: The unit file, source configuration file or drop-ins of usbmount@dev-sdb1.service changed on disk. Run 'systemctl daemon-reload' to reload units.
× usbmount@dev-sdb1.service
     Loaded: loaded (/etc/systemd/system/usbmount@.service; static)
     Active: failed (Result: exit-code) since Sat 2025-01-25 01:18:48 WAT; 6min ago
   Duration: 45.826s
    Process: 635 ExecStart=/usr/share/usbmount/usbmount add (code=exited, status=1/FAILURE)
    Process: 2317 ExecStopPost=/bin/umount /dev/sdb1 (code=exited, status=32)
   Main PID: 635 (code=exited, status=1/FAILURE)
        CPU: 22ms

Jan 25 01:18:02 box systemd[1]: Started usbmount@dev-sdb1.service.
Jan 25 01:18:47 box usbmount[2310]: cannot acquire lock /var/run/usbmount/.mount.lock
Jan 25 01:18:47 box systemd[1]: usbmount@dev-sdb1.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 01:18:48 box umount[2317]: umount: /dev/sdb1: not mounted.
Jan 25 01:18:48 box systemd[1]: usbmount@dev-sdb1.service: Control process exited, code=exited, status=32/n/a
Jan 25 01:18:48 box systemd[1]: usbmount@dev-sdb1.service: Failed with result 'exit-code'.
```

jvonau, 2025-01-25:
>The unit file, source configuration file or drop-ins of usbmount@dev-sdb1.service changed on disk. Run 'systemctl daemon-reload' to reload units.

I'm going to guess that was hand editing the unit file, the warning speaks for itself, you must run `systemctl daemon-reload` when an edit is done on any unit file as the old version is what is loaded into memory and will be run until reloaded or rebooted There is a pid for the stop command so something called 'stop'

Guess I should add use `sudo systemctl stop usbmount@dev-sdX0.service` to unmount to have a accurate picture when later when you use 'status'

 for me:

`sudo systemctl stop usbmount@dev-sda2.service`
`sudo systemctl status usbmount@dev-sda2.service`
>usbmount@dev-sda2.service
     Loaded: loaded (/etc/systemd/system/usbmount@.service; static)
     Active: inactive (dead) since Fri 2025-01-24 19:32:13 CST; 4s ago
   Duration: 4h 52min 45.146s
    Process: 546 ExecStart=/usr/share/usbmount/usbmount add (code=exited, status=0/SUCCESS)
    Process: 2158 ExecStop=/usr/sbin/iiab-clean-usb.sh dev/sda2 (code=exited, status=0/SUCCESS)
    Process: 2171 ExecStopPost=/bin/umount /dev/sda2 (code=exited, status=0/SUCCESS)
   Main PID: 546 (code=exited, status=0/SUCCESS)
        CPU: 62ms
Jan 24 14:39:28 box systemd[1]: Started usbmount@dev-sda2.service.
Jan 24 14:39:33 box usbmount[916]: executing command: mount -tvfat -osync,noexec,nodev,noatime,>
Jan 24 14:39:33 box usbmount[924]: executing command: run-parts /etc/usbmount/mount.d
Jan 24 14:39:33 box 70-usb-library[928]: Display entire USB drive is True. Checking for rootfs >
Jan 24 14:39:33 box 70-usb-library[944]: Found Share Directory /media/usb0.
Jan 24 14:39:33 box 70-usb-library[948]: Creating link to /library/www/html/local_content/USB0.
Jan 24 19:32:13 box systemd[1]: Stopping usbmount@dev-sda2.service...
Jan 24 19:32:13 box usbmount[2170]: /library/www/html/local_content/USB0 removed.
Jan 24 19:32:13 box systemd[1]: usbmount@dev-sda2.service: Deactivated successfully.
Jan 24 19:32:13 box systemd[1]: Stopped usbmount@dev-sda2.service.

avni, 2025-01-25:
✅ Test Case 4 (w/ New code) - Success 

🔎 Test Case 2 & 5 (w/ New code) - when I manually mount/unmount stick with `mount -o umask=0000 /dev/sdb1 /media/usb2` and `umount /media/usb2`, there seems to be inconsistent behavior where the server reboots with the disk not mounted, but after a few minutes will be, other times, upon reboot, the disk does not mount. I can't reproduce it consistently and it seems to work the majority of the time so I think it's good enough. We can advise teachers to unplug and replug the USB if needed. 

🔎 Test Case 3 (w/ New code) - If I do `umount /media/usb2` before shutting down, test case 3 is successful, otherwise upon reboot, drive is not mounted. 

🔎 Test Case 4.5 (w/ New code) [unmount disk but leave plugged in, turn off IIAB, turn on) - drive not mounted upon reboot. 

=====

**An example of Test Case 5 when it was unsuccessful in case it helps with debugging.** 

**Before reboot**

```
root@box:~# umount /media/usb2
umount: /media/usb2: not mounted.

root@box:~# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
....
sdb      8:16   1  58.6G  0 disk 
└─sdb1   8:17   1  58.6G  0 part 

root@box:~# mount -o umask=0000 /dev/sdb1 /media/usb2

root@box:~# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
....
sdb      8:16   1  58.6G  0 disk 
└─sdb1   8:17   1  58.6G  0 part /media/usb2

```

**After reboot**
```
iiab-admin@box:~ $ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
....
sdb      8:16   1  58.6G  0 disk 
└─sdb1   8:17   1  58.6G  0 part 

root@box:~# dmesg |grep sdb
[    3.810404] sd 1:0:0:0: [sdb] 122880000 512-byte logical blocks: (62.9 GB/58.6 GiB)
[    3.810603] sd 1:0:0:0: [sdb] Write Protect is off
[    3.810618] sd 1:0:0:0: [sdb] Mode Sense: 0b 00 00 08
[    3.810809] sd 1:0:0:0: [sdb] No Caching mode page found
[    3.810821] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[    3.814790]  sdb: sdb1
[    3.815260] sd 1:0:0:0: [sdb] Attached SCSI removable disk

root@box:~# sudo systemctl status usbmount@dev-sdb1.service
× usbmount@dev-sdb1.service
     Loaded: loaded (/etc/systemd/system/usbmount@.service; static)
     Active: failed (Result: exit-code) since Sat 2025-01-25 01:52:53 WAT; 1min 31s ago
   Duration: 30.137s
    Process: 641 ExecStart=/usr/share/usbmount/usbmount add (code=exited, status=1/FAILURE)
    Process: 2002 ExecStopPost=/bin/umount /dev/sdb1 (code=exited, status=32)
   Main PID: 641 (code=exited, status=1/FAILURE)
        CPU: 44ms

Jan 25 01:52:23 box systemd[1]: Started usbmount@dev-sdb1.service.
Jan 25 01:52:53 box usbmount[1995]: cannot acquire lock /var/run/usbmount/.mount.lock
Jan 25 01:52:53 box systemd[1]: usbmount@dev-sdb1.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 01:52:53 box umount[2002]: umount: /dev/sdb1: not mounted.
Jan 25 01:52:53 box systemd[1]: usbmount@dev-sdb1.service: Control process exited, code=exited, status=32/n/a
Jan 25 01:52:53 box systemd[1]: usbmount@dev-sdb1.service: Failed with result 'exit-code'.
```

=====
**Test Case 4.5 output**

```
root@box:~# sudo systemctl stop usbmount@dev-sdb1.service
root@box:~# sudo systemctl status usbmount@dev-sdb1.service
× usbmount@dev-sdb1.service
     Loaded: loaded (/etc/systemd/system/usbmount@.service; static)
     Active: failed (Result: exit-code) since Sat 2025-01-25 02:52:13 WAT; 1min 29s ago
   Duration: 1min 41.846s
    Process: 646 ExecStart=/usr/share/usbmount/usbmount add (code=exited, status=1/FAILURE)
    Process: 2376 ExecStopPost=/bin/umount /dev/sdb1 (code=exited, status=32)
   Main PID: 646 (code=exited, status=1/FAILURE)
        CPU: 20ms

Jan 25 02:50:31 box systemd[1]: Started usbmount@dev-sdb1.service.
Jan 25 02:52:13 box systemd[1]: usbmount@dev-sdb1.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 02:52:13 box umount[2376]: umount: /dev/sdb1: not mounted.
Jan 25 02:52:13 box systemd[1]: usbmount@dev-sdb1.service: Control process exited, code=exited, 

```

jvonau, 2025-01-25:
It the revised code fails to mount on boot in the future grab the usbmount status as above and `ps -AH`
Are there any current problems?

avni, 2025-01-25:
I think the biggest issue is when you unmount the USB stick, and reboot, the drive is not mounted. It seems to perhaps remember the prior state? 

Here's the status and ps -AH from that use case. 

```
root@box:~# sudo systemctl stop usbmount@dev-sdb1.service
root@box:~# sudo systemctl status usbmount@dev-sdb1.service
× usbmount@dev-sdb1.service
     Loaded: loaded (/etc/systemd/system/usbmount@.service; static)
     Active: failed (Result: exit-code) since Sat 2025-01-25 03:09:40 WAT; 18s ago
   Duration: 45.443s
    Process: 649 ExecStart=/usr/share/usbmount/usbmount add (code=exited, status=1/FAILURE)
    Process: 2489 ExecStopPost=/bin/umount /dev/sdb1 (code=exited, status=32)
   Main PID: 649 (code=exited, status=1/FAILURE)
        CPU: 22ms

Jan 25 03:08:55 box systemd[1]: Started usbmount@dev-sdb1.service.
Jan 25 03:09:40 box usbmount[2488]: cannot acquire lock /var/run/usbmount/.mount.lock
Jan 25 03:09:40 box systemd[1]: usbmount@dev-sdb1.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 03:09:40 box umount[2489]: umount: /dev/sdb1: not mounted.
Jan 25 03:09:40 box systemd[1]: usbmount@dev-sdb1.service: Control process exited, code=exited, status=32/n/a
Jan 25 03:09:40 box systemd[1]: usbmount@dev-sdb1.service: Failed with result 'exit-code'.
```

**Output of root@box:~# ps -AH is at: https://pastebin.com/qfX5Rvms**

avni, 2025-01-25:
> Edit:
> /var/run/usbmount/.mount.lock seems to be the issue
> /var/run is a symlink to /run that is on a tmpfs and needs to be created first.

Yes, agree. But not sure how a teacher can resolve this? What's a good next step?

jvonau, 2025-01-25:
> > Edit:
> > /var/run/usbmount/.mount.lock seems to be the issue
> > /var/run is a symlink to /run that is on a tmpfs and needs to be created first.
> 
> Yes, agree. But not sure how a teacher can resolve this? What's a good next step?

Just trying to think how something on a tmpfs survives a reboot, that is the whole point of using a tmpfs, the hierarchy of /run needs to be recreated upon boot. If you can reproduce this on demand the contents of /var/run/usbmount/ would be useful otherwise it is a boot time race between usbmount and what creates the /run hierarchy.

jvonau, 2025-01-25:
Thinking 'ConditionPathExists=' would be useful since the error is about /var/run and that is a link to /run. I think we might try ConditionPathExists=/var/run to ensure the rootfs is mounted first (in case of a fsck delay) and thus /var/run would be accessible.

jvonau, 2025-01-25:
The addition doesn't cause my machine any issues but I don't have the reproducer to see if that fixes the issue

avni, 2025-01-25:
> That would be used in the [Unit] section of the service file

Forgive my ignorance, which service file is this?

jvonau, 2025-01-25:
Forgive me how do you umount the key?

jvonau, 2025-01-25:
Think its a race, the main pid seems pretty low, lets turn on some additional logging.
Use sudo when editing /etc/usbmount/usbmount.conf
At the bottom of the file change to
>VERBOSE=yes

Now lets edit "/etc/systemd/system/usbmount@.service"
in the [unit] section add
>ConditionPathExists=/var/run

That should get rid of the mount failure but if not we have better logging.

`journalctl -b0 --no-pager | grep usbmount > ~/journal.txt` if you need to post something 

If you are doing a `umount /media/usbX` that is leaving a bunch of USB0|1|2|3 directories hanging around in local_content. Use `sudo systemctl stop usbmount@dev-<your device>.service` and  local_content gets cleaned up with 'systemctl stop' or a reboot.

jvonau, 2025-01-25:
Hold the phone, just bumped into this, fsck delayed the rw mount so /var/run can't be written to yet.

jvonau, 2025-01-25:
@avni just do the first part to enable the better logging. Question, are you usb booting your machine or have /library mounted to sda1?  

Sample of logging added when viewing systemctl status:
>Jan 25 09:47:39 box usbmount[568]: loaded usbmount configurations
Jan 25 09:47:39 box usbmount[582]: trying to acquire lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[597]: acquired lock /var/run/usbmount/.mount.lock

There are 2 possible causes, usbmount didn't create its /var/run directory or the lock timed out. 
The usbkey used below is an ubuntu installer key so there are 4 partitions so it's very tough and papers over one issue.
>journalctl -b0 --no-pager | grep usbmount 
Jan 25 09:47:38 box systemd[1]: Created slice system-usbmount.slice - Slice /system/usbmount.
Jan 25 09:47:39 box systemd[1]: Started usbmount@dev-sda.service.
Jan 25 09:47:39 box systemd[1]: Started usbmount@dev-sda1.service.
Jan 25 09:47:39 box systemd[1]: Started usbmount@dev-sda2.service.
Jan 25 09:47:39 box systemd[1]: Started usbmount@dev-sda3.service.
Jan 25 09:47:39 box systemd[1]: Started usbmount@dev-sda4.service.
Jan 25 09:47:39 box usbmount[568]: loaded usbmount configurations
Jan 25 09:47:39 box usbmount[570]: loaded usbmount configurations
Jan 25 09:47:39 box usbmount[566]: loaded usbmount configurations
Jan 25 09:47:39 box usbmount[569]: loaded usbmount configurations
Jan 25 09:47:39 box usbmount[567]: loaded usbmount configurations
Jan 25 09:47:39 box usbmount[582]: trying to acquire lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[579]: creating /var/run/usbmount directory
Jan 25 09:47:39 box usbmount[581]: trying to acquire lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[583]: trying to acquire lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[584]: creating /var/run/usbmount directory
Jan 25 09:47:39 box usbmount[588]: trying to acquire lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[591]: trying to acquire lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[597]: acquired lock /var/run/usbmount/.mount.lock
Jan 25 09:47:39 box usbmount[632]: dev/sda contains filesystem type iso9660
Jan 25 09:47:39 box usbmount[634]: usbmount execution finished
Jan 25 09:47:44 box usbmount[837]: acquired lock /var/run/usbmount/.mount.lock
Jan 25 09:47:44 box usbmount[865]: dev/sda1 contains filesystem type iso9660
Jan 25 09:47:44 box usbmount[868]: usbmount execution finished
Jan 25 09:47:54 box usbmount[1237]: acquired lock /var/run/usbmount/.mount.lock
Jan 25 09:47:54 box usbmount[1253]: dev/sda2 contains filesystem type vfat
Jan 25 09:47:54 box usbmount[1256]: mountpoint /media/usb0 is available for dev/sda2
Jan 25 09:47:54 box usbmount[1261]: executing command: mount -tvfat -osync,noexec,nodev,noatime,nodiratime,umask=0000 dev/sda2 /media/usb0
Jan 25 09:47:54 box usbmount[1269]: executing command: run-parts /etc/usbmount/mount.d
Jan 25 09:47:54 box usbmount[1299]: usbmount execution finished
Jan 25 09:48:09 box usbmount[1453]: acquired lock /var/run/usbmount/.mount.lock
Jan 25 09:48:09 box usbmount[1466]: dev/sda4 does not contain a filesystem or disklabel
Jan 25 09:48:09 box systemd[1]: usbmount@dev-sda4.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 09:48:09 box systemd[1]: usbmount@dev-sda4.service: Failed with result 'exit-code'.
Jan 25 09:48:29 box usbmount[1474]: acquired lock /var/run/usbmount/.mount.lock
Jan 25 09:48:29 box usbmount[1487]: dev/sda3 does not contain a filesystem or disklabel
Jan 25 09:48:29 box systemd[1]: usbmount@dev-sda3.service: Main process exited, code=exited, status=1/FAILURE
Jan 25 09:48:29 box systemd[1]: usbmount@dev-sda3.service: Failed with result 'exit-code'

code from usbmount
>if [ ! -e /var/run/usbmount ]; then
    mkdir -p /var/run/usbmount
    log debug "creating /var/run/usbmount directory"
fi
umask 022
if [ "$1" = add ]; then
   #Acquire lock.
    log debug "trying to acquire lock /var/run/usbmount/.mount.lock"
   #lockfile-create --use-pid --retry 5 /var/run/usbmount/.mount || \
    lockfile-create --retry 3 /var/run/usbmount/.mount || \
        { log err "cannot acquire lock /var/run/usbmount/.mount.lock"; exit 1; }
    trap '( lockfile-remove /var/run/usbmount/.mount )' 0
    log debug "acquired lock /var/run/usbmount/.mount.lock"

The code is called for all partitions plus the parent device, this is one failure point. In the logging above there are 2 notes of 
> creating /var/run/usbmount directory

This is the other failure point: 
>lockfile-create --retry 3 /var/run/usbmount/.mount 

notes are here: https://linux.die.net/man/1/lockfile-create