iiab/iiab #3907: Rename usb_lib_umask0000_for_kolibri and require var to be true for the usb_lib role.

avni, 2025-01-19:
- Rename usb_lib_umask0000_for_kolibri to umask0000_for_writable_sticks 
- make roles/usb_lib refuse to install if usb_lib_umask0000_for_writable_sticks: True has not been set.

@holta, assign this to me and I'll work on this after #3875 (upload2usb v0.1) is merged.

holta, 2025-01-23:
High priority & easy, now that PR #3875 is merged!

@avni the name `umask0000_for_writable_sticks` is fine.  If others prefer `usb_lib_writable_sticks` that's also OK.

holta, 2025-01-25:
> * make roles/usb_lib refuse to install if umask0000_for_writable_sticks: True has not been set.

I'd suggest maybe dropping that idea for now?

(While exceedingly rare, there might a small demographic who want read-only USB sticks...)

avni, 2025-01-25:
> (While not the norm, there might be a small demographic who legitimately want read-only USB sticks...)

Thanks, that's a valid pt. (Also makes it a lot easier to close this out).

avni, 2025-01-25:
### Test Cases

**Setup:** 
- curl iiab.io/install.txt | bash -s 3913
- Reboot server. 

**Test case 1:** 
- Go to: http://box/local_content/upload/ and test a couple of the test cases from here: [LINK](https://github.com/iiab/iiab/pull/3875#issuecomment-2599628892)

**Test case 2:** 
- Go to: http://box/kolibri/en/device/#/content/export_channels and export a small channel such as "_Using Studio: Your Content Workspace for Kolibri_" (~200MB) 
- Confirm that the KOLIBRI_DATA folder was created on your USB stick and has content from the channel in it.

EMG70, 2025-01-25:
Test case 1:

[image]
[image]

Test case 2:

[image]

[image]

[image]
[image]
[image]
[image]
[image]

avni, 2025-01-25:
Thx @EMG70 for testing! 🚀

closed 2025-01-31 by holta