iiab/iiab #3912: bogus country codes suppied by upstream

jvonau, 2025-01-24:
> @holta When are you opening issues around the bogus countries in raspi-imager and raspi-config that are not part of [wireless-regdb](https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/)'s [db.txt](https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/plain/db.txt)?
> 
> https://github.com/iiab/iiab/issues/3890#issuecomment-2601007839
> [real_clean.txt](https://github.com/user-attachments/files/18470294/real_clean.txt) Anything with a + are in iso  but not in db.txt.
> 
> https://github.com/iiab/iiab/issues/3890#issuecomment-2608227256 
> [imager-diff.txt](https://github.com/user-attachments/files/18511346/imager-diff.txt) Anything with a + are in raspi-imager's countries.txt but not in db.txt.
> 
> These are a bit of a problem as wpa_supplicant logging is getting spammed really bad and leaves 00 as an unset set country code. 

 _Originally posted by @jvonau in [#3890](https://github.com/iiab/iiab/issues/3890#issuecomment-2611256121)_

jvonau, 2025-01-24:
From 767bf738 snipped because it is overwhelming
>COMMAND: /usr/bin/sudo journalctl -b | grep wpa_supplicant    # wpa_supplicant log since boot
 Jan 23 11:36:21 iiab-test-cd systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 23 11:36:21 iiab-test-cd wpa_supplicant[532]: Successfully initialized wpa_supplicant
Jan 23 11:36:21 iiab-test-cd systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 23 11:36:23 iiab-test-cd wpa_supplicant[532]: nl80211: kernel reports: Registration to specific type not supported
Jan 23 11:36:23 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:29 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:29 iiab-test-cd wpa_supplicant[532]: wlan0: WPS-PBC-ACTIVE
Jan 23 11:36:29 iiab-test-cd wpa_supplicant[532]: wlan0: WPS-CANCEL
Jan 23 11:36:33 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:33 iiab-test-cd wpa_supplicant[532]: wlan0: Trying to associate with SSID 'SHAW-1F3C'
Jan 23 11:36:33 iiab-test-cd wpa_supplicant[532]: wlan0: Associated with 8c:76:3f:63:1f:40
Jan 23 11:36:33 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 23 11:36:33 iiab-test-cd wpa_supplicant[532]: wlan0: WPA: Key negotiation completed with 8c:76:3f:63:1f:40 [PTK=CCMP GTK=CCMP]
Jan 23 11:36:33 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-CONNECTED - Connection to 8c:76:3f:63:1f:40 completed [id=0 id_str=]
Jan 23 11:36:37 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:40 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:43 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:47 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:50 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:53 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:36:57 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:00 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:24 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:27 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:30 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:34 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:37 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:40 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:44 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:47 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:50 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:54 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:37:57 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:00 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:04 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:07 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:10 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:14 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:17 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:20 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:24 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:27 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:30 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:34 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:37 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:40 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:44 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:47 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:50 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:54 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:38:57 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan 23 11:39:00 iiab-test-cd wpa_supplicant[532]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
[... cut ...]
[   16.063953] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[ 1331.506722] warning: `cat' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[ 1474.122248] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

Related links for brcmf_set_channel: set chanspec XXXX fail, reason -52
https://github.com/RPi-Distro/firmware-nonfree/issues/38
https://forums.raspberrypi.com/viewtopic.php?t=376855
https://forums.raspberrypi.com/viewtopic.php?t=372396
https://github.com/raspberrypi/linux/issues/6049

jvonau, 2025-01-24:
Side note: Figure out what is altering cmdline.txt's country code automatically.

holta, 2025-01-24:
> From 767bf738

Background is https://paste.centos.org/view/767bf738 from:

- PR https://github.com/iiab/iiab/pull/3908#issuecomment-2610763668

jvonau, 2025-01-24:
Thoughts: powersave, roaming for my install. For nzola much the same but other issues might include a battery hat that might be installed.

holta, 2025-01-24:
> Thoughts: powersave, roaming for my install. For nzola much the same but other issues might include a battery hat that might be installed.

@nzola do your Raspberry Pis include a battery hat or anything similar?  Question regards your postings here:

- https://github.com/iiab/iiab/pull/3908#issuecomment-2608979981 (19h ago)
- https://github.com/iiab/iiab/pull/3908#issuecomment-2609888083 (12h ago)

holta, 2025-01-24:
@nzola clarifies that he's not using a "battery hat" here.

nzola, 2025-01-24:
> [@nzola](https://github.com/nzola) clarifies that he's not using a "battery hat" here.

Correct. No battery hat. Just a bare RPI4 board without enclosure.

jvonau, 2025-01-27:
#2975

jvonau, 2025-03-11:
Guess there is no interest in helping upstream sort out the mess that has been uncovered.

holta, 2025-03-11:
- Also keep in mind that the Raspberry Pi enterprise really struggles to keep up, even after their IPO which cannot solve everything... (e.g. Raspberry Pi OS's chromium-browser fork is often months out of date, lacking important security updates which very materially and very regrettably endangers many worldwide).  😟

- So it'd be really great if we articulated a specific and clear recommendation this year (or soon, if we get there, very clearly expressed as to what kind of simple fix would be genuinely efficient to implement?)  i.e. to help them better support folks in all countries — if this hopefully/eventually turns out to be possible... !

jvonau, 2025-03-11:
Upstream will never know of the issue uncovered if is not raised with them to investigate. Over to you again to take the lead.

holta, 2025-03-11:
I'd like to understand WiFi country code dynamics in RasPiO and similar distros much better in coming months, definitely if there is time:

- While rpi-imager is excellent software, it's also very apparent that the Raspberry Pi enterprise is struggling to maintain it, as the official release ([[1.8.5 currrently](https://www.raspberrypi.com/software/)) and the actual release when installing RasPiOS and apt updating (1.9.3) are far apart.

- So if in the end a few of us can (in fact) come up with a very concrete suggestion for them to implement in 2025, that'd be great indeed.

jvonau, 2025-03-11:
What more needs to be articulated? All the material needed is in the links provided above but in a nutshell use db.txt as the source for both raspi-config and raspi-imager then things might play nicer across the board going forward. Back over to you to push upstream.

holta, 2025-03-11:
> in a nutshell use db.txt as the source for both raspi-config and raspi-imager then things might play nicer across the board

That's a good start, but things like "country codes" affect so many different things... like keyboard setting, locale, time zone, , WiFi ETC.  So I wonder (perhaps wrongly) if rpi-imager's overall design might/possibly need to evolve a bit here.  Ensuring this is as pragmatic and self-contained as possible for rpi-imager's maintainers / devs:

- Just one example (to keeping things as simple as possible within rpi-imager) is a UX challenge that both @tim-moody and I noticed: rpi-imager only allows WiFi country code to be set if you're also setting an SSID.

- So I hope to improve my own understanding of several of these related challenges over coming weeks — to make sure any recommendation / bug report is genuinely very grounded & very clear/understandable to all (if a relatively "easy fix" is hopefully/really possible).

tim-moody, 2025-03-11:
We should use the term Regulatory Domain Code, not Country Code when speaking of Wi-Fi as these are different sets of codes. And while I agree that using db.txt for RegDom is better than locale, it is not perfect as the true test is the firmware, where, for example, HT is in db.txt, but not accepted by the firmware.

jvonau, 2025-03-11:
Looking forward to the testing of Ubuntu's server and desktop images coming from @holta as he works to expand his understanding of the concepts involved. I agree with Tim's understanding of the situation and the only way to clear the air from the smokescreen that was put forth by Adam is to file an upstream issue to get more eyes on the subject outside of the IIAB world and to solicit a response from those who know the firmware the best.

jvonau, 2025-03-11:
> Just one example (to keeping things as simple as possible within rpi-imager) is a UX challenge that both @tim-moody and I noticed: rpi-imager only allows WiFi country code to be set if you're also setting an SSID.

Agreed, the reason would be to unlock the WiFi to be able to do a headless deployment on a stage2 image over WiFi. In the case of wired headless install the regulatory domain doesn't matter much until the WiFi needs to be provisioned for use as an access point.

jvonau, 2025-03-15:
https://forums.raspberrypi.com/viewtopic.php?t=364908

jvonau, 2025-04-01:
https://github.com/raspberrypi/linux/issues/6669