iiab/iiab #3917: get-adm-conf failed iiab-cmdsrv failed to initialize

Dphilly-215, 2025-01-26:
I've installed iiab twice now using the pi imager. The first time I was able to download maps and change the default pw. After reboot I try to login to admin console locally from the pi and get this error get-adm-conf failed iiab-cmdsrv failed to initialize. Tried to restart the iiab cmdsrv service, it didn't help. Reinstalled using the pi imager again. Didn't do anything but change my location in pi settings from UK to US and again same error can't get into admin. On first boot I'm able to get into admin not sure why changing location breaks the ability to login. On my 3rd reinstall now not sure what the issue is. Plenty of space on SD, 128gb sd card.

holta, 2025-01-26:
Hi @Dphilly-215

Can you run `sudo iiab-diagnostics` and post the resulting URL here to help us diagnose?

tim-moody, 2025-01-26:
@Dphilly-215 you seem comfortable at the linux terminal. You could also help me diagnose what is happening by doing the following as root:

```
cd /opt/admin/cmdsrv/
systemctl stop iiab-cmdsrv
./iiab-cmdsrv3.py
```

You are now running cmdsrv manually and errors will not be trapped, so you should get more information about any error.

Please paste in any errors that appear on the terminal screen.

jvonau, 2025-01-26:
>Didn't do anything but change my location in pi settings from UK to US

How did you do that, with raspi-imager or with raspi-config, did you reboot afterwards?

Dphilly-215, 2025-01-26:
@jvonau i used the raspi config manager on pi image in preferences. I'm on my 3rd install and haven't touched anything in the pi config, no issues so far. The last install what I changed was location and wlan country. Changed wlan from unset to US. The other suggestions above I will try soon and provide a diag report. @tim-moody I'm doing another install on a different sd card so I don't mess up the work I've done already. I'll "break" this one and send information.

Dphilly-215, 2025-01-26:
[8.1_raspbian-11_2025-01-26_13%3A44%3A56_EST_dphilly.txt](https://github.com/user-attachments/files/18551256/8.1_raspbian-11_2025-01-26_13.3A44.3A56_EST_dphilly.txt)

Dphilly-215, 2025-01-26:
@tim-moody from how I understand this, changing the location changed the ISO type causing my issue?

root@box:/opt/admin/cmdsrv# systemctl stop iiab-cmdsrv
root@box:/opt/admin/cmdsrv# ./iiab-cmdsrv3.py
ERROR: Ansible requires the locale encoding to be UTF-8; Detected ISO8859-1.
Traceback (most recent call last):
  File "/opt/admin/cmdsrv/./iiab-cmdsrv3.py", line 3932, in <module>
    main()
  File "/opt/admin/cmdsrv/./iiab-cmdsrv3.py", line 188, in main
    init()
  File "/opt/admin/cmdsrv/./iiab-cmdsrv3.py", line 3376, in init
    get_ansible_facts()
  File "/opt/admin/cmdsrv/./iiab-cmdsrv3.py", line 3509, in get_ansible_facts
    outp = subprocess.check_output(args, shell=False, universal_newlines=True, encoding='utf8', env=ansible_env)
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/local/bin/ansible', 'localhost', '-i', '/opt/iiab/iiab/ansible_hosts', '-m', 'setup', '-o', '--connection=local']' returned non-zero exit status 1.
root@box:/opt/admin/cmdsrv# root@box:/opt/admin/cmdsrv# systemctl stop iiab-cmdsroot@box:/opt/admin/cmdsrv# root@box:/opt/admin/cmdsrv# systemctl stop iiab-cmdsroot@box:/opt/admin/cmdsrv# root@box:/opt/admin/cmdsrv# systemctl stop iiab-cmdsroot@box:/opt/admin/cmdsrv#

tim-moody, 2025-01-26:
@Dphilly-215 I think you  nailed it. I'm still not clear on what you changed and how.  Did you use raspi-config to change the wifi regulatory domain or some other means to change the locale?

tim-moody, 2025-01-26:
you could also run /usr/local/bin/ansible localhost -i /opt/iiab/iiab/ansible_hosts -m setup -o --connection=local to see if you get more error text

Dphilly-215, 2025-01-26:
@tim-moody I clicked the pi symbol in the top left, preferences, then raspberry pi configuration. from here I clicked on the last tab, "localisation", set locale. default was UK (GB) and character set UTF-8. When I changed to US it made the character set ISO8859-1. Then I had to reboot. after reboot the error occurred when logging into admin interface on IIAB. After your suggestion I saw the error, went back and changed character set back to UTF-8, another reboot and the issue is gone. I'm able to login as admin again. I appreciate everyone's help here!!

tim-moody, 2025-01-26:
>I clicked the pi symbol in the top left, preferences, then raspberry pi configuration

This is on the desktop? I notice in raspi-config the locale has en_US.UTF-8

tim-moody, 2025-01-26:
@holta I vaguely think we've seen this before. Should there be an faq?

jvonau, 2025-01-26:
raspi-config should be only selecting to use .UTF8 as the default choice for US, just like the default for GB is set.
That is 2 issues that have popped up in the last little bit using raspi-config.

holta, 2025-01-27:
@Dphilly-215

Can you say how relevant #3882 is/was in the end?  As @tim-moody and @jvonau alluded to above:

- `/etc/default/locale` must contain a `.UTF-8` language.
- If language(s) there do not end in `.UTF-8` (presumably because a ISO8859-1 language was chosen?) then extremely serious problems appear inevitable!
- Specific recovery instructions that can help: https://github.com/iiab/iiab/issues/3882#issuecomment-2579414146

holta, 2025-01-27:
@Dphilly-215 apologies you ended up installing an old IIAB image!  FYI you'll get a much better result if you:

1. Use [Raspberry Pi Imager](https://www.raspberrypi.com/software/) to install the very latest 64-bit Raspberry Pi OS.

   In other words, pick one of these 3 OS images, to install onto your microSD card: https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit

2. Boot your new Raspberry Pi OS (microSD card) in your Raspberry Pi 4.

3. Then install IIAB's latest software, by running the following at the Raspberry Pi OS command-line:
 
   `curl iiab.io/install.txt | bash`

Dphilly-215, 2025-01-27:
@holta I ran this cmd on my installs: rpi-imager --repo http://iiab.io/fast.json  I did actually try what you suggested during my testing. I did just a pi install to the SD, after booting up did apt-get update and upgrade. then ran curl iiab.io/install.txt | bash as per a guide I found. Install took a long time but once it finished iiab didnt work. I was unable to get the home page to come up. The only success I've had to get iiab to work was using the rpi-imager --repo http://iiab.io/fast.json from cmd. also after running this I was only presented with 3 image options, medium, small or medical. when I ran the curl iiab.io/install.txt | bash from a stock pi os install, there was a large option which is what I selected but again this failed so I went back to the rpi-imager --repo http://iiab.io/fast.json method. any suggestions on why it might have failed and Ill do another install and try again, but there where no errors at the end of the install everything looked to have installed ok. 

[image]

holta, 2025-01-28:
1. @Dphilly-215 you definitely want `curl iiab.io/install.txt | bash` on a clean Raspberry Pi OS.  In other words, I must apologize that `rpi-imager --repo http://iiab.io/fast.json` is NOT the way forward at this particular moment, as that will result in an archaic/ancient IIAB from 2023. :pray:

2. If anything goes wrong after you run `curl iiab.io/install.txt | bash` on a fresh/clean Raspberry Pi OS — please reply immediately with (1) what you expect to be working, (2) how precisely are you testing that — AND (3) the URL that results from you running `sudo iiab-diagnostics`

_Thank You!!_

PS send any screenshot/photo of any error message(s) too, please. 💯

Dphilly-215, 2025-01-29:
@holta I just did a fresh install of the 64bit pi OS and ran into the same issue where iiab does not work. If I got to http://box/home its blank. if I go to http://box/admin I get the error shown. The only way I've been able to get iiab to work is with that script rpi-imager --repo http://iiab.io/fast.json. I did see that in the about this is an old 2023 version. I would really like the most up to date version but this is the 2nd time that the curl iiab.io/install.txt | bash did not work for me. what am I doing wrong? I've rebooted a few times after install and set my wlan country to US. not sure what I'm missing.  my diag report is here: https://paste.centos.org/view/32f96c94  

[image]    

[image]   

[image]

holta, 2025-01-29:
_Thank You @Dphilly-215:_

1. Error seems to be:

> 2025-01-28 20:52:08,059 p=3442 u=root n=ansible INFO| TASK [jupyterhub : pip install 3 packages into virtual environment: /opt/iiab/jupyterhub (~326 MB total, after 2 Ansible calls)] ***
> 2025-01-28 20:53:14,628 p=3442 u=root n=ansible INFO| fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/opt/iiab/jupyterhub/bin/pip3", "install", "--no-cache-dir", "--prefer-binary", "pip", "wheel", "jupyterhub"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nRequirement already satisfied: pip in /opt/iiab/jupyterhub/lib/python3.11/site-packages (23.0.1)\nRequirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.38.4)\nCollecting jupyterhub\n  Downloading https://www.piwheels.org/simple/jupyterhub/jupyterhub-5.2.1-py3-none-any.whl (9.2 MB)\n     M-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^AM-bM-^TM-^A                        3.9/9.2 MB 75.2 kB/s eta 0:01:10\n\n:stderr: ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.\n    jupyterhub from https://www.piwheels.org/simple/jupyterhub/jupyterhub-5.2.1-py3-none-any.whl#sha256=ab6107e930a20d0f4cd76e8440dd8ff5872979c6761f02211dcba3f37166fc4e:\n        Expected sha256 ab6107e930a20d0f4cd76e8440dd8ff5872979c6761f02211dcba3f37166fc4e\n             Got        a824f8b8fdb1a4a9de8f46117a3207c58007b21a58320078786efc8265732eb7\n\n"}

2. Can you reply with the results of each of the 2 commands below?

```
sudo cat /etc/default/locale

sudo localectl
```

3. We'll look into it ASAP!

holta, 2025-01-29:
@Dphilly-215

I tried to reproduce the JupyterHub problem you ran into above, and it all works for me on Raspberry Pi 5.
Please send more details if you can, including output from the 3 `locale` commands above!

jvonau, 2025-01-29:
>Collecting jupyterhub\n  
Downloading https://www.piwheels.org/simple/jupyterhub/jupyterhub-5.2.1-py3-none-any.whl (9.2 MB)\n     ━━━━━━━━━━━━━━━━━                        3.9/9.2 MB 75.2 kB/s eta 0:01:10\n
\n
:stderr: ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.\n    
jupyterhub from https://www.piwheels.org/simple/jupyterhub/jupyterhub-5.2.1-py3-none-any.whl#sha256=ab6107e930a20d0f4cd76e8440dd8ff5872979c6761f02211dcba3f37166fc4e:\n        
Expected sha256 ab6107e930a20d0f4cd76e8440dd8ff5872979c6761f02211dcba3f37166fc4e\n
             Got        a824f8b8fdb1a4a9de8f46117a3207c58007b21a58320078786efc8265732eb7\n
\n"}

I would suggest being behind a proxy might be an issue but having a look at
https://www.piwheels.org/project/jupyterhub/ and the failure log
https://www.piwheels.org/logs/0000/1258/4972.txt I'm surprised it worked for him.

holta, 2025-01-29:
Just in case this comparison helps, here's my iiab-diagnostics showing the fresh/new install of JupyterHub on Raspberry Pi 5, that worked earlier this morning: https://paste.centos.org/view/cbcd3337

jvonau, 2025-01-29:
Just did a quick install no error
`cd /opt/iiab/jupyterhub`
`source bin/activate`
`bin/jupyterhub --version`
5.2.1

holta, 2025-01-29:
> Just did a quick install no error `cd /opt/iiab/jupyterhub` `source bin/activate` `bin/jupyterhub --version` 5.2.1

@jvonau's above test is likely on Raspberry Pi 400 much like @Dphilly-215's RPi 4?

Either way, JupyterHub 5.2.1 somehow miraculously (despite piwheels.org drama!) installed successfully for me too, on RPi 5 here:

```
root@raspberrypi:~# /opt/iiab/jupyterhub/bin/jupyterhub --version
5.2.1
root@raspberrypi:~# python3 --version
Python 3.11.2
```

jvonau, 2025-01-29:
Rpi 4

Dphilly-215, 2025-01-29:
@jvonau & @holta This install was done on a pi4. I have a pi 5 too but using it for something else right now so did the install on a 4. After I wrote my last message I was trying to figure out what might have happened. I remembered in the beginning of my install there was a message that said, if anything fails run this cmd sudo iiab. While I didnt see any failures after my first install I figured, I cant break it anymore so why not try it. The cmd ran for a while but once it finished I rebooted and now everything is working! you guys would know better than me what that cmd did but it fixed whatever was happening after the first install.

holta, 2025-01-29:
@Dphilly-215, can you send us the output from these 3 commands just to be sure?

```
sudo cat /etc/default/locale

sudo localectl

sudo locale -a
```

holta, 2025-01-29:
> I remembered in the beginning of my install there was a message that said, if anything fails run this cmd sudo iiab. While I didnt see any failures after my first install I figured, I cant break it [...]

Ok!  If it happens again, @Dphilly-215 please do try to help us understand the pattern / cause! 🔎 🔍

Dphilly-215, 2025-01-29:
@holta Will do! happy to help however I can. this is a really cool project. I appreciate the response and help from everyone, ty so much!

holta, 2025-02-04:
> Should there be an faq?

3 or 4 people have now run into this:

- So [draft FAQ NUMBER 12 (question and answer)](https://wiki.iiab.io/go/FAQ#What_to_do_if_Raspberry_Pi_OS_messes_up_language_locale%3F) is now posted.
- The situation undoubtedly evolves in coming years, if not earlier if Raspberry Pi OS dedicates time to cleaning up their locale UX (graphical widgets and raspi-config too !)

tim-moody, 2025-03-11:
this is a raspi-os problem, not iiab