iiab/iiab #3921: bug: permission denied when copying from /tmp to USB stick mount point on RPi Desktop

avni, 2025-01-27:
🐞 As part of #3875, we missed testing the upload2usb app on RPi OS Desktop. From a test this weekend, the USB stick is mounted differently and we get a permission denied error when uploading a file. The error triggers when copying from /tmp to the USB stick mount point. 

The mount point is owned by iiab-admin on Desktop, unlike on Headless, where it is owned by root. ⛑️

holta, 2025-01-28:
RELATED: Uploading into /tmp is also a bug (causing permanent damage to IIAB microSD cards, when the real destination that matters is the USB stick!)  But perhaps we should open a separate ticket to solve that "architectural flaw" ?

Tangentially related:

- https://github.com/iiab/iiab/issues/3403#issuecomment-2615004263

jvonau, 2025-01-28:
Seen before #3105 #2456

jvonau, 2025-01-28:
`lsblk` it will be mounted twice once by usbmount in the correct place and once by the desktop, last mount wins the permissions race.

jvonau, 2025-01-28:
The same could hold true for a usb booted machine also

jvonau, 2025-01-28:
Not a BUG, it's a lack of a user disabling the auto-mount function in the desktop. After all we build servers and that is what the design is geared towards. Think the desktop has changed underneath so one must fiddle around and document the steps needed.

avni, 2025-01-28:
> it will be mounted twice once by usbmount in the correct place and once by the desktop, last mount wins the permissions race.

Thanks @jvonau That's exactly the behavior. USB stick is mounted 2x. Updated issue indicate it's not a bug. 

> It's a lack of a user disabling the auto-mount function in the desktop. 

I'll test to confirm this and we can documentation the top of usb_lib/tasks/install.yml

Thank you!

avni, 2025-01-28:
In the RPi Desktop OS File system preferences pane (see screenshot), "Volume Management", unselecting the second option ("") under auto-mount makes it so the volume is only mounted once! The upload2usb app then works and we no longer get the "permission denied" error.

holta, 2025-01-28:
> In the RPi Desktop OS File system Preferences pane (see screenshot), in "Volume Management", unselecting the second option ("Mount removable media automatically when they are inserted") under auto-mount makes it so the volume is only mounted once! The upload2usb app then works and we no longer get the "permission denied" error.

1) Thank for confirming.  I mentioned this Saturday, but now the big picture should be more clear. (:

2) @avni can you please investigate automating detection of this troublesome automount setting?  IIAB displays 1-or-2-or-3 graphical pop-ups right after "RasPiOS with desktop" boots — to get implementers on track, to clean up their IIAB installation in the face of serious/common mistakes.  Arguably Chromium (browser) is one of those pop-ups!

   In any case, we should strongly consider adding a 3rd or 4th.  It's very easy to do, in coming weeks if you have time?

   CLARIF: Run `grep -r labwc` within `/opt/iiab/iiab` to see details of how-and-why these pop-ups appear.

avni, 2025-01-28:
1. You did! The screenshot you shared helped me find the setting once I understood better what was going on. I should have mentioned that above.

2. Yes, will look. Thx for the pointer.

holta, 2025-04-07:
@avni would you have time to work on AUTO-DETECTION of this misconfiguration on "RasPiOS with desktop" in the coming month?

TBD:
- Once auto-detection of the automount misconfiguration is in place, there's a quite strong argument that this misconfiguration should also be auto-fixed — e.g. whenever IIAB is install with roles/usb_lib — or if `./runrole --reinstall usb_lib` is later run after the fact.
- Particularly as there are already way too many pop-ups (three so far) that IIAB can display after "RasPiOS with desktop" is rebooted.  UX 101: Absolutely nobody wants 4 pop-ups immediately after booting — these become visual pollution that will be ignored, accomplishing nothing.

holta, 2025-04-07:
COROLLARY: A bit more than "legalese" would be appropriate, so that everyone installing IIAB onto RasPiOS with desktop... as least has the opportunity to realize that the OS's automount default option is being turned off... i.e. documenting this loudly in the most common {small, medium, large} local_vars.yml files [in the usb_lib section](https://github.com/iiab/iiab/blob/61310b8aafa3fc2c4b57d86b6ce605a67b1aa2f6/vars/local_vars_medium.yml#L195-L197) at the minimum.

OBVIOUS, but it still needs to be state: 90% of IIAB implementers won't read it, but this kind of lower-level technical nitty-gritty likely belongs in {small, medium, large} local_vars.yml files more than in than in higher-level docs like FAQ.IIAB.IO — with as many hyperlinks as are feasible to make this/these flood of implementation details more discoverable ;)