iiab/iiab #3924: upload2usb: Writing N times to /tmp as part of file upload (before copying to USB stick) damages SD card

avni, 2025-01-28:
HTML form file upload is writing to /tmp and then copying to final destination on USB stick. This will damage SD cards (on Pis) over time with the numerous writes. 

PHP docs indicate that we can change tmp dir to another location: https://www.php.net/manual/en/ini.core.php#ini.upload-tmp-dir

**From post by @holta in [#3921](https://github.com/iiab/iiab/issues/3921#issuecomment-2617479517)_**

> Uploading into /tmp is also a bug (causing permanent damage to IIAB microSD cards, when the real destination that matters is the USB stick!)  But perhaps we should open a separate ticket to solve that "architectural flaw" ?
> 
> Tangentially related:
> 
> - https://github.com/iiab/iiab/issues/3403#issuecomment-2615004263 

RELATED: 
- #3875 
- #3923
- #3919 
- #3915

jvonau, 2025-01-28:
Might just want to set the tmp dir to a scratch directory on the usbkey

In the getTargetUSBDriveLocation function is really too much calling lsblk, all you really just need is to scan /library/www/html/local_content/ to get USBX to filter out the usb boot root directory as that filtering is already done in roles/usb_lib/templates/mount.d/70-usb-library.j2

tim-moody, 2025-01-28:
>Might just want to set the tmp dir to a scratch directory on the usbkey

good idea, and makes a mv easier.

also we have run out of space on /tmp on some occasions.

holta, 2025-01-28:
@avni feel free too "unassign" yourself.  If you don't have time to work on this ticket.  Thanks!

jvonau, 2025-01-28:
If you are interested in reducing the writes on the sdcards then /tmp could be a tmpfs but that limits the size to the amount of ram available. That change would have to be very early in the boot process but I just tested /etc/fstab
> ls /tmp
admin-console.sock
systemd-private-ceb23b05cae143da8a6a0bb12585db4d-ModemManager.service-j2AXHi
systemd-private-ceb23b05cae143da8a6a0bb12585db4d-systemd-logind.service-dmnxYr
systemd-private-ceb23b05cae143da8a6a0bb12585db4d-systemd-resolved.service-9uNDx6
systemd-private-ceb23b05cae143da8a6a0bb12585db4d-systemd-timesyncd.service-tkZjdt

https://www.reddit.com/r/Ubuntu/comments/17698i2/tmp_as_tmpfs_vs_tmp_as_directory_over_zfs/
https://wiki.centos.org/TipsAndTricks(2f)TmpOnTmpfs.html

jvonau, 2025-01-28:
If you really want to make the sdcard last longer get rid of the system logging & analytics, anything that writes to the filesystem post install after content provisioning. Think like a read only root filesystem is the goal.

holta, 2025-01-28:
That might not be the case.  FYI:

- Many schools in Mexico have demonstrated very convincingly to me that writing large amounts of data (e.g. very large files) is what damages (effectively destroys) microSD cards.

- Whereas writing very small files (e.g. logging, routine OS writes to the main disk, etc) has definitely not been a problem so far, over many years to date.

holta, 2025-09-17:
@chapmanjacobd says he might be able to work on this:

@Ark74 I might give you a quick call, to see if you can help guide along the way?!

holta, 2025-09-17:
@chapmanjacobd @Ark74 let's please consider the imminent Raspberry Pi OS 13 "Trixie" as our baseline going forward here:

- #4058
- https://downloads.raspberrypi.com/nightlies/

_i.e. just in case the systemd version or other particulars matter here, in coming up with a "sane enough" fix (:_

jvonau, 2025-09-20:
https://github.com/iiab/iiab/pull/4086#issuecomment-3315034157