iiab/iiab #3933: Every menu link gives Undefined Zim

pixel5, 2025-02-03:
I have downloaded a large amount of content, almost entirely zims. No matter what I add to the content menu, every link brings up the Undefined Zim page. I have done my due diligence and searched for a solution, but I'm unable to resolve my issue. The following information might be useful when discussing this issue:

- Hardware: Raspberry Pi 3B+, external 500 GB Samsung EVO 860
- All zims are accounted for in /library/zims/content/
- I have run `iiab-make-kiwix-lib` a dozen times with no apparent effect, but get the following output every time:

 ```Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Cannot add zim /library/zims/content/devdocs_en_vue_2025-01.zim to the library.
Cannot add zim /library/zims/content/libretexts.org_en_bio_2025-01.zim to the library.
Cannot add zim /library/zims/content/wikivoyage_en_all_maxi_2024-10.zim to the library.
Cannot add zim /library/zims/content/wiktionary_en_all_nopic_2024-05.zim to the library.
Writing zim_versions_idx
Zim files and library.xml are out of sync. Please run iiab-make-kiwix-lib.
Finished making Kiwix library.xml
```
- I have also run `/usr/bin/iiab-make-kiwix-lib.py -f --no_tmp` with no effect on the issue
- The menu file(s) have not been manually altered in any way, nor has any other file
- There are no errors in command logs or in the nginx error.log

I am unsure of what to do at this point. Any help is appreciated.

tim-moody, 2025-02-03:
>Zim files and library.xml are out of sync

/library/www/html/common/assets/zim_version_idx.json maps the generic name of a zim to the actual one installed, such as 
wiktionary_en_all_nopic to wiktionary_en_all_nopic_2024-05.zim

If the menu system does not find an entry in that file you will get the undefined error message. It sounds like zim version is empty.

This is probably an error in python, and I'm surprised you don't see that when running `iiab-make-kiwix-lib`. We need to discover that error.

You could start by moving the 4 zims that cannot be added to another directory and rerunning `iiab-make-kiwix-lib` to see if that helps.

pixel5, 2025-02-03:
Hello @tim-moody, thanks for responding. All of the zims have that naming scheme, a few examples from my server: 

```nih_rarediseases_en_all_maxi_2020-12.zim
opensource.stackexchange.com_en_all_2024-11.zim
pets.stackexchange.com_en_all_2024-11.zim
raspberrypi.stackexchange.com_en_all_2024-11.zim
urban-prepper_en_all_2024-12.zim
wikem_en_all_maxi_2021-02.zim```

I moved those 4 zims into a directory outside of IIAB and re-ran `iiab-make-kiwix-lib`, all of the links are now working, which is great!

I would, however, like to get those four to work as well... is there a way I could manually fix those?

tim-moody, 2025-02-03:
just did a download of wiktionary_en_all_nopic_2024-05.zim onto a test server using Admin Console.

It an all other zims are on the menu. I did have to refresh the browser to get the new one not to report undefined.

could you try moving that zim back into zims/content running make kiwix lib

pixel5, 2025-02-04:
Apologies for the delayed response. I just moved those four zims back and re-ran, gave the same error as before. I should note, the menu links for everything else are still working and haven't been re-broken by this action.