iiab/iiab #3934: Test on NM AP with no IIAB with stock standard firmware

tim-moody, 2025-02-05:


tim-moody, 2025-02-05:
Raspberry Pi: 5
OS: 2024-11-19-raspios-bookworm-arm64-lite.img
apt full upgrade: Yes
IIAB Installed: No
raspi-config nonint do_wifi_country: No

- Use 8 connection firmware
/etc/alternatives/cyfmac43455-sdio.bin -> /lib/firmware/cypress/cyfmac43455-sdio-standard.bin
 
- Create hotspot with NetworkManager:

```
rfkill unblock wifi
nmcli radio wifi on
nmcli connection add type wifi ifname wlan0 con-name local-ap autoconnect yes ssid local-ap
nmcli connection modify local-apt 802-11-wireless.mode ap 802-11-wireless.band bg ipv4.method shared
nmcli connection up local-ap

```

As expected this configuration supports a **maximum of 8 wifi connections**.

holta, 2025-02-05:
> As expected this configuration supports a **maximum of 8 wifi connections**.

And hopefully the WiFi firmware doesn't crash when a 9th "student" tries to connect? :pray:

(As used to occur in 2020!)

Related:

- #823

tim-moody, 2025-02-05:
>And hopefully the WiFi firmware doesn't crash when a 9th "student" tries to connect?

My laptop was the 9th and was simply unable to connect.