iiab/iiab #3935: Test on NM AP with no IIAB with stock minimal firmware

tim-moody, 2025-02-05:


tim-moody, 2025-02-05:
Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:12 version 7.45.241 (1a2f2fa CY) FWID 01-5209c09

holta, 2025-02-05:
> cyfmac43455-sdio-minimal.bin

> Version corresponds to IIAB rpi3bplus_rpi4_wifi_firmware: os setting.

Tiny correction: Above corresponds to [rpi3bplus_rpi4_wifi_firmware: 19](https://github.com/iiab/iiab/blob/f08478545256e08e0b5842b445d00f17a473bdfa/roles/firmware/tasks/install.yml#L19-L21) (arranged by pelwell on #2853).

FYI Raspberry Pi OS continues to offer these exact same 2 files for the 43455 Wi-Fi chip since 2021: (i.e. identical contents in 2025; don't worry about Raspberry Pi OS having changed both paths/filenames over recent years)

- https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/cypress/cyfmac43455-sdio-minimal.bin (548728 bytes)
- https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/cypress/cyfmac43455-sdio.clm_blob (2676 bytes)

(Which presumably means RPi 5 and RPi 500 can handle 19-or-20 "student" connections using this particular W-Fi firmware — just like with RPi 3B+ and RPi 4.)

holta, 2025-02-05:
Every bit is identical in the 2 files that matter — nothing has changed from 2021 to 2025.

> doesn't seem to be exactly the same version as this one is 7.45.241.

I'm not quite understanding what you are saying.  The minimal firmware produced in Q4 2021 remains 7.45.241 (literally nothing has changed).

> but the correspondence with '19' is not exact either

- 19 is very precise from a teacher's perspective, running the default setup for this "minimal" WiFi firmware.
- Technical gurus however, who prefer to set `wifi_up_down: False` can of course claim 20 is God's Truth, if they prefer, no problem (-:

tim-moody, 2025-02-05:
Perhaps I'm confused in thinking that os and 19 are different in IIAB.

tim-moody, 2025-02-06:
@holta Please confirm:
- setting rpi3bplus_rpi4_wifi_firmwares to either os or 19 will result in making brcmfmac43455-sdio.bin_2021-11-30_minimal the wifi firmware.

I agree that brcmfmac43455-sdio.bin_2021-11-30_minimal is identical bit for bit with currently shipping cyfmac43455-sdio-minimal.bin (7.45.241).

I was expecting os to set the firmware to cyfmac43455-sdio-standard.bin or some historical equivalent.

holta, 2025-02-06:
Setting `os` uses whatever Wi-Fi firmware was shipped by the OS.

Which at the moment (February 2025) presumably corresponds to 8 "student" connections, thanks to your 43455 testing of Raspberry Pi OS on #3934.

tim-moody, 2025-02-06:
The comment to line 14 above says 'os' installs 7.45.241, which is minimal.

holta, 2025-02-06:
Raspberry Pi OS changes regularly — every year or two (sometimes more often) its "standard" Wi-Fi firmwares evolve:

The point of the `os` setting is exactly that (to use the Wi-Fi standard/stock/default/generic firmware of the underlying OS) at the moment in time when IIAB was installed.

ASIDE: OS firmware is backed up to 2 .orig files during IIAB installation, in order to protect the IIAB implementer, allowing them BOTH options if in future the OS updates itself (e.g. as a result of apt updates).

jvonau, 2025-02-07:
There are two parts to the firmware, the bin file and the blob file. Should check for clm_blob differences also. Perhaps the version 7.45.265 is just an updated blob file.