iiab/iiab #3938: Any recommendations for reducing logging on Pi zero 2W ?

PeterTGarner, 2025-02-08:
**Raspberry Pi 2W, Debian 12 Lite (no desktop), IIAB 8.3**

I'm running IIAB 8.3 for myself and a couple of family members, so I don't need most of the logging. I'm keen to reduce the number of writes to the SD card. I've already disabled kalite.service and jupyterhub.service using systemctl. I don't need any statistics on what's been accessed, etc. Are there any recommendations for what else I could potentially disable to save writes and memory?

jvonau, 2025-02-11:
Glad somebody else brought this to the forefront, thought this was in the Wiki somewhere. Ill let @holta assume the lead response but I have a couple of pointers for you if you wish. One problem with using systemctl to disable the service directly is local_vars file will re-enable the service should there be an ansible run at some point in future. It would be best to edit local_vars.yml and set *_enabled to False for that service, then run one of `sudo ./ runrole service name` ,`sudo ./iiab-configure` 

Or by changing the services in admin-console by pressing on 'Configuration'  ->  'Services Enabled'  then unchecking the box(es). Next press the 'Save Configuration' button which saves the changes to local_vars for you and then press 'Install Configured Options' to apply the changes which will run ansible for you.

holta, 2025-02-11:
Thanks @PeterTGarner for writing.

- Start with @jvonau suggestions above.

- We need a better answer here in 2025, especially as the [Upload2USB](https://wiki.iiab.io/go/FAQ#Can_students_upload_their_own_work?) feature becomes more widely adopted.

Just FYI, microSD cards in Mexico schools show this multi-year pattern so far:

- IIAB microSD cards fail after too many very-large-scale write operations (writing of too many very large files, or dd'ing to rewrite to the entire microSD card a few years later).

- On the bright side, IIAB microSD cards continue to work for years (more than expected) across many reboots, in some very harsh field conditions.  As if most logging and log files do not appear to be the most serious threat to the health of SD cards.

(In any case, we need more data points in 2025.  And a better understanding of which logging is truly excessive, as even solid-state SSD drives with "real MTBF warranties" fail prematurely on occasion!)

As we all try to dig deeper here (hard-nosed testing of microSD cards would be very welcome!) please see these 2 related questions:

- https://github.com/iiab/iiab/pull/3875#issuecomment-2571350116
- https://github.com/iiab/iiab/issues/3924

jvonau, 2025-03-04:
> **Raspberry Pi 2W, Debian 12 Lite (no desktop), IIAB 8.3**
> 
> I'm running IIAB 8.3 for myself and a couple of family members, so I don't need most of the logging. I'm keen to reduce the number of writes to the SD card.  

> Are there any recommendations for what else I could potentially disable to save writes and memory?

@PeterTGarner Back in the [XO](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/xo.yml.unused) days /var/log was in a tmpfs and iiab undid that configuration so we could monitor the older logs after a reboot. You could try to implement that with a entry in 
/etc/fstab for /var/log see https://wiki.archlinux.org/title/Tmpfs

tim-moody, 2025-03-04:
From what I read you can rewrite every sector of an sdcard every day for 10 years without an issue. The logs are probably < 1% of the total. IIAB is mostly static.

jvonau, 2025-03-04:
One might want to use /var/log on a tmpfs for some other reasons:

- To prevent filling the 'drive' to the point of not being able to login on a shell
- For privacy reasons to suppress the recording of unique identifiers of clients in hostile governmental areas. One example would be the macaddresses of wifi clients

`journalctl -b --no-pager`
>Mar 04 13:30:48 box dnsmasq-dhcp[1159]: DHCPDISCOVER(br0) xx:xx:xx:xx:xx:xx
Mar 04 13:30:48 box dnsmasq-dhcp[1159]: DHCPOFFER(br0) 10.10.10.206 xx:xx:xx:xx:xx:xx
Mar 04 13:30:48 box dnsmasq-dhcp[1159]: DHCPDISCOVER(br0) xx:xx:xx:xx:xx:xx
Mar 04 13:30:48 box dnsmasq-dhcp[1159]: DHCPOFFER(br0) 10.10.10.206 xx:xx:xx:xx:xx:xx