iiab/iiab #3956: I enabled MFA for SSH which is working, then changed the password on the IIAB console and the new password works for SSH with the MFA token but can't login to the IIAB console

giglioroninomicon, 2025-02-22:
Hi there,

I enabled MFA for SSH which is working, then changed the password on the IIAB console and the new password works for SSH with the MFA token, but I can't login to the IIAB console now.

I'm not sure how the IIAB console login works, if it sends the authentication request to the system, how would I enter the MFA token? Or is the MFA only enforced for the SSH daemon?

Any help is appreciated!
Thanks

holta, 2025-02-22:
Hi @giglioroninomicon 

Which kind of MFA did you enable?

giglioroninomicon, 2025-02-22:
I followed this guide and am using the password based auth. https://www.linuxbabe.com/ubuntu/two-factor-authentication-ssh-key-ubuntu

giglioroninomicon, 2025-02-22:
And I'm running on 
Linux box 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

giglioroninomicon, 2025-02-22:
Quick update - I am in a good place with my install, I updated what I needed via SSH cli. You can take your time on this request or add it as a feature enhancement. Appreciate what you have built here, it is really cool.

tim-moody, 2025-02-23:
Admin Console does not support MFA. It authenticates by computing a hash for the entered password and comparing to /etc/shadow. If adding MFA to SSH changed the hash algorithm then the password might no longer work. The line in /etc/shadow (at least the salt and algorithm part) would need to be examined to say more.

journalctl or iiab-diagnostics could help.