iiab/iiab #3959: Ran iiab-update, now admin page fails to load

1n5aN1aC, 2025-03-02:
The iiab admin page was working, but I was having troubles with Calibre-web.

I updated iiab by updating the base system then running `iiab-update` which appears to have fixed my issues with calibre-web, however, now the admin console does not load.

Before I even have a chance to put in a username or password, a browser modal box pops-up with the following:

```Connection to Server failed.
 Please make sure your network settings are correct,
 that the server is turned on,
 and that the web server is running.
```

System:  Debian Trixie   (`Linux internetinabox 6.12.12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.12-1 (2025-02-02) x86_64 GNU/Linux`)

1n5aN1aC, 2025-03-02:
Appears I posted a little too hastily, as I believe I have solved this.

I tracked the errors down via nginx's /var/log/nginx/error.log:
`2025/03/01 19:25:59 [crit] 1663#1663: *105 connect() to unix:/run/php/php8.2-fpm.sock failed (2: No such file or directory) while connecting to upstream, client: 10.0.0.x, server: internetinabox, request: "GET /admin/server-info.php?_=1740885957786 HTTP/1.1", upstream: "fastcgi://unix:/run/php/php8.2-fpm.sock:", host: "10.0.0.x", referrer: "http://10.0.0.x/admin/"`

Upon investigating my system, I appear to now have php8.4 installed, not php8.2.

I manually edited `/etc/nginx/nginx.conf`, and edited the line near the bottom:  
`server unix:/run/php/php8.2-fpm.sock;`  
changing it to:  
`server unix:/run/php/php8.4-fpm.sock;`

After running `service nginx restart` the admin panel works now, and it appears everything else is working.  It does make me worry I may have hosed something on my install, and may need to re-install from scratch at some point, but everything seems good for now....

holta, 2025-03-02:
Thank.you @1n5aN1aC for posting. 🙏

As you know Debian 13 Trixie is not yet supported (and [iiab-update](https://wiki.iiab.io/go/FAQ#Can_I_upgrade_IIAB_software?) is never fully guaranteed) — but at the same time these are great questions to begin probing as Raspberry Pi OS 13 / Trixie approaches later this year !

1n5aN1aC, 2025-03-02:
@holta Yeah, thanks.  Everything seems to still be good to go,

I've been running Debian Trixie for a fair bit of time, I think around the same time as this guy, #3736 and haven't had any problems at all, except those caused by periodic apt updates, like this one.

I think it should be pretty close to good-to-go once Trixie becomes stable, and doesn't break things like this.  :)

Anyways, I would consider this issue 'resolved', and would be glad to have it closed unless you wanted to leave it open as a reminder of what may be required before official support of Debian 13 Trixie.

holta, 2025-03-02:
Great news:

Thanks again @1n5aN1aC for confirming!

jvonau, 2025-03-03:
Seems like this situation was based on older IIAB install that was updated to trixie as [nginx.conf](https://github.com/iiab/iiab/blob/master/roles/nginx/templates/nginx.conf.j2) has a {{ php_version }} which is a [detected value](https://github.com/iiab/iiab/blob/master/scripts/local_facts.fact#L24). Reinstalling the nginx role would fix that up.

1n5aN1aC, 2025-03-03:
@jvonau just a note, this install was NOT previously upgraded from an older Debian version, though it has been around on Trixie for a decent chunk of time.  I set it up probably around a year ago, directly on Trixie.

Regardless, it appears Trixie is only a few months from freezes, at which point major package upgrades should become pretty rare.

Thanks for the point about reinstalling the nginx role.  I should probably give that a go to make sure I don't have any other cruft that isn't quite working correctly.

jvonau, 2025-03-03:
> This install was NOT previously upgraded from an older Debian version, though it has been around on Trixie for a decent chunk of time. I set it up probably around a year ago, directly on Trixie. [...]

@1n5aN1aC Have a look at /var/log/apt/history.log or the older compressed log files that would show when php moved from 8.2 to 8.4. I suspect your year old install may have other issues. Just remember the install is a one hit wonder and re-installs are not officially supported despite my best efforts to be able to support rolling upgrades of both the OS and [IIAB](https://github.com/iiab/iiab/pull/2500)

jvonau, 2025-03-03:
https://github.com/iiab/iiab/pull/2771

tim-moody, 2025-03-11:
resolved by reinstalling nginx

closed 2025-03-11 by tim-moody