iiab/iiab #3960: Looking for architecture advice - single network Tiny computer and a GLi MT300A travel router

giglioroninomicon, 2025-03-04:
Hi all,
I have a working setup I have been building out. I have all the content I want on the IIAB device (Lenovo M93p Tiny) which has a single wired nic. I am front ending that with a GLi MT300A travel router running OpenWRT latest version. 

Everything works great if you know to go to http://box.lan/home - Since I want this to be a publicly accessible device and it won't have an attendant, what is the best way to send anyone that connects to the "Internet In A Box" SSID to the proper URL? 

I've been looking at possibly doing a captive portal where I could put the URL in the splash page but wonder if there is a more straightforward way to do this.

Thanks in advance!

tim-moody, 2025-03-04:
IIAB has a captive portal that in my view works quite well with the built in hotspot. I suggest you install it with the Admin Console (Configure -> Internal Wi-Fi AP) and test it with the external Wi-Fi adapter. I have never tried this, so would be interested in the results.

giglioroninomicon, 2025-03-04:
Thanks Tim, I'll check it out. Part of my design goal is to install the M93p Tiny indoors and then have an outdoor AP mounted up high that is fed by 50 ft of Ethernet cable with POE for maximum reach.

jvonau, 2025-03-05:
Could add the Ethernet connection under the LAN bridge (br0), things should work the same as accessing the IIAB box via WiFi.

giglioroninomicon, 2025-03-05:
I'll check that out, thanks!

giglioroninomicon, 2025-03-09:
Hi all,
I've been playing with things and am at the point where I need to enable DHCP/DNS and also the captive portal on my device. 

My device has a single nic and is currently in Appliance mode. Until now I have been using a small travel router as my DHCP/DNS server but wish to simplify my configuration for stability reasons. 

I have all the content I need for now so I don't need to hook things up to the Internet. 

What I'd like to do is plug the wifi AP device into the LAN wired port on the device. I will use a small ethernet switch for this. 

The AP derives it's DHCP address from the network and I'd like that to be the IIAB device.

I'd also like the captive portal or some mechanism so when a user connects to the AP via wifi they automatically go to the homepage on the IIAB device.

I've read through some of the networking FAQs but don't have a clear enough picture in my head how to get where I want to be.

For my desired configuration do I need to be in Appliance or LAN Controller mode?

Which config file do I update to enable dnsmasq to be the DHCP/DNS server?

For jvonau's comment, what file do I edit to add the ethernet connection under the LAN bridge? 

Sorry for the long winded update, I'm close and don't want to blow anything up as I have almost 500GB of content downloaded and working currently. Ultimately I just need users that connect to be taken to the box.lan url once they are connected via the external wifi AP that is connected to the ethernet connection on the device.

Thanks and I appreciate everyone's comments/feedback!

tim-moody, 2025-03-09:
@jvonau will no doubt correct me if I am wrong, but your architecture sounds like what we call Lan Controller. Your external AP would then just be pass through. (plug the ethernet cable into the lan side not the wan side so it acts like a bridge and turn off dhcp/dns) There have been a number of users who have operated this way. The AP IP can also be hard coded as it is acting as a bridge, though that makes for some difficulty with configuration.

The changes to IIAB can be done through the Admin Console and should not require editing of files.

giglioroninomicon, 2025-03-09:
Thanks Tim, when I try just changing to Lan Controller via the Admin console, Save the config, Apply the config, I don't get the DHCP/DNS server and the device just waits for a DHCP server to give it an address. Do I need to unplug everything and rerun sudo iiab-network so nothing is detected and the script reinstates the DHCP/DNS server etc?

tim-moody, 2025-03-09:
> I don't get the DHCP/DNS server

a reboot can't hurt.

do other devices get an ip? probably the AP does not request an IP on its lan port.

giglioroninomicon, 2025-03-09:
Hi all,
So I reran this advice with my unit unhooked from any networks.

A stronger version is also available if necessary: cd /opt/iiab/iiab then sudo ./runrole --reinstall network

It ran through and modified a bunch of stuff but failed on starting dnsmasq, see attached photo, any advice? it doesn't like "eno1" but that is what the interface is named. Still looking through old cases for a clue lol. Thanks!

[image]

giglioroninomicon, 2025-03-09:
Reading through this chain at the moment https://github.com/iiab/iiab/issues/3385

giglioroninomicon, 2025-03-09:
Getting closer lol

giglioroninomicon, 2025-03-09:
So I was able to get DHCP/DNS running on the IIAB box, added eno1 to br0, the external outdoor AP is able to connect and get an IP address and have clients connect over wifi. However, since the external AP is not controlled by IIAB I am unable to get the captive portal to work. So my unknown users would need to enter "box.lan" into their browser to get to the IIAB homepage.

Any ideas on how I can send them there?

Really appreciate the support as well - thanks Tim and @jvonau

giglioroninomicon, 2025-03-09:
I came up with a work around solution. I plugged in my trusty Atheros based USB 2.4Ghz wifi adapter and the captive portal started working. I reconfigured the external outdoor AP into repeater mode leveraging the Atheros USB wifi. So far so good. 

I believe you are good to close this case. I will close it myself in a couple days in case I come up with a better methodology.

Thanks again!

jvonau, 2025-03-10:
Without knowing which OS is being tinkered with I can't give step by step instructions. The automatic addition of a wired connection under br0 was [disabled](https://github.com/search?q=repo%3Aiiab%2Fiiab%20bridge%20eth0&type=code) at @holta request to be able to plug the wired connection back into a different network at some point in time in the future. 

>external outdoor AP is able to connect and get an IP address
>I reconfigured the external outdoor AP into repeater

Seems what you want is to have the external AP to be in 'bridge mode' so that the external AP is transparent to the WiFi clients and the requests get forwarded directly to the IIAB 'up the wire' without really 'seeing' the AP.

giglioroninomicon, 2025-03-11:
@jvonau I think I am in 'bridge mode' even though the AP manufacturer calls it "repeater mode". To test some things, I unplugged the USB wifi adapter from the USB port. The external AP is still handling traffic and sending it in over the wire. The interesting thing is I have both the IIAB device ethernet cable on 10.10.10.10 performing DHCP/DNS/Captive portal, and the external AP device ethernet cable plugged into a small intelligent switch. I have a monitor port turned on that watches the IIAB ethernet port and the AP ethernet port. Running wireshark to capture the mirror port data shows the external AP doesn't currently appear to be using TCP/IP but the traffic is still being served up correctly (unless I've misconfigured the monitor port). It's late for me and I'll play more tomorrow but I'm guessing what you said "the requests get forwarded directly to the IIAB 'up the wire' without really 'seeing' the AP" may be happening. Thanks again as well!

jvonau, 2025-03-11:
Using the switch between the IIAB machine and the AP is a good demonstration of being able to scale from 1 AP to as many as the switch can handle, just configure the additional access points the same way as the first. All the requests will show up on 10.10.10.10

jvonau, 2025-03-11:
PS you never really disclosed which OS you are playing with, I be interested to know.

giglioroninomicon, 2025-03-11:
@jvonau Good morning, I'm running Ubuntu 24.04.2 server version. Thanks again for all the help!