iiab/iiab #3961: Can IIAB offer better Ansible messaging when hostapd fails to (re)start during IIAB installs e.g. on "RasPiOS with desktop" ?

holta, 2025-03-05:
Can we offer better Ansible messaging to implementers who get stuck during vanilla IIAB installs like this?

> TASK [network : Waiting 10 seconds for network to stabilize for ap0] ***********
changed: [127.0.0.1]
>
> TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service hostapd: Job for hostapd.service failed because the control process exited with error code.\nSee \"systemctl status hostapd.service\" and \"journalctl -xeu hostapd.service\" for details.\n"}
>
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=787  changed=441  unreachable=0    failed=1    skipped=193  rescued=0    ignored=2

```
root@raspberrypi:~# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2025-03-05 13:36:38 EST; 11min ago
    Process: 57593 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
        CPU: 17ms

Mar 05 13:36:38 box hostapd[57593]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Mar 05 13:36:38 box hostapd[57593]: Could not read interface ap0 flags: No such device
Mar 05 13:36:38 box hostapd[57593]: nl80211 driver initialization failed.
Mar 05 13:36:38 box hostapd[57593]: ap0: interface state UNINITIALIZED->DISABLED
Mar 05 13:36:38 box hostapd[57593]: ap0: AP-DISABLED
Mar 05 13:36:38 box hostapd[57593]: ap0: CTRL-EVENT-TERMINATING
Mar 05 13:36:38 box hostapd[57593]: hostapd_free_hapd_data: Interface ap0 wasn't started
Mar 05 13:36:38 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Mar 05 13:36:38 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Mar 05 13:36:38 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
```

```
root@raspberrypi:~# journalctl -u hostapd
Mar 05 12:42:01 box systemd[1]: hostapd.service - Access point and authentication server for Wi-Fi and Ethernet was skipped because of an unmet condition check (ConditionFileNotEmpty=/etc/hostapd/hostapd.conf).
Mar 05 13:23:54 box systemd[1]: Starting hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Mar 05 13:23:54 box hostapd[57545]: Could not read interface ap0 flags: No such device
Mar 05 13:23:54 box hostapd[57545]: nl80211: Driver does not support authentication/association or connect commands
Mar 05 13:23:54 box hostapd[57545]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Mar 05 13:23:54 box hostapd[57545]: Could not read interface ap0 flags: No such device
Mar 05 13:23:54 box hostapd[57545]: nl80211 driver initialization failed.
Mar 05 13:23:54 box hostapd[57545]: ap0: interface state UNINITIALIZED->DISABLED
Mar 05 13:23:54 box hostapd[57545]: ap0: AP-DISABLED
Mar 05 13:23:54 box hostapd[57545]: ap0: CTRL-EVENT-TERMINATING
Mar 05 13:23:54 box hostapd[57545]: hostapd_free_hapd_data: Interface ap0 wasn't started
Mar 05 13:23:54 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Mar 05 13:23:54 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Mar 05 13:23:54 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Mar 05 13:36:38 box systemd[1]: Starting hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Mar 05 13:36:38 box hostapd[57593]: Could not read interface ap0 flags: No such device
Mar 05 13:36:38 box hostapd[57593]: nl80211: Driver does not support authentication/association or connect commands
Mar 05 13:36:38 box hostapd[57593]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Mar 05 13:36:38 box hostapd[57593]: Could not read interface ap0 flags: No such device
Mar 05 13:36:38 box hostapd[57593]: nl80211 driver initialization failed.
Mar 05 13:36:38 box hostapd[57593]: ap0: interface state UNINITIALIZED->DISABLED
Mar 05 13:36:38 box hostapd[57593]: ap0: AP-DISABLED
Mar 05 13:36:38 box hostapd[57593]: ap0: CTRL-EVENT-TERMINATING
Mar 05 13:36:38 box hostapd[57593]: hostapd_free_hapd_data: Interface ap0 wasn't started
Mar 05 13:36:38 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Mar 05 13:36:38 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Mar 05 13:36:38 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
```

```
root@raspberrypi:~# cat /etc/hostapd/hostapd.conf
# Basic configuration

interface=ap0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=Internet in a Box
channel=6
bridge=br0

# Hardware configuration
driver=nl80211
country_code=US
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g
```

```
root@raspberrypi:~# rfkill
ID TYPE      DEVICE      SOFT      HARD
 0 bluetooth hci0   unblocked unblocked
 2 wlan      phy1   unblocked unblocked
```

```
root@raspberrypi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:70:06:af brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.184/24 brd 192.168.0.255 scope global noprefixroute eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::659e:c6f0:bad6:e291/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
4: tailscale0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1280 qdisc pfifo_fast state UNKNOWN group default qlen 500
    link/none
5: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether dc:a6:32:70:06:b0 brd ff:ff:ff:ff:ff:ff
6: br0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 96:d3:d7:07:28:21 brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.10/24 brd 10.10.10.255 scope global br0
       valid_lft forever preferred_lft forever
```

iiab-diagnostics: https://paste.centos.org/view/0e9547bf

jvonau, 2025-03-06:
>Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
/etc/debian-version: 12.9

>=IIAB==========================================================================
COMMAND: /usr/bin/sudo dmesg | grep -i "command line:"    # Kernel boot parameters
 [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  numa=fake=2 system_heap.max_order=0 smsc95xx.macaddr=DC:A6:32:70:06:AF vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=50cbe625-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

Refuse to install if the country code for the wifi has not been set first before installing, there is no cfg80211.ieee80211_regdom= present

holta, 2025-03-06:
Something to think about: ✅

- Interesting that headless RasPiOS (i.e. Lite) is able to proceed without a country code, successfully completing IIAB install.

- Whereas RasPiOS with desktop fails here.

So...

- Explanatory Ansible messaging, actionable so implementers learn new requirements, would be the minimum (in my opinion anyway!)

- And strict blocking can also be considered yes (if it's carefully tailored!?)

holta, 2025-03-06:
Any idea why `host_country_code: US` in /etc/iiab/local_vars.yml did not have take effect in this case?

[image]

iiab-diagnostics: https://paste.centos.org/view/0e9547bf

jvonau, 2025-03-06:
My guess would be is because NetworkManager has control of wlan0 or something in the desktop tied up wlan0 and looks like cloning for ap0 failed.

jvonau, 2025-03-06:
Suppose one could vary {{ hostapd_wait }} in local_vars and see if it just a race situation while NM gets fully restarted.

jvonau, 2025-03-06:
Might want to do a no fail at that point, there is a second restart if need a [little below](https://github.com/iiab/iiab/blob/master/roles/network/tasks/restart.yml#L104-L114)

holta, 2025-03-06:
Great ideas both (just above).  I'm busy today but will help over the coming few days as soon as I can.

> Question when you set the country code in interactive (menu) mode with raspi-config is the system forced to reboot or tells you to reboot?

Useful question from 4 hours ago: the answer is that... I've confirmed raspi-config does NOT reboot:

- I've also confirmed raspi-config now requires SSID immediately after you input WiFi country code (unlike the graphical widget in the top-right of "RasPiOS with desktop" — where those are 2 separate manual steps).
- Long story short, both approaches (raspi-config and the graphical widget) both connect you to a nearby WiFi SSID without rebooting.

holta, 2025-03-06:
> or tells you to reboot?

I did not see it asking for reboot (both within text-based raspi-config, and also within "RasPiOS with desktop" widget in the top-right).

holta, 2025-03-07:
One option might be to make the `STAGE` variable (counter) in `/etc/iiab/iiab.env` rise to 10 — instead of just to [9 (as  set in iiab-install)](https://github.com/iiab/iiab/blob/423e21c52ca87aa780135d04a29eb627c49ea58d/iiab-install#L169):

- e.g. to help implementers who type in `sudo iiab` so as to run the equivalent of `iiab-network` repeatedly  — at least in the common case where Ansible variable `network_install: True` is set.

- A counterargument might be that `iiab-network` (i.e. IIAB's network role) should perhaps only be run 3-times-or-so before giving up on that?  (Anyway that's one possible idea... not sure it's the best idea... during initiall `iiab-install` it might well be best to respect [SKIP_ROLE_ON_ERROR=false](https://github.com/iiab/iiab/blob/423e21c52ca87aa780135d04a29eb627c49ea58d/iiab-install#L18) across the board...)

tim-moody, 2025-03-07:
seems like overkill. just rerun hostapd start if fails. (I think sometimes the first run sets up the country code if it is a change and only the second actually starts hostapd.

holta, 2025-03-07:
The point is that non-technical will never know to run `systemctl restart hostapd` and when to give up (and if they should give up!)

Technically proficient users are completely off-topic for the matter of this discussion.

The question is about everyday users who keep typing in `sudo iiab` until IIAB appears to be fully installed.

In other words, some best effort attempt to really complete the Ansible role (roles/network) can and should be considered for non-technical users.  I'll look into it in coming day.

jvonau, 2025-03-07:
I feel blaming the network role for something that is clearly related to a firmware issue is trying to sully my good name. I feel personally attacked.

holta, 2025-03-07:
Let's definitely NOT blame the network role!

Just like any other role, it may need to be run several times until it completes properly, that's all.

I'll look into it.  There may be several good ways to make this happen.

holta, 2025-03-07:
Alongisde AND in addition to considering @jvonau's 2 excellent suggestions from 24+ hours ago:

> Suppose one could vary {{ hostapd_wait }} in local_vars and see if it just a race situation while NM gets fully restarted.

> Might want to do a no fail at that point, there is a second restart if need a [little below](https://github.com/iiab/iiab/blob/master/roles/network/tasks/restart.yml#L104-L114)

jvonau, 2025-03-07:
Related: https://forums.raspberrypi.com/viewtopic.php?t=376246 The forced replacement of firmware might need a rethink as the firmware appears to be embedded in the inittramfs image.

holta, 2025-03-07:
> just rerun hostapd start if fails.

If only (:  FWIW that failed (naturally `systemctl restart hostapd` was the first thing I tried 2 days ago).

I'll explore more in the coming week to see if this is consistently reproducible, or it it's more intermittent.

- In any case, respecting and honoring [SKIP_ROLE_ON_ERROR=false](https://github.com/iiab/iiab/blob/423e21c52ca87aa780135d04a29eb627c49ea58d/iiab-install#L18) is something that should be at least considered, as one of several UX-oriented solutions to help non-technical / newbie implementers understand when they should (or should not!) give up.

holta, 2025-03-07:
> I'll explore more in the coming week to see if this is consistently reproducible, or if it's more intermittent.
> 
> * In any case, respecting and honoring [SKIP_ROLE_ON_ERROR=false](https://github.com/iiab/iiab/blob/423e21c52ca87aa780135d04a29eb627c49ea58d/iiab-install#L18) is something that should at least be considered, as one of several UX-oriented solutions to help non-technical / newbie implementers understand when they should (or should not!) give up.

- And if in the end a reboot proves necessary in some of these NM-or-whatever situations, that's unfortunate but so be it! (:

- What really matters is that `sudo iiab` newbie implementers get clear + actionable Ansible messaging, so they know what to do next! ✅

holta, 2025-03-07:
An Ethernet IIAB install onto "Raspberry Pi with desktop" on Raspberry Pi 500 failed in the exact same way as on RPi 4:

- Both `systemctl restart hostapd` and `iiab-network` appear to fail every time (until reboot).
- After reboot, both commands work!

ASIDE: Ansible messaging and/or respecting and honoring [SKIP_ROLE_ON_ERROR=false](https://github.com/iiab/iiab/blob/423e21c52ca87aa780135d04a29eb627c49ea58d/iiab-install#L18) will still need to be improved, in due course, but one step at a time.  As we think through different options around how best to materially help newbie implementers understand what's happening — when things (inevitably!) go wrong year-by-year as technical environments evolve in one way or another, ETC  (:

holta, 2025-03-07:
@jvonau I'll help overnight or over the weekend, if you have any recommendations on how to proceed here.

tim-moody, 2025-03-07:
personally my workflow includes apt update and upgrade after which I reboot. I set country code before that reboot.

holta, 2025-03-07:
@jvonau asked some [good questions Wednesday/Thursday](https://github.com/iiab/iiab/issues/3961#issuecomment-2704513104):

Interestingly (FWIW!) RasPiOS can set the WiFi country code and establish a WiFi connection without rebooting, regardless whether one uses the CLI tool (raspi-config) or the graphical widget in the top-right of their graphical desktop.

jvonau, 2025-03-08:
> Any idea why `host_country_code: US` in /etc/iiab/local_vars.yml did not take effect in this case?
 iiab-diagnostics: https://paste.centos.org/view/0e9547bf
 
Think you are mistaken, the raspi-config routine to auto enable wifi within the role worked as shown below.
>COMMAND: /usr/sbin/iw reg get    # Detected WiFi country code / legal frequencies
 global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
phy#1
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)
 
 
Think the desktop scan to populate the pick menu and/or poking at the presented list left the wifi chip set to the last seen Access Point.
>COMMAND: /usr/sbin/iw dev    # List wireless interfaces
phy#1
        Unnamed/non-netdev interface
                wdev 0x100000002
                addr de:a6:32:70:06:b0
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 5
                wdev 0x100000001
                addr dc:a6:32:70:06:b0
                type managed
                channel 165 (5825 MHz), width: 20 MHz, center1: 5825 MHz
                txpower 31.00 dBm

Now with the channel being at '165' the starting of hostapd fails as the the channel used with hostapd is '6' from the default setting in local_vars.yml

tim-moody, 2025-03-08:
good sleuthing

holta, 2025-03-08:
> Think the desktop scan to populate the pick menu and/or poking at the presented list left the wifi chip set to the last seen Access Point.

Not quite.  Interestingly, what's actually happening is `raspi-config nonint do_wifi_country {{ host_country_code }}` on [Line 69 of roles/network/tasks/main.yml](https://github.com/iiab/iiab/blob/423e21c52ca87aa780135d04a29eb627c49ea58d/roles/network/tasks/main.yml#L67-L70) changes both rather immediately:

1. `iw reg get` output shows "country" changed from 00: DFS-UNSET to US: DFS-FCC.
2. `iw dev` output shows "channel" changed from 34 to 120 — **and then a few seconds later, channel changes from 120 to 165.**

(I've reconfirmed this on a clean / freshly installed "RasPiOS with desktop" by running `raspi-config nonint do_wifi_country US` right after applying apt updates and rebooting.)

> Now with the channel being at '165' the starting of hostapd fails as the the channel used with hostapd is '6' from the default setting in local_vars.yml

Yes.

Thanks for explaining what's happening.

holta, 2025-03-08:
> * Both `systemctl restart hostapd` and `iiab-network` appear to fail every time (until reboot).
> * After reboot, both commands work!

I haven't yet figured out why `iiab-network` starts working after reboot (but @jvonau probably has an idea?)

jvonau, 2025-03-08:
> > Think the desktop scan to populate the pick menu and/or poking at the presented list left the wifi chip set to the last seen Access Point.
> 
> The graphical widget in the top-right of the OS desktop was never touched, so it wasn't that.
> 
>     2. `iw dev` output shows "channel" changing from 34 to 120 — **and then a few seconds later, channel changes again, from 120 to 165.**
> 
> 
> (I've reconfirmed this on a clean / freshly installed 64-bit "RasPiOS with desktop" on RPi 500, by running `raspi-config nonint do_wifi_country US` right after applying apt updates and rebooting.)
> 

Seems to support the theory of populating the AP pick list in the GUI.

jvonau, 2025-03-08:
> > * Both `systemctl restart hostapd` and `iiab-network` appear to fail every time (until reboot).
> > * After reboot, both commands work!
> 

Did you even bother to check if hostapd was running before restarting the service? I bet that hostapd was already running based on the staging I built into the unit files that start the required services.

holta, 2025-03-08:
RECAP from the top of this ticket — hostapd failed:

```
root@raspberrypi:~# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2025-03-05 13:36:38 EST; 11min ago
    Process: 57593 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
        CPU: 17ms

Mar 05 13:36:38 box hostapd[57593]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Mar 05 13:36:38 box hostapd[57593]: Could not read interface ap0 flags: No such device
Mar 05 13:36:38 box hostapd[57593]: nl80211 driver initialization failed.
Mar 05 13:36:38 box hostapd[57593]: ap0: interface state UNINITIALIZED->DISABLED
Mar 05 13:36:38 box hostapd[57593]: ap0: AP-DISABLED
Mar 05 13:36:38 box hostapd[57593]: ap0: CTRL-EVENT-TERMINATING
Mar 05 13:36:38 box hostapd[57593]: hostapd_free_hapd_data: Interface ap0 wasn't started
Mar 05 13:36:38 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Mar 05 13:36:38 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Mar 05 13:36:38 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
```

holta, 2025-03-08:
> 2\. `iw dev` output shows "channel" changing from 34 to 120 — **and then a few seconds later, channel changes again, from 120 to 165.**

To show more detail, I ran `watch -n1 iw dev` while ./iiab-install's roles/network was running:

- What happens is that the "channel" number (as output by `iw dev`) dances around erratically for about a full minute.  In other words, the "channel" value output by command `iw dev` jumps around to all sorts of different numbers (from about 1 to 165 or so, it seems?)
- Until this channel number finally settles down (finalizes on) channel 165 in the end.
- From what I can tell, the channel is jumping to a new number every second or so (but possibly it's happening even more often than that, as `watch -n1` was only checking the value once per second).

holta, 2025-03-09:
Recap, with a bit more detail:

- `iw dev` outputs an almost random number as the channel, while scanning for nearby WiFI SSIDs: https://forums.raspberrypi.com/viewtopic.php?t=175480

- This happens often, and repeatedly, soon after boot especially, but also well after boot — regardless whether one hovers over the WiFi SSID selector graphical widget or not (in the top-right of 'RasPiOS with desktop').

- Tests were done on RPi 4, 5 and 500.

- Freshly installed OS; IIAB was not installed; fully apt updated & rebooted prior to testing with `raspi-config nonint do_wifi_country US` and then `watch -n.1 iw dev`

jvonau, 2025-03-10:
At least I didn't waste my time to prove myself correct, thanks for confirming I know what I speak of without me touching a keyboard. Now you know why I don't engage. The sky is green, yup the sky is green. Live is to short to always be proving myself.

holta, 2025-04-01:
Good enough:

- Until we know more, PR #3965 tightens up substantially.
- @EMG70 will help with Linux Mint testing in coming days, if he has time, to make sue that common path also remains solid.

holta, 2025-04-03:
> * [@EMG70](https://github.com/EMG70) will help with Linux Mint testing in coming days, if he has time, to make sure that common path also remains solid.

Confirmed:

- Huge thanks @EMG70 for having verified that IIAB installs well onto Linux Mint 22.1 are on track.

- @avni will hopefully assist in the coming week to figure out why 💾 ["Upload to USB"](https://wiki.iiab.io/go/FAQ#Can_students_upload_their_own_work%3F) 💾 is not currently working on Linux Mint.