iiab/iiab #3963: Unable to log into IIAB admin console

4casts, 2025-03-11:
I just downloaded IIAB and am unable to log into the admin console using the default login.

Diagnostics file URL: https://paste.centos.org/view/4457a347

Thank you for your help!

holta, 2025-03-11:
Hi @4casts,

What exact "default login" are you trying?

(Using which URL?  Include a screenshot or photo of any errors if possible?)

4casts, 2025-03-11:
[image]

I'm using the iiab-admin login from the FAQs on http://box/admin. I'm also on Raspberry Pi if that helps.

holta, 2025-03-11:
Any chance you might have changed the password for Linux user `iiab-admin` during your IIAB install 2+ weeks ago on Saturday February 22nd?

4casts, 2025-03-11:
I didn't think I did but it's definitely possible - I attempted to install IIAB then but kept hitting an error with one of the apps downloading, so I tabled it until I was able to revisit the download tonight. This is the first time I've been able to access the admin console login page.

Any advice for correcting that user error if so?

4casts, 2025-03-11:
Update: I reset the admin password using `sudo passwd iiab-admin` and will mark this closed. Thank you so much for your time!

4casts, 2025-03-11:
It was JupyterHub.

jvonau, 2025-03-11:
>/opt/iiab/iiab/iiab-install.log-2025-02-22 00:28:00,000 p=8987 u=root n=ansible INFO| fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/opt/iiab/jupyterhub/bin/pip3", "install", "--no-cache-dir", "--prefer-binary", "jupyterlab", "jupyterhub_firstuseauthenticator", "jupyterhub-systemdspawner", "ipywidgets"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\n
Collecting jupyterlab\n  
Downloading https://www.piwheels.org/simple/jupyterlab/jupyterlab-4.3.5-py3-none-any.whl (11.7 MB)\n
     ━━━━━━━━━━                               3.0/11.7 MB 63.4 kB/s eta 0:02:17\n
\n
:stderr: ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.\n
    jupyterlab from https://www.piwheels.org/simple/jupyterlab/jupyterlab-4.3.5-py3-none-any.whl#sha256=571bbdee20e4c5321ab5195bc41cf92a75a5cff886be5e57ce78dfa37a5e9fdb:\n
        Expected sha256 571bbdee20e4c5321ab5195bc41cf92a75a5cff886be5e57ce78dfa37a5e9fdb\n
             Got        9bb3e728ef58c75d3602b7ac7fda875dc817e0f0f20d19de9a7432b635b499d3\n\
n"}
/opt/iiab/iiab/iiab-install.log:2025-02-22 00:28:00,052 p=8987 u=root n=ansible INFO| TASK [jupyterhub : SEE ERROR ABOVE (skip_role_on_error: False)] ****************

2 observations:

- The above looks really bad with checksums not matching would be a huge security concern to be further investigated.
- Why on earth would an aarch64 machine be looking at piwheels when the wheels there are intended for 32bit OS-images?

jvonau, 2025-03-11:
To answer question 2, it is the indiscriminate adding of pip.conf within scripts/ansible
Ref: https://github.com/piwheels/packages/issues/390

holta, 2025-03-11:
> It was JupyterHub.

Thank You @4casts:

I'll re-open this issue if that's Ok — in case JupyterHub installs can be made simpler and more reliable.

jvonau, 2025-03-11:
@holta Spin this off into 2 issues based on the observations above, no need to spam @4casts with the technical details of the upcoming discussion. Hint pip.conf's presents might be an upstream issue.

holta, 2025-03-11:
I'll humbly disagree.  This ticket is a fine place to start, preserving context for anybody who wants to make improvements here.  With a big thank you to @4casts for his/her shining a light an participating as he/she chooses (anybody can unsubscribe from any ticket if they prefer!)  💯

jvonau, 2025-03-11:
Ok then start with tracking down where pip.conf is coming from on 64bit (aarch64) images. As you should know from the top left corner of https://www.piwheels.org/ that is a link to  https://github.com/piwheels/packages/issues/119 Happy hunting.

holta, 2025-03-11:
Yes, and a few more tips for whoever works on this:

- The piwheels project aspires to support 64-bit not just 32-bit as they've explained: piwheels/packages#390

- This means that 2025 (and especially the release of RasPiOS 13 Trixie around Q3 or so?) will start to shine a much brighter light as to what's really happening on this front, in a few short months.

holta, 2025-03-11:
> What does this have to do with Admin Console?

@4casts couldn't figure out how to log into Admin Console, after JupyterHub interrupted his/her IIAB install.  He/She has since figured it out, after changing the iiab-admin password at the CLI.

jvonau, 2025-03-11:
> What does this have to do with Admin Console?

Other than the password needed to be reset for some reason, nothing. 

I just spotted the repeated failure of jupyterlab with a really scary warning about **someone may have tampered with them** that should warrant a closer look.

jvonau, 2025-03-11:
> Yes, and a few more tips for whoever works on this:
> 
>     * The piwheels project aspires to support 64-bit not just 32-bit as they've explained: [Package issue: cryptography 41.0.3 leads to cffi 1.15.1 failure on "32-bit" Raspberry Pi OS [REASON: /etc/pip.conf missing on some Bookworm pre-releases] piwheels/packages#390](https://github.com/piwheels/packages/issues/390)

Please read pywheels 119 above, arm64|aarch64 wheels should exist on PyPI already, no need to reinvent the wheels. (pun fully intended)

holta, 2025-03-11:
CLAIF: I prefer more keywords in subject lines, to help make concepts and tickets much more findable.

jvonau, 2025-03-15:
> To answer question 2, it is the indiscriminate adding of pip.conf within scripts/ansible

That is not really accurate /etc/pip.conf is installed by upstream which originates from http://archive.raspberrypi.com/debian/pool/main/p/python-pip/python3-pip-whl_23.0.1+dfsg-1+rpt1_all.deb via the /etc/apt/sources.list.d/raspi.sources file that is present. The /etc/pip.conf file should not be present on arm64 images for the reasons mentioned in 119 above. I suggest tweaking scripts/ansible to remove that file on arm64 based installs instead of forcing /etc/pip.conf being present.