iiab/iiab #3967: Output of iiab-summary on the 10-minute CI run (Ubuntu 24.04) "fatal: no names found, cannot describe anything"

Akatama, 2025-03-15:
While working on the changes for the 10 minute CI run (the one that runs on Ubuntu 24.04) I noticed that iaab-summary command has some strange output. I am very new to this project, but just as a new user if I saw something like this I would assume something was wrong with how Internet-in-a-Box was set up on my system if I saw output like this.

Let me know if this is actually the expected output, or if something is awry in the setup so I can look into fixing it.

The example below is actually taken from the last CI run on this repo before my recent pull request, so its not something that is only happening on my fork:

`fatal: No names found, cannot describe anything.
/usr/bin/iiab-summary: line 25: cd: /opt/iiab/iiab-admin-console: No such file or directory
fatal: No names found, cannot describe anything.
install_date = 2025-03-12T07:57:55Z   Current TZ: UTC

iiab: ef1bb46, 0 PR's / 0 commits since tag 
    "Merge 89b31ee085017fe8d149bf54e56a06d75dff859a into 752ab0ee19552d7f2cdff3b1a048ea6761d4301d"
    none   branch: 

    WARNING: Directory /opt/iiab/iiab-admin-console does not exist!

Ubuntu 24.04.2 LTS   trixie/sid
display-manager? inactive   Arch1: amd64   Arch2: 
box 6.8.0-1021-azure #25-Ubuntu SMP Wed Jan 15 20:45:09 UTC 2025 x86_64
AMD EPYC 7763 64-Core Processor   CPU(s): 4   Mem: 15990 Swap: 4095

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        72G   48G   24G  67% /      ZIMs: 0 OER2Go: 0 Apps2B: 1

lo eth0 docker0
box box.lan fv-az1441-793.lnin1eoc0irezeqatpkuni3jdf.cx.internal.cloudapp.net localhost localhost.localdomain
10.1.0.126 172.17.0.1 
`

tim-moody, 2025-03-15:
>/opt/iiab/iiab-admin-console: No such file or directory

assuming this is not permissions, I assume it means that that repo has not been cloned yet.

Akatama, 2025-03-15:
@tim-moody at the point this is called, iiab should be fully installed. Based on what I am seeing later (e.g. /etc/iiab/iiab_state.yml is populated) I suspect that iiab is either mostly or fully set up. Permissions issue is possible. Let me test that. 

Thanks for the thoughts!

My goal with this project is to give these github workflows a look over so the CI runs can be more useful. In the future we shouldn't have any steps that are like this - they should either be failing or succeeding. Not exist in some kind of superposition between the two because we don't have any validation on the output.

jvonau, 2025-03-15:
Not being cloned would be the reason why that repo is not present. Perhaps the workflow should include installing iiab-admin-console? This workflow does not mirror what the curl routine does when calling [`iiab`](https://github.com/iiab/iiab-factory/blob/master/iiab)

Akatama, 2025-03-15:
The first step in the test-install job is actions/checkout@v4, which as it is currently set up with all default values is the equivalent to  ```git clone --depth 1 https://github.com/iiab/iiab.git``` or whatever the URL to your fork is. This is checked out in the default location for $GITHUB_WORKSPACE.

A. Holt told me about the simple set up at SCaLE. I think you are both correct that workflow maybe has an old way to set up IIAB or something. I had noticed this before, but since I am new to the project I didn't want to make such a major change without knowing more. I will test the simple setup on the my local machine first, then try it as a CI run. If that works then that is what I will change it too.

holta, 2025-03-15:
Good point @Akatama:

scripts/iiab-summary should probably be refined with cleaner output — things like /opt/iiab/iiab-admin-console sometimes do not exist for good reason (e.g. during tiny IIAB installs when Admin Console is not needed, during testing, etc).

Akatama, 2025-03-15:
Thank you!

I am installing with local_vars_small.yml in this workflow. Looking at that, it should be installing iiab-admin-console, but the output suggests it is not. I will look into that.

But if installing without it is a valid scenario, then we need to consider it for future workflows and tests. In such a case some nicer output would make a lot of sense.

jvonau, 2025-03-15:
Combined iiab & iiab-admin-console testing might be better served in iiab-factory with an eye towards using the routine to generate finished images for distribution.
Prior work
 - https://github.com/iiab/iiab-factory/
 - https://github.com/georgejhunt/iiab-smoketest
 - https://github.com/jvonau/XSCE-builder
 - https://github.com/jvonau/mk-image
 - https://github.com/jvonau/iiab-upgrader

holta, 2025-03-16:
@Akatama can you restest now, thanks to the following quickly merged PR?

- PR #3968

Akatama, 2025-03-16:
@holta I apologize, my wife worked nigh shift last night so I had to take care of my daughters alone most of the day. I saw your PR but I didn't have a chance to look at it until now. 

My first thought is that while you got rid of one fatal line, we still have the first one, which still seems to me like a "something is horribly wrong" situation. However, as already previously stated, something is wrong with the setup for that workflow, so I want to try really quickly on multipass with a minimal install. I will do that and report back here. 

```
fatal: No names found, cannot describe anything.
install_date = 2025-03-16T02:19:22Z   Current TZ: UTC

iiab: 78eb42e, 0 PR's / 0 commits since tag 
    "Merge pull request #3968 from holta/iiab-summary_cleaner"
    https://github.com/iiab/iiab   branch: master

    WARNING: Directory /opt/iiab/iiab-admin-console does not exist!

Ubuntu 24.04.2 LTS   trixie/sid
display-manager? inactive   Arch1: amd64   Arch2: 
box 6.8.0-1021-azure #25-Ubuntu SMP Wed Jan 15 20:45:09 UTC 2025 x86_64
AMD EPYC 7763 64-Core Processor   CPU(s): 4   Mem: 15990 Swap: 4095

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        72G   48G   24G  67% /      ZIMs: 0 OER2Go: 0 Apps2B: 1

lo eth0 docker0
box box.lan fv-az1043-573 localhost localhost.localdomain
10.1.0.33 172.17.0.1 
```

holta, 2025-03-16:
@Akatama no immediate rush, but when you can:

Any chance you can tell why exactly `fatal: No names found, cannot describe anything.` is appearing?

holta, 2025-03-16:
FYI:

"git describe fails with "fatal: No names found, cannot describe anything.""
https://stackoverflow.com/questions/4916492/git-describe-fails-with-fatal-no-names-found-cannot-describe-anything

holta, 2025-03-16:
Root Cause: GitHub Actions (GHA) appears to be clone repos with `--depth 0` (e.g. cloning of git repo iiab/iiab in this case!)

https://stackoverflow.com/questions/71819174/shallow-update-not-allowed-with-github-actions/71819349#71819349

1. One quick-and-dirty workaround (imperfect, but possibly good enough for now??) might be to force GHA to clone with depth 1000 — thereby using up an additional 15MB in the runner's `iiab/.git/objects` — but that's probably a negligeable amount...

   ```
   - uses: actions/checkout@v3
     with:
       fetch-depth: 1000
   ```

2. Or there might be a more resilient / graceful solution?!

holta, 2025-03-16:
I don't like...

```
fetch-depth: 0
```

...as that causes GHA to fetch all branched unnecessarily!

So my recommendation (for now!) is that we jack up 1000 and make it 10000 — just in case the most recent tag retrieved by [git describe --tags --abbrev=0](https://github.com/holta/iiab/blob/d971af7ac839282bca2c1930472bd14ad6ecb2ad/scripts/iiab-summary#L10) happens to be buried deep...

```
fetch-depth: 10000
```

Akatama, 2025-03-16:
@holta I was going to look into why when I got multipass working, but alas, it seems like even though multipass is installed there is something wrong with it on my system (granted I am on openSUSE Tumbleweed). Might be worth it to dual-boot Ubuntu in the near future for this project if I cannot figure out a fix for it.

I am testing with the fetch-depth suggestion right now.

I am honestly thinking a more graceful solution would be to use the install instructions that you provide to your users:

```
curl iiab.io/install.txt | bash
```

At least for this simple workflow. But I can also see why we would want to do a regular checkout for the run.

Akatama, 2025-03-16:
@holta Oops, the 10 minute CI run just finished. Still has fatal: https://github.com/Akatama/iiab/actions/runs/13880082698/job/38837519339

I did a run before but I forgot to pull your recent PR for the iiab-summary script.

Let me try the curl command.

```
fatal: No names found, cannot describe anything.
install_date = 2025-03-16T04:59:51Z   Current TZ: UTC

iiab: 653eaaac3, 0 PR's / 0 commits since tag 
    "iiab-summary: No need to twice mention dir absence"
    https://github.com/Akatama/iiab   branch: 1000_depth

    WARNING: Directory /opt/iiab/iiab-admin-console does not exist!

Ubuntu 24.04.2 LTS   trixie/sid
display-manager? inactive   Arch1: amd64   Arch2: 
box 6.8.0-1021-azure #25-Ubuntu SMP Wed Jan 15 20:45:09 UTC 2025 x86_64
AMD EPYC 7763 64-Core Processor   CPU(s): 4   Mem: 15990 Swap: 4095

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        72G   50G   23G  69% /      ZIMs: 1 OER2Go: 0 Apps2B: 1

lo eth0 docker0 tailscale0
box box.lan fv-az1660-24.qcwxoje0g1iujdwyioltluxobh.ex.internal.cloudapp.net localhost localhost.localdomain
10.1.0.167 172.17.0.1 
````

holta, 2025-03-16:
@Akatama

`fetch-depth: 10000` is my vote at this point — unless we find something much better!  (As it's actually quite resilient.  True it adds ~16MB to runner VMs, but in 2025 that's nearly negligeable.)

PS three little suggestions at https://github.com/iiab/iiab/pull/3966#issuecomment-2727192032 when you have a second.

Akatama, 2025-03-16:
@holta Yeah, my concern isn't so much the amount of data to GitHub's runners.

My concern is that something in the install process is wrong in at least both the old and new 10 minute workflow. I say this because while you are correct with your additions to iiab-summary script if not installing admin-console is legitimate, but local_vars_small.yml actually should result in admin-console being installed. On lines 33 and 34 of local_vars_small.yml, we see this:
```
admin_console_install: True
admin_console_enabled: True
```
So, is it installed but for some reason iiab-summary can't find it? Or did it fail to install? That's what I need to find out.

holta, 2025-03-16:
> On lines 33 and 34 of local_vars_small.yml, we see this:
> 
> ```
> admin_console_install: True
> admin_console_enabled: True
> ```

Right!  Recall that [iiab/iiab-admin-console](https://github.com/iiab/iiab-admin-console) is a separate repo, that should probably have its own CI/QA scripts at some point.  But if you really want to test Admin Console here (as an exercise, or for whatever reason) then you'd need add these steps equivalent to:

```
cd /opt/iiab
sudo git clone https://github.com/iiab/iiab-admin-console
cd iiab-admin-console
sudo ./install
```

Akatama, 2025-03-16:
@holta I see. So just to confirm, that means that the steps in 10 minute workflow doesn't actually install admin-console, even though local_vars_small.yml has them marked as to be installed?

holta, 2025-03-16:
> just to confirm, that means that the steps in 10 minute workflow doesn't actually install admin-console, even though local_vars_small.yml has them marked as to be installed?

Correct.

In contrast, everyday IIAB implementers who run `curl iiab.io/install.txt | bash` will have Admin Console installed when they run the interactive install script... on Line 514 here:

https://github.com/iiab/iiab-factory/blob/f8ecb0e03bc43481fc8af2409906a35f9fee28cf/iiab#L513-L515

Akatama, 2025-03-17:
Thank you, @jvonau. That makes a lot of sense. 

Also, very cool to see some of the history of this project!

Akatama, 2025-03-27:
Fixed in PR #3976

closed 2025-03-27 by Akatama