iiab/iiab #3975: Add support for a school management system / school information system

setupify, 2025-03-26:
It would be great if iiab could include an opensource app for school management.

For example we could install one of these opensource apps:
- https://github.com/frappe/education
- https://github.com/GibbonEdu/core
- https://github.com/francoisjacquet/rosariosis
- ...

avni, 2025-03-27:
Hello! 👋 

IIAB currently has the option to install these three open source LMSs (learning management systems) that support school management with student and teacher accounts: 
- Kolibri: https://github.com/iiab/iiab/tree/master/roles/kolibri#readme; 
- Moodle: https://docs.moodle.org/en/About_Moodle_FAQ; or 
- NextCloud: https://github.com/iiab/iiab/tree/master/roles/nextcloud#nextcloud-readme

Depending on what functionality you are seeking perhaps one of those would work? Or let us know if there's specific functionality you are seeking that is in the apps you shared!

Related info: 
- https://wiki.iiab.io/go/FAQ#What_services_(IIAB_apps)_are_suggested_during_installation?

setupify, 2025-04-01:
Hello @avni, thank you for your reply.

I don't know much of these LMS yet, I need to take time to learn those.

Here are some needed features for the school where I will deploy iiab:
- School admin and users admin: Administer users data, families, student enrollment, teachers, and control aspects of the school such as school years, classrooms.
- Manage student attendance, generate school absence reports.
- Timetable: plan and visualize timetables (schedules) for every student, and every teacher.
- Plan and manage lessons and homework.
- Finance: Manage fees, billing schedules and invoices, receipts and payment reminders.
- Communication: intranet (offline) website, intranet (offline) communication channels (forum, mail, instant messaging).

Do you know if these various features are supported by Kolibri, Moodle, or NextCloud?

setupify, 2025-04-01:
Btw, it seems Gibbon already provides integration with Moodle LMS:
https://docs.gibbonedu.org/getting-started/configuration/integrations/connecting-to-moodle

setupify, 2025-04-01:
We will have limited internet bandwith in this school in Burkina Faso. Kolibri will be very much useful for us, but probably not for the features I've described in my last comment.

After going in more details about Moodle, I think this is not what I'm looking for and it will not address all above features I need.
Moodle is a Learning Management System, but I'm rather looking for a Student Information System:
https://www.reddit.com/r/moodle/comments/1fouysu/creating_student_information_system_on_moodle/

Would you be open to integrate [Gibbon SIS](https://gibbonedu.org/) into IIAB?
I don't know Ansible yet, but I'm familiar with Saltstack (which is quite similar to Ansible), so I may be able to submit a pull request for that.

holta, 2025-04-01:
> Would you be open to integrate [Gibbon SIS](https://gibbonedu.org/) into IIAB?
> I don't know Ansible yet, but I'm familiar with Saltstack (which is quite similar to Ansible), so I may be able to submit a pull request for that.

Interesting.  Thanks for digging.  While I personally don't enough about Gibbon SIS yet, but feel free to give me a call — to see if if we can help you out here?

Thanks @setupify.  Email to make contact here: https://internet-in-a-box.org/contributing.html

tim-moody, 2025-04-02:
Gibbon looks to be php + mysql, both of which are available on IIAB. It would also need to work with nginx rather than apache and not insist on /var/www/html as the web root. It should be possible to install with ansible.

bbinet, 2025-04-03:
Thank you @tim-moody for these informations.
What do you mean by "not insist on /var/www/html as the web root"? Where should Gibbon static content be hosted instead?

holta, 2025-04-03:
@bbinet Gibbon content (as with most Internet-in-a-Box content) should generally be stored within /library

If the content is very static, it's possible it might be stored within /library/www/html

setupify, 2025-04-03:
Ok, thank you.