iiab/iiab #3986: Add support for deleting .zim files, to allow for updates.

prepperdisk, 2025-04-03:
Rather than requiring system access to delete a zim file, consider adding support via iiab-admin to delete zim files.  Useful for when existing resources update.

prepperdisk, 2025-04-03:
Updated to clarify, zim file support already exists.  But updating maps does not appear to be possible?

tim-moody, 2025-04-03:
>But updating maps does not appear to be possible?

by this you mean removing tiles?

adding regions is possible

prepperdisk, 2025-04-03:
Unless I'm mistaken, after downloading a region I cannot update it.

tim-moody, 2025-04-03:
there is only one version of each region, and currently no one is updating the tiles

prepperdisk, 2025-04-03:
Thank you, I wasn't aware of that.  The latest tiles appear to be from 2022?  I would be interested in helping to update them if there is a resource I can connect with to ramp up.

holta, 2025-04-03:
Thanks @prepperdisk:

I don't yet know how much work updating OpenStreetMap (to more recent vector tiles) will be.

Take a quick look here at the details / outline if you haven't yet:

https://github.com/iiab/iiab/wiki/IIAB-Maps

prepperdisk, 2025-04-03:
Thanks, initially I thought this was an issue with the client not support vector-based map packs but now I see that migration was done (based on the wiki above).  I'm available (30 years of software development) to help in the effort @tim-moody if you or a colleague has a starting point.  Cheers.

holta, 2025-04-03:
FYI I've connected with @prepperdisk directly:

- We'll try to meet up as we happen to be quite nearby, to see what's hopefully possible. 🤞

- We'll involve @tim-moody and others if we hopefully come up with an OpenStreetMap outline or plan. 💫

tim-moody, 2025-04-03:
>I'm available (30 years of software development) to help in the effort

That would be great. I spent some time trying get the Wikimedia Foundation to give us some tiles, but I never got a straight answer on whether they build a cache or always create on demand. We could consider creating our own tiles, say every 6 months, but we have to get some funds for the computational resources.

prepperdisk, 2025-04-03:
I would be very open to helping fund the compute depending on costs.  I have been looking for a way to contribute to this project and that could be one way to do so.