iiab/iiab #3991: Triage troublesome USB stick [Upload2USB]

avni, 2025-04-14:
Creating this issue to track a troublesome USB stick that is causing issues with Upload2USB stick testing. @EMG70 has a stick that was is partitioned with /media/usb0 pointing to a 5B partition: 

```
root@box:/media/usb0/UPLOADS.2025-04-14# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1 117.4G  0 disk 
├─sda1        8:1    1   5.8G  0 part 
├─sda2        8:2    1     5M  0 part /media/usb0
└─sda3        8:3    1   300K  0 part 
mmcblk0     179:0    0 119.1G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 118.6G  0 part /
```

```
root@box:/media/usb0/UPLOADS.2025-04-14# lsusb -v

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            6.12
  iManufacturer           3 Linux 6.12.20+rpt-rpi-v8 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:01:00.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x0009
[... 390 lines cut ...]
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered

```

holta, 2025-04-14:
> /media/usb0 pointing to a 5B partition

I think you mean 5M partition (:

Can you post the URL output by `sudo iiab-diagnostics` ?

avni, 2025-04-14:
sudo iiab-diagnostics: https://paste.centos.org/view/9395c96d

As a final test, i will try to unmount and remount the USB drive from the command line, just to see if it mounts differently.🤞

avni, 2025-04-14:
> As a final test, i will try to unmount and remount the USB drive from the command line, just to see if it mounts differently.

Mounting the 5.8G partition did not work! 

```
root@box:/opt/iiab/iiab# sudo umount /media/usb0
root@box:/opt/iiab/iiab# sudo mount -o umask=0000 /dev/sda1 /media/usb0
mount: /media/usb0: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
```

I was able to mount the 5M partition. 

root@box:/opt/iiab/iiab# sudo mount -o umask=0000 /dev/sda2 /media/usb0
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
root@box:/opt/iiab/iiab# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1 117.4G  0 disk 
├─sda1        8:1    1   5.8G  0 part 
├─sda2        8:2    1     5M  0 part /media/usb0
└─sda3        8:3    1   300K  0 part 
mmcblk0     179:0    0 119.1G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 118.6G  0 part /

EMG70, 2025-04-14:
The 2.1TB USB stick which was causing problems when uploading files has been formatted in last few hours.It has not caused any problems since.I can upload OK ,tried unmounting it and inserted several times without problems.

sudo iiab-diagnostics https://paste.centos.org/view/1654f660
 
```
root@box:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04d9:0007 Holtek Semiconductor, Inc. Raspberry Pi Internal Keyboard
Bus 001 Device 006: ID 048d:04d2 Integrated Technology Express, Inc. UDisk 3.0
Bus 001 Device 003: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@box:~#

```

```
root@box:~# lsusb -v

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            6.12
  iManufacturer           3 Linux 6.12.20+rpt-rpi-v8 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:01:00.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
[... 391 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
root@box:~#

```

holta, 2025-04-14:
@EMG70 thanks so much!
Keep an eye on it nevertheless!

PS Do you believe the `1.9T` USB stick size (as output by lsblk) is for genuine & for real?

EMG70, 2025-04-14:
I think It is actually correct size ,I have written over 1000GB on  it before despite having issues at the start.It wouldn't show full size at first attempt.The price on ebay was very cheap ,I had bought it to test if it was genuine.

holta, 2025-04-14:
Ok!

> price on ebay was very cheap

Maybe don't store your family's archives on there? 😉

EMG70, 2025-04-14:
True,it's not a reliable stick.

avni, 2025-04-14:
Thank you @EMG70 ! :heart: :rocket:

jvonau, 2025-04-20:
`sudo fdisk -l /dev/sda` would of been useful to debug the original issue.