iiab/iiab #3994: PDF downloads for every page

isuzumac, 2025-04-18:
When I first set up my IIAB and offline websites everything was fine, I was able to use the hyperlinks in Wikipedia to go to other Wikipedia pages, look at the book of knots, and all the other offline pages with no issues. Now, no matter what I do, it downloads EVERYTHING I click on as a pdf. Wikipedia hyperlinks stopped working saying they're broken (because the page I am viewing is now a PDF download) My main question is can I reinstall IIAB on my raspberry pi and m.2 and keep all my current downloaded information or do I need to start from scratch? Thank you!

tim-moody, 2025-04-18:
Can you run iiab-diagnostics and paste the link here?

This could be a browser problem. Please tell us what browser and version you are using.

Please provide an example of a link that downloads as a pdf.

isuzumac, 2025-04-22:
I have narrowed my issue down to what I believe is a misunderstanding on my end. The browser I am using is Mozilla Firefox, as Chrome likes to give me weird problems. The only documents that download instead of open in a new tab to view are the "Library of Knots" books. I can right click and force it to open in a new tab rather than download and open in a new tab. All my browser settings are set to "Open in Firefox" for PDF viewing. I will post a diagnostic link when I am back home!