iiab/iiab #3996: IIAB Error on Rasberry Pi I don't know what to do

DryFudge, 2025-04-20:
I installed IIAB on a Raspberry Pi 3 b+ and when I run:

sudo iiab

I get this error:

`ERROR! We are unable to read either as JSON nor YAML. These are the errors we get from each:
JSON: Expecting value: line 1 column 1 (char 0)

Syntax Error while loading YAML:
did not find expected <document start>. did not find expected <document start>
in "<unicode string>", line 16, column 1

The error appears to be in '/etc/iiab/local_vars.yml" : line 16, column 1, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

#Ansible's default timeout for "get_url:" downloads (10 seconds) often fails 
download_timeout: 100
^here `

I have no idea what to do to fix this as I have only just started this project. I know very little about the coding. Any help is appreciate! Thanks.

holta, 2025-04-20:
Hi @DryFudge 

Can you please run `iiab-diagnostics` and reply here with the resulting URL ?

holta, 2025-04-20:
A photo of your screen showing the complete error(s) might also help?

jvonau, 2025-04-21:
Really need to view the local_vars.yml file, perhaps a copy could be rename local_vars.txt and attached to this issue or perhaps @holta can guide in the use of a pastebin like service.

DryFudge, 2025-04-21:
> Hi [@DryFudge](https://github.com/DryFudge)
> 
> Can you please run `iiab-diagnostics` and reply here with the resulting URL ?

I tried running iiab-diagnostics [on its own and with `sudo iiab-diagnostics` and get "iiab-diagnostics: command not found"

DryFudge, 2025-04-21:
> A photo of your screen showing the complete error(s) might also help?

[image] 

Here is what my screen is showing at the time. Thank you for any help! Very much a novice but excited about the iiab.

holta, 2025-04-21:
@DryFudge can you please... ?

1. Make sure your Raspberry Pi is connected to the Internet.

2. Run:

   ```
   sudo /opt/iiab/iiab/scripts/iiab-diagnostics
   ```

3. Hit the [Enter] key, twice.

4. Let us know the resulting URL, Thanks!