iiab/iiab #4005: Publish checksums for each release

victorb, 2025-04-29:
After downloading the .img from https://download.iiab.io/8.2/, I was looking for any sort of checksum (sha256sum, or even md5) for verifying that the download was done correctly, but couldn't find it on the download page nor in any of the files. Would be great to have available for verification purposes.

Would also make it easier to pass along the .img in a offline-scenario, where as long as you know the checksum of the official release, you could feel safe that a stranger passed you the right .img without modifications.

holta, 2025-04-29:
Thanks @victorb for suggesting.

Just FYI at the moment [IIAB 8.3 pre-releases](https://github.com/iiab/iiab/wiki/IIAB-8.3-Release-Notes) are strongly recommended, so that important fixes are improvements are included.

victorb, 2025-04-29:
> Thanks @victorb for suggesting.
> 
> Just FYI at the moment [IIAB 8.3 pre-releases](https://github.com/iiab/iiab/wiki/IIAB-8.3-Release-Notes) are strongly recommended, so that important fixes are improvements are included.

Thanks for the recommendation, but for essentials/archival I'll continue to use only stable releases and to be frank, I'd advise others to do the same. Happy to use pre-releases for less vital software :)