iiab/iiab #4010: tailscale dnsmasq conflict

tim-moody, 2025-05-08:
As currently configured all name resolution goes through tailscale when it is up so the following fail:

- ping download.kiwix.org
- ping github.com

both work after tailsale down

tim-moody, 2025-05-08:
This was true on a VPS.

Tested on rpi 5 with both IIAB and my personal tailnet and neither had this problem?!

holta, 2025-05-08:
Presumably this affects DNS lookup of every public Internet site / name?

( When it's happening, e.g. when IIAB is installed on some VPS's ?)

jvonau, 2025-05-12:
Not enough information to be useful, are DNS lookups failing from the IIAB on the VPS or from the client connecting to the IIAB via tailscale? If the former think you might want to see: https://github.com/tailscale/tailscale/issues/13863

tim-moody, 2025-05-12:
Tailscale takes over dns and in this case only resolves nodes in its VPN.

There is an option to add 1.1.1.1 or similar to solve that, but I suspect the vps I was on has not done this.

I don't think the IIAB tailnet has this problem.

jvonau, 2025-05-12:
Just a little google foo:
>all name resolution goes through tailscale
 
AI Overview

Yes, Tailscale can be configured to route all name resolution through its network, effectively making it the primary source for resolving domain names and IP addresses for devices within a Tailscale network. This can be achieved by enabling "Override DNS servers" in the Tailscale admin console, which directs devices to use the Tailscale-defined global nameservers instead of their local DNS settings. 
Here's a more detailed explanation:

    Tailscale DNS:
    Tailscale provides its own DNS service, known as MagicDNS, that can be used to resolve domain names within the Tailscale network. 

Override DNS:
By enabling the "Override DNS servers" option in the Tailscale admin console, you can force all devices within your Tailscale network to use the Tailscale-defined DNS settings instead of their local DNS settings. 
MagicDNS:
MagicDNS allows devices within the Tailscale network to resolve names by IP addresses, and it can be used to point to services running on those devices. 
Split DNS:
Tailscale also supports split DNS, which allows you to configure different DNS settings for different types of traffic, for example, routing DNS lookups for internal services to an internal DNS server while still allowing public DNS lookups through the Tailscale-defined global nameservers. 
resolv.conf:
Tailscale can automatically configure the resolv.conf file on Linux devices, which defines the DNS servers a device uses for name resolution. 

In summary:
By enabling "Override DNS servers" in the Tailscale admin console, you can ensure that all name resolution for devices within your Tailscale network is handled by the Tailscale DNS service. This means that devices will use the Tailscale-defined global nameservers to resolve both public and private domain names, making it the primary source for DNS resolution within the Tailscale network. 

AI responses may include mistakes.