iiab/iiab #4012: Patch up Luanti role for (evolving!) Ubuntu 25.10 pre-releases

holta, 2025-05-10:
A quick job: I'll do it & test it coming days.

(And of course things will inevitably change in the coming year when Ubuntu 25.10 or 26.04 inevitably catch up to Debian 13 Trixie, repackaging / renaming so many different aspects of Minetest and its internals to Luanti.)

Related:

- PR #4011

fivemeyestore, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

myeyestore, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

fivemhostingserver, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyesgodcode, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

fivemeyestoreserver, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

godofcodeeyes, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

eyestorefivem, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

fivemserverhostings, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

eyestorefivemstore, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

ayestoro, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

eyesaider, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

fivemservereyestore, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyestoreoye, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

scriptforfivem, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

holta, 2025-05-11:
Mitigated by...

- PR #4013

closed 2025-05-11 by holta