iiab/iiab #4047: Logging across reboots needs to be easier on RasPiOS: a recent upstream change (in RasPiOS) is causing these logs to be lost

holta, 2025-07-18:
IIAB should very likely restore logging across boots, given this recent change in Raspberry Pi OS:

- https://forums.raspberrypi.com/viewtopic.php?p=2324941#p2324941
- https://github.com/raspberrypi/bookworm-feedback/issues/415

Though we (IIAB) might want to wait a few more weeks before finalizing this (maybe!) as Raspberry Pi OS is still a bit in flux here (-:

jvonau, 2025-07-18:
#3938

jvonau, 2025-10-05:
Think that point would be too late to capture and preserve the instillation routine. I would suggest in `iiab` prior to the reboot for an apt upgrade. A drop-in file with a higher number might be a better approach.

holta, 2025-10-05:
> Think that point would be too late to capture and preserve the instillation routine. I would suggest in `iiab` prior to the reboot for an apt upgrade. A drop-in file with a higher number might be a better approach.

Excellent questions.  Food for thought.

(Even if we should _try_ to avoid going down the road of overloading [sudo iiab](https://github.com/iiab/iiab-factory/blob/master/iiab), e.g. with RasPiOS-ish customizations.)

holta, 2025-10-06:
Old news from July, just FWIW:

- RasPiOS seems to ignore the `Storage=` setting (`Storage=auto` or whatever) setting in /etc/systemd/journald.conf

- Which is yet another reason that a drop-in within `/etc/systemd/journald.conf.d/` is advised, at least according to seemingly official/upstream comments in /etc/systemd/journald.conf

closed 2025-10-06 by holta