iiab/iiab #4052: Raspberry Pi 5, successful install then stops working after adding new content and reboot

technofox01, 2025-08-02:
Hi everyone,

I am getting pretty fed up with trying to get IIAB to run reliably. I have successfully installed IIAB on my Raspberry Pi 5 (4GB) model with a 1tb Samsung microSD card. I do a Raspberry Pi OS lite install, then update and upgrade all of the packages, reboot, then run the famous one line installer, get everything working and setup, download Kiwix content and Global maps, reboot, and then no more web interface for either users or admin console. I have been battling this issue for several different installs and am at wits end. I have generated the Pastebin URL using the iiab-diagnostics command:

https://paste.centos.org/view/3013b0d4

Can someone please help?

I will greatly appreciate it.

holta, 2025-08-03:
Thanks @technofox01 for having sent iiab-diagnostics output:

> and then no more web interface for either users or admin console.

- What exact URL(s) are you trying to browse to — ideally from several different WiFi client devices?

- What's the nature of this failure, as viewed from several different WiFi client devices?  (Ideally send a screenshot or photo or similar, revealing any error messages, if possible?)

jvonau, 2025-08-03:
>COMMAND: /usr/bin/df -ah    # Disk usage detail
 /dev/mmcblk0p2  940G  327G  594G  36% /var/hdd.log
log2ram         128M  5.6M  123M   5% /var/log`

>COMMAND: /usr/bin/lsblk    # Partition mount points
mmcblk0     179:0    0   955G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 954.5G  0 part /var/hdd.log
                                                                        /

Wondering if the double mount of  mmcblk0p2 at both "/ /var/hdd.log" is somehow blocking /library from being seen... #4047

jvonau, 2025-08-05:
Well looks like somebody found some old [log2ram install instructions](https://raspberrytips.com/install-log2ram-raspberry-pi/) as there is no apt packaging available without using a custom repo or tar file for [ log2ram](https://github.com/azlux/log2ram) so the issue appears to be end user generated.
> Assuming you can still SSH in, have you tried restarting nginx?

Looks to be running 
>  1710 ?        00:00:00   nginx
   1711 ?        00:00:00     nginx
   1712 ?        00:00:00     nginx
   1713 ?        00:00:00     nginx
   1714 ?        00:00:00     nginx

but could be throwing errors in to the journal (journalctl is your friend) while restarting nginx should puke if my theory about double mounting of mmcblk0p2 is somehow blocking /library from being accessible to nginx.

CrimsonPaladinActual, 2025-08-07:
You're right in that nginx looks to be running, but considering this issue happened after reboot I'm wonder if nginx came up before any of the service configs ever got generated or reloaded... so it could be running with an empty or stale config and simply never proxies /kiwix, /admin, /books, etc.

jvonau, 2025-08-07:
Might be related #3997. Ok @CrimsonPaladinActual, we are both kind of on the same page, but actual logging from nginx would be more useful. With the double mounting of the mmcblk0p2 partition I assume to be some how related to the installation of log2ram. There are some issues documented in which double mounting has messed up the output of some very basic commands like `df` resulting in only the last mount point being displayed, see above. This behavior is what might be causing
> so it [nginx] could be running with an empty or stale config and simply never proxies /kiwix, /admin, /books, etc.

that is being observed.
 
@technofox01 To test my theory I suggest trying:
`sudo df /dev/mmcblk0p2` note the mountpoint 
`sudo systemctl stop log2ram`
`sudo df /dev/mmcblk0p2` note the mountpoint 
`sudo systemctl restart nginx` 

Then retry accessing the web paths.

technofox01, 2025-08-08:
> Might be related [#3997](https://github.com/iiab/iiab/pull/3997). Ok [@CrimsonPaladinActual](https://github.com/CrimsonPaladinActual), we are both kind of on the same page, but actual logging from [...]

Sorry for the late reply. I figured it out. The IIAB software does not like to have an external USB drive after reboot for some reason. After removing the USB drive and just rebooting off of my microSD card, it works fine now. 

I appreciate your help.

jvonau, 2025-08-08:
> Sorry for the late reply. I figured it out. The IIAB software does not like to have an external USB drive after reboot for some reason. After removing the USB drive and just rebooting off of my microSD card, it works fine now.
> 
> I appreciate your help.

No problem, @holta not mounting the external usb drive on boot is a bug. Wondering if there is something is amiss with lite stage2 base installs and exfat support

>Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage2
/etc/debian-version: 12.11
display-manager? inactive   Arch1: arm64   Arch2: armhf

>/dev/sda1: LABEL="usb-ssd" UUID="688B-8A00" BLOCK_SIZE="512" TYPE="exfat" PARTLABEL="Basic data partition" PARTUUID="ce50a628-affa-4e0e-8571-fe8e973e43cf"

technofox01, 2025-08-08:
> No problem, [@holta](https://github.com/holta) not mounting the external usb drive on boot is a bug. Wondering if there is something is amiss with lite stage2 base installs and exfat support

Yeah, it was an exfat formatted USB drive. I also had issues with IAB directly installed on a USB flash drive. There is something messed up with IIAB's USB support/driver. At least from what I can see. I have reimaged and successfully got it work with a few exceptions (e.g. issues with Lokole and Internet Archive not working but those will be needed for another thread).

holta, 2025-08-08:
If anybody discovers hints as to why exFAT USB drive(s) can fail to mount on RasPiOS, please do share with others/all :pray:

(A big thank you to @technofox01 @jvonau and @CrimsonPaladinActual for having pushed things forward this far.)