iiab/iiab #4058: RasPiOS trixie testing

jvonau, 2025-08-22:
Test builds of RasPiOS trixie available to try 
https://downloads.raspberrypi.com/nightlies/

holta, 2025-08-22:
So far testing of on Debian 13 "Trixie" and pre-releases of RasPiOS 13 "Trixie" has gone better than expected: (FWIW!)

- #3736

PS I'd expect some rough spots with wheels, but (so far!) have not seen any, just yet anyway:

- https://github.com/piwheels/packages/issues/210#issuecomment-3036781727

cc: thanks you @EMG70 for all the testing so far, in recent months!  (@EMG70 if you can try to download and install the latest .xz from https://downloads.raspberrypi.com/nightlies/ onto your fastest Raspberry Pi, that'd be really great in coming days?)

EMG70, 2025-08-25:
IIAB Size 3 installed on Debian 13 on 8/23 nightly Trixie img successfully.

sudo iiab-diagnostics https://paste.centos.org/view/668c3243

tim-moody, 2025-08-26:
>Can Tailscale be installed in a more air-gapped way?

Don't understand the issue. Tailscale requires a central server to enforce network permissions. It can be theirs or someone else's, like yours. Am I missing something?

holta, 2025-08-26:
The goal of [roles/tailscale/tasks/install.yml](https://github.com/iiab/iiab/blob/master/roles/tailscale/tasks/install.yml) _(should be!)_ to install Tailscale's client software, not to use it:

- Actually using Tailscale's client software should happen later, with whatever coordination server's desired — e.g.:
  `tailscale login --login-server=https://my-custom-control-server.example.com`
  Whereas the default corresponds to: `tailscale login --login-server=https://controlplane.tailscale.com`
- So (in principle anyway!!) there really should be no need for roles/tailscale/tasks/install.yml to "call home" during Tailscale's software installation.
- But possibly Tailscale's client software attempts to transmit a bit of telemetry or similar, every time it's installed??  I'll investigate when I have a bit more time — until then here is (below) a paste from Google's AI/gossip engine (:

[image]

holta, 2025-08-26:
Ah, good catch: so these NGINX warnings occur on Linux Mint 22.2 and RasPiOS 13 Trixie — and likely other OS's too:

- https://github.com/iiab/iiab/issues/4061#issuecomment-3225235266

jvonau, 2025-08-26:
> > > channel 34 (5170 MHz), width: 20 MHz, center1: 5170 MHz
> > 
> > 
> > If the rfkill is in place and the country code was not set, how did the wifi device end up on channel 34?
> 
> Perhaps that's entirely normal behavior for RasPiOS. At least I thought so (or maybe I'm remembering wrong?)

Gathering some leads from https://github.com/raspberrypi/bookworm-feedback/issues/386 looks like pplug-netman is involved. Why, if rfkill has a soft block in place, would be the question.

holta, 2025-09-24:
> cool! so it has zram already

So promising!  Then oddkyymy install of IIAB (or the subsequent reboot, or something, unclear what yet!) wiped out the `zram0` line from the output of `lsblk` here:

- https://github.com/iiab/iiab/pull/4088#issuecomment-3325214565

> most likely this is from systemd-zram-generator
> 
> can you check these two commands?
> 
> ```
> sudo systemctl status systemd-zram-setup@zram0.service
> sudo zramctl
> ```

Apologies this particular OS in ROi 3  is no longer clean/vanilla — so this is all we get, for now:

```
root@box:~# systemctl status systemd-zram-setup@zram0.service
Unit systemd-zram-setup@zram0.service could not be found.
root@box:~# zramctl
root@box:~#
```

> can you also run
> 
> ```
> apt list raspberrypi\*

```
root@box:~# apt list raspberrypi\*
raspberrypi-archive-keyring/testing,testing,now 2025.1+rpt1 all [installed]
raspberrypi-net-mods/testing,testing,now 1.4.4 all [installed]
raspberrypi-sys-mods/testing,now 1:20250630 arm64 [installed]
raspberrypi-sys-mods/testing 1:20250630 armhf
raspberrypi-ui-mods/testing 1.20250411+b1 arm64
raspberrypi-ui-mods/testing 1.20250411+b1 armhf
```

> apt list linux-image\*

```
root@box:~# apt list linux-image\*
linux-image-6.12.25+rpt-rpi-2712-dbg/testing 1:6.12.25-1+rpt1+trixie arm64
linux-image-6.12.25+rpt-rpi-2712/testing 1:6.12.25-1+rpt1+trixie arm64
linux-image-6.12.25+rpt-rpi-v6-dbg/testing 1:6.12.25-1+rpt1+trixie armhf
linux-image-6.12.25+rpt-rpi-v6/testing 1:6.12.25-1+rpt1+trixie armhf
linux-image-6.12.25+rpt-rpi-v7-dbg/testing 1:6.12.25-1+rpt1+trixie armhf
linux-image-6.12.25+rpt-rpi-v7/testing 1:6.12.25-1+rpt1+trixie armhf
linux-image-6.12.25+rpt-rpi-v8-dbg/testing 1:6.12.25-1+rpt1+trixie arm64
linux-image-6.12.25+rpt-rpi-v8-rt-dbg/testing 1:6.12.25-1+rpt1+trixie arm64
linux-image-6.12.25+rpt-rpi-v8-rt/testing 1:6.12.25-1+rpt1+trixie arm64
linux-image-6.12.25+rpt-rpi-v8/testing 1:6.12.25-1+rpt1+trixie arm64
linux-image-6.12.31-arm64-16k-dbg/stable-security 6.12.31-1 arm64
linux-image-6.12.31-arm64-16k-unsigned/stable-security 6.12.31-1 arm64
linux-image-6.12.31-arm64-16k/stable-security 6.12.31-1 arm64
linux-image-6.12.31-arm64-dbg/stable-security 6.12.31-1 arm64
linux-image-6.12.31-arm64-unsigned/stable-security 6.12.31-1 arm64
linux-image-6.12.31-arm64/stable-security 6.12.31-1 arm64
linux-image-6.12.31-armmp-dbg/stable-security 6.12.31-1 armhf
linux-image-6.12.31-armmp-lpae-dbg/stable-security 6.12.31-1 armhf
linux-image-6.12.31-armmp-lpae/stable-security 6.12.31-1 armhf
linux-image-6.12.31-armmp/stable-security 6.12.31-1 armhf
linux-image-6.12.31-cloud-arm64-dbg/stable-security 6.12.31-1 arm64
linux-image-6.12.31-cloud-arm64-unsigned/stable-security 6.12.31-1 arm64
linux-image-6.12.31-cloud-arm64/stable-security 6.12.31-1 arm64
linux-image-6.12.31-rt-arm64-dbg/stable-security 6.12.31-1 arm64
linux-image-6.12.31-rt-arm64-unsigned/stable-security 6.12.31-1 arm64
linux-image-6.12.31-rt-arm64/stable-security 6.12.31-1 arm64
linux-image-6.12.31-rt-armmp-dbg/stable-security 6.12.31-1 armhf
[... 92 lines cut ...]
linux-image-arm64-16k/stable-security 6.12.48-1 arm64
linux-image-arm64-dbg/stable-security 6.12.48-1 arm64
linux-image-arm64-signed-template/stable-security 6.12.48-1 arm64
linux-image-arm64/stable-security 6.12.48-1 arm64
linux-image-armmp-dbg/stable-security 6.12.48-1 armhf
linux-image-armmp-lpae-dbg/stable-security 6.12.48-1 armhf
linux-image-armmp-lpae/stable-security 6.12.48-1 armhf
linux-image-armmp/stable-security 6.12.48-1 armhf
linux-image-cloud-arm64-dbg/stable-security 6.12.48-1 arm64
linux-image-cloud-arm64/stable-security 6.12.48-1 arm64
linux-image-rpi-2712-dbg/testing 1:6.12.47-1+rpt1 arm64
linux-image-rpi-2712/testing,now 1:6.12.47-1+rpt1 arm64 [installed]
linux-image-rpi-v6-dbg/testing 1:6.12.47-1+rpt1 armhf
linux-image-rpi-v6/testing 1:6.12.47-1+rpt1 armhf
linux-image-rpi-v7-dbg/testing 1:6.12.47-1+rpt1 armhf
linux-image-rpi-v7/testing 1:6.12.47-1+rpt1 armhf
linux-image-rpi-v8-dbg/testing 1:6.12.47-1+rpt1 arm64
linux-image-rpi-v8-rt-dbg/testing 1:6.12.47-1+rpt1 arm64
linux-image-rpi-v8-rt/testing 1:6.12.47-1+rpt1 arm64
linux-image-rpi-v8/testing,now 1:6.12.47-1+rpt1 arm64 [installed]
linux-image-rt-arm64-dbg/stable-security 6.12.48-1 arm64
linux-image-rt-arm64/stable-security 6.12.48-1 arm64
linux-image-rt-armmp-dbg/stable-security 6.12.48-1 armhf
linux-image-rt-armmp/stable-security 6.12.48-1 armhf
```

jvonau, 2025-09-29:
> > cool! so it has zram already
> 
> So promising! Then oddly my install of IIAB (or the subsequent reboot, or something, unclear what yet!) wiped out the `zram0` line from the output of `lsblk` here:
> 
>     * [WIP detect double mounting [for Upload2USB] #4088 (comment)](https://github.com/iiab/iiab/pull/4088#issuecomment-3325214565)
> 
> 
> > most likely this is from systemd-zram-generator
> > can you check these two commands?
> > ```
> > sudo systemctl status systemd-zram-setup@zram0.service
> > sudo zramctl
> > ```
> 
> Apologies this particular OS on RPi 3 is no longer clean/vanilla — so this is all we get, for now:
> 
> ```
> root@box:~# systemctl status systemd-zram-setup@zram0.service
> Unit systemd-zram-setup@zram0.service could not be found.
> root@box:~# zramctl
> root@box:~#
> ```

Did you run `apt autoremove`? and ran into https://github.com/iiab/iiab/pull/4087#issuecomment-3326921491

holta, 2025-10-02:
RasPiOS 13 Trixie is officially now released:

- https://raspberrypi.com/software/operating-systems/
- https://downloads.raspberrypi.com/raspios_arm64/release_notes.txt

Good News:

- They've now put 64-bit versions on top (on their above page), increasingly steering people away from the 32-bit version of the OS.
- [rpi-imager 2.0.0. RC1 ("Open Beta")](https://github.com/raspberrypi/rpi-imager/releases/tag/v2.0.0-rc1) was released yesterday, which should help new people getting involved with DIY, when that's released soon, hopefully in coming weeks.  Backstory: rpi-imager 1.8.x and 1.9.x while helpful, also had quite a large number of bugs ;)

tim-moody, 2025-10-24:
>I'm betting the newest raspi-imager is creating user-data and optionally network-config files in the same directory location as cmdline.txt

As an aside, I use dd instead of raspi-imager to write to an sd card, so should be possible to isolate any problems deriving from the latter.

jvonau, 2025-10-25:
https://github.com/raspberrypi/trixie-feedback/issues/3

jvonau, 2025-11-02:
Interesting reads
https://github.com/raspberrypi/bookworm-feedback/issues/220
https://github.com/raspberrypi/rpi-imager/issues/1252

jvonau, 2025-11-30:
Revised in 2025-11-24 https://www.raspberrypi.com/news/cloud-init-on-raspberry-pi-os/
https://github.com/raspberrypi/rpi-imager/issues/1320#issuecomment-3592682406