iiab/iiab #4060: U25.10 strangeness

jvonau, 2025-08-25:
sudo ip r
/etc/sudoers:58:14: unknown setting: 'logfile'
Defaults     logfile = /var/log/sudo.log
             ^~~~~~~~

jvonau, 2025-08-30:
https://github.com/iiab/iiab/pull/4074

holta, 2025-09-06:
Ubuntu Desktop 25.10's very latest Rust-based coreutils [1] [2] have also now broken a core aspect of our `sudo iiab` installer — as their new `date` command has been reimplemented to give bogus results here...

```
date +%s -r "/boot/$(ls -t /boot | head -1)"
```

https://github.com/iiab/iiab-factory/blob/01765621774c37fc73c5ef96c226a375cd078db7/iiab#L452

...in short the above command erroneously gives the actual time — instead of the timestamp of the most recently-modified file/folder in `/boot` — which in turn triggers a reboot every time (instead of proceeding to install IIAB!)

WORKAROUND, FOR NOW, if IIAB needs to be installed on (a similarly buggy) Ubuntu Desktop 25.10:

- Run `curl iiab.io/risky.txt | bash` (and subsequently `sudo iiab -r`) to proceed — which bypasses initial apt updates.

Related:

- PR #4074 

[1] "Ubuntu 25.10 Proceeding With Its Rust Coreutils Transition"
https://phoronix.com/news/Rust-Coreutils-Release-Pocket

[2] "Rust Coreutils 0.2 Released With "Massive" Performance Gains, Production-Ready Ubuntu Support"
https://phoronix.com/news/Rust-Coreutils-0.2

jvonau, 2025-09-20:
https://itsfoss.com/sudo-vs-sudo-rs/

jvonau, 2025-10-07:
> To get the old coreutils back run `apt install coreutils-from-gnu` within `iiab`

Since the hood is up on `iiab` might want to look at this also?

holta, 2025-10-23:
No Comment (-:

- "Ubuntu 25.10 Unattended Upgrades Broken Due To Rust Coreutils Bug"
   https://phoronix.com/news/Ubuntu-25.10-Broken-Upgrade

holta, 2025-10-25:
FWIW "Rust Coreutils 0.3 Released With Some Major Speed-Ups, Better GNU Compatibility"

https://phoronix.com/news/Rust-Coreutils-0.3-Released

jvonau, 2025-11-22:
```
jvonau@pi500:~$ curl iiab.io/risky.txt | bash -s 4093
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2707  100  2707    0     0  14785      0 --:--:-- --:--:-- --:--:-- 14873
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 34232  100 34232    0     0   257k      0 --:--:-- --:--:-- --:--:--  259k
Hit:1 http://ports.ubuntu.com/ubuntu-ports questing InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports questing-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports questing-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports questing-security InRelease
All packages are up to date.    

DANGER: User 'iiab-admin' has public password 'g0adm1n' per https://FAQ.IIAB.IO

This is for login to Internet-in-a-Box's Admin Console (http://box.lan/admin)
Technical Details: https://github.com/iiab/iiab/blob/master/roles/iiab-admin

Installing Internet-in-a-Box requires /etc/iiab/local_vars.yml

PLEASE PICK (1) 🚵 SMALL (2) 🚢🚣 MEDIUM or (3) 🚂🚃🚃 LARGE (NEEDS 2+ GB RAM).
Or try (m) for 🚑 MEDICAL.

See "What services (IIAB apps) are suggested during installation?" and
"What can I do with E-books and Internet-in-a-Box?" in https://FAQ.IIAB.IO

WARNING: This Can Take More Than An Hour, depending on Internet speed, CPU,
temperature, and the speed of your microSD card.

Please press a key — 1, 2, 3 or m: /usr/sbin/iiab: line 325: read: read error: 0: Input/output error
```

jvonau, 2025-11-22:
```
jvonau@pi500:~$ cd /etc/iiab
jvonau@pi500:~$ curl https://raw.githubusercontent.com/iiab/iiab/refs/heads/master/vars/local_vars_none.yml | sudo tee local_vars.yml
```
```
'apt update' is checking for OS updates, which may force a reboot...

All packages are up to date.

Hit any key to confirm you'll TRY TO RERUN 'sudo iiab' IF THERE IS A PROBLEM: 
REBOOTS
```

jvonau, 2025-11-23:
raspi-imager 1.9.6  with 'Ubuntu Server' the custom setting for wifi, user/pw, and ssh were offered to be applied to the image and prior to booting raspi-imager wrote on the first vfat partition:
- user-data for userid/pw and ssh
- network-config for the first-boot config (mine had wifi only)
- meta-data for the general setup

with the final result: https://github.com/iiab/iiab/pull/4093#issuecomment-3566889112

raspi-imager 1.9.6  with 'Ubuntu Desktop' didn't offer to apply any customization.