iiab/iiab #4061: Linux Mint 22.2 (Zara) Testing

avni, 2025-08-26:
Linux Mint 22.2 (Zara) - Beta was released on Aug 12th" https://blog.linuxmint.com/

Did a successful "LARGE" sized install on ThinkPad T14!
iiab-diagnostics: https://paste.centos.org/view/cc22e829

holta, 2025-08-26:
fyi @EMG70 it looks like we're all set, for Linux Mint 22.2's final release in coming weeks!

holta, 2025-08-26:
> Above page's recommended resolution can be considered later, if Linux Mint communities really want/need this (and upstream folk like Mint distro maintainers don't fix this themselves?)

CLARIF: These NGINX warnings occur not just on Linux Mint 22.2, but on RasPiOS 13 Trixie too (and likely other OS's as well...)

Thanks to @jvonau who confirmed this:

- https://github.com/iiab/iiab/issues/4058#issuecomment-3225211464