iiab/iiab #4066: [Upload2USB] Permission denied error when uploading file on Ubuntu Desktop 24.04, Ubuntu Desktop 25.10, and Mint 22.2

avni, 2025-08-28:
From #4063 
> Can you expand on the 'double mounting' aspect of the report please. Is the Gui mounting the usb drive also?

The GUI is mounting the USB drive, but I don't think it's 'double mounting' like I originally thought. When uploading a file via Upload2USB on either Ubuntu Desktop 24.04, Ubuntu Desktop 25.10, and Mint 22.2, a permission denied error is thrown. Of note is that, Raspberry Pi Trixie Desktop, and Ubuntu 24.10 Server do not have this error. 

**Error when folder for the day does not exist:**
```
Aug 27 22:27:30 box nginx[10442]: box nginx: 2025/08/27 22:27:30 [error] 10442#10442: *4 FastCGI sent in stderr: "PHP message: PHP Warning:  mkdir(): Permission denied in /library/www/html/upload2usb/upload2usb.php on line 66; PHP message: UPLOAD2USB ERROR: RuntimeException: {"usb_count":-1,"exception_msg":"Not able to create upload directory. <br\/>Make sure 'usb_lib_writable_sticks' is set to 'True'. <br\/><br\/>"} in /library/www/html/upload2usb/upload2usb.php:66
                                  Stack trace:
                                  #0 /library/www/html/upload2usb/upload-file.php(11): getTargetFolderPath()
                                  #1 {main}" while reading response header from upstream, client: 127.0.0.1, server: box, request: "POST /upload2usb/upload-file.php HTTP/1.1", upstream: "fastcgi://unix:/run/php/php8.3-fpm.sock:", host: "box", referrer: "http://box/usb/"
```

**Error when folder for the day already exists:** 
`box nginx: 2025/08/27 21:43:09 [error] 10441#10441: *1 FastCGI sent in stderr: "PHP message: PHP Warning:  move_uploaded_file(/media/usb0/UPLOADS.2025-08-27/SCaLE 22x Booth Assets.pdf): Failed to open stream: Permission denied in /library/www/html/upload2usb/upload-file.php on line 43; PHP message: PHP Warning:  move_uploaded_file(): Unable to move &quot;/tmp/phpPDHsiQ&quot; to &quot;/media/usb0/UPLOADS.2025-08-27/SCaLE 22x Booth Assets.pdf&quot; in /library/www/html/upload2usb/upload-file.php on line 43; PHP message: UPLOAD2USB ERROR: RuntimeException: {"usb_count":-1,"exception_msg":"There was an error uploading your file. <br\/><br\/>"} in /library/www/html/upload2usb/upload-file.php:53`

avni, 2025-08-28:
**Screencast**

https://github.com/user-attachments/assets/ec1fc9a8-bcc1-46bd-addb-12ecda88797e

holta, 2025-08-28:
@avni Would you have time to install Debian 13 "Trixie" onto x86_64 bare metal?  (This could potentially offer valuable insights!) 🤞

Ark74, 2025-08-28:
If I got some steps to reproduce, I'll be glad to test on Trisquel 12, 24.04 with some Celtic spices.

jvonau, 2025-08-28:
> Two possible approaches to resolve this issue:
> 
>     1. Consider adding a pop-up advisory that guides folks to disabling automount on their desktop OS's. This could show up on all desktop OS's. The downside of this approach is that it seems that the issue no longer surfaces on RaspiOS (Trixie) - disabling automount is not required and Upload2USB works just fine. So users on this newer OS (and perhaps others) would be seeing this popup directing them to take action unnecessarily.
> 
>     2. Auto-detect the double mounting and unmount the USB stick from the mount in the home directory.
> 
> 
> `#2` seems like a better user experience - would appreciate input from others on this!

Every desktop has a slightly different method of implementing this functionally depending on the backend for the desktop being used. Where/how do you plan on doing this?  Seems like something the upload function should be handling.

avni, 2025-08-29:
> @avni Would you have time to install Debian 13 "Trixie" (with graphical desktop) onto x86_64 bare metal? (This could potentially offer valuable insights!) 🤞

Debian 13 (w/ graphical desktop) also double mounts the USB stick. Akin to Ubuntu 24.04 LTS (w/ graphical desktop), using dconf Editor to disable automount allows for Upload2USB to work as expected. 

> Every desktop has a slightly different method of implementing this functionally depending on the backend for the desktop being used. Where/how do you plan on doing this? Seems like something the upload function should be handling.

I'm not sure where/how yet - I was thinking of starting with usbmount. What do you mean by upload function? Do you mean Upload2USB file upload should handle this? How if so?

avni, 2025-08-30:
Interestingly enough, Linux Mint 22.2 (Zara) using the Cinnamon desktop, does not behave the same as Ubuntu 24.04 and Debian 13 (both w/ graphical desktops). Disabling automount via dconf Editor does not resolve the Upload2USB permission denied error. 

Running: `sudo systemctl stop udisks2.service` and stopping the udisks2 service does prevent double mounting and get Upload2USB working.  

I'll update the README with the current findings.

holta, 2025-08-30:
> I'll update the README with the current findings.

Thank you!

➡️ Please consider linking to the most insightful tickets, when that's a much more efficient way of documenting structural architecture (or fringe issues, if they deserve any mention at all??) within [usb_lib's README](https://github.com/iiab/iiab/blob/master/roles/usb_lib/README.rst) ⬅️

avni, 2025-08-30:
Well. It seems that changing the settings in dconf Editor when called from the command line (`dconf-editor`) does not save them in the same location as opening dconf Editor from the Applications menu! Perhaps they are not synced?! See screenshot. Should this be reported as a bug to Linux Mint Cinnamon? 

Disabling automount via the dconf Editor opened via command line does not resolve the Upload2USB issue but disabling automount via the instanced opened from the Applications menu does! 

 
[image]

holta, 2025-08-30:
> changing the settings in dconf Editor when called from the command line (`dconf-editor`) does not save them in the same location as changing the settings from dconf Editor opened from the Applications menu!

Any chance that's entirely normal??

Certainly dconf-editor edits dconf database settings for the "currently logged-in user" — but that can mean different things from CLI (e.g. root?) OR the user logged in and running the graphical/desktop session, etc.

avni, 2025-09-13:
> Certainly dconf-editor edits dconf database settings for the "currently logged-in user" — but that can mean different things from CLI (e.g. root?) OR the user logged into & running the graphical/desktop session, etc.

Good point, just not what I initially expected to happen.