iiab/iiab #4077: Unable to Host Hotspot on IIAB

GarlicToast1, 2025-08-31:
Device is a Raspberry Pi 5 (8GB RAM)
Running on Raspberry Pi OS (64-bit)
Running latest IIAB version (8.3)

All went smoothly via the one line installer, the Pi rebooted properly, but no hotspot was created. The entire setup process was done through ethernet. I ran iiab-diagnostics, and the result is attached below. 

[IIAB-diagnostics.txt](https://github.com/user-attachments/files/22062144/IIAB-diagnostics.txt)

holta, 2025-08-31:
Hi @GarlicToast1 

Can you reply with the URL generated by iiab-diagnostics ?

Or if you prefer, please attach the **complete file** generated by iiab-diagnostics...

```
/etc/iiab/diag/8.3_raspbian-12_2025-08-30_19:57:25_MDT_toasted
```

Thank you!

GarlicToast1, 2025-08-31:
> Hi [@GarlicToast1](https://github.com/GarlicToast1)
> 
> Can you reply with the URL generated by iiab-diagnostics ?
> 
> Or if you prefer, please attach the **_complete file_** generated by iiab-diagnostics...
> 
> ```
> /etc/iiab/diag/8.3_raspbian-12_2025-08-30_19:57:25_MDT_toasted
> ```
> 
> Thank you!

Thank you! the link is:
https://paste.centos.org/view/d4d32bc30

holta, 2025-08-31:
> https://paste.centos.org/view/d4d32bc3

Thanks much @GarlicToast1

The issue is:

> "Authorization required, but no authorization protocol specified"
>
> "Cant connect to display: :0"

As detailed here:

> 2025-08-30 18:22:52,982 p=48146 u=root n=ansible INFO| TASK [network : Run "sudo -u '#1000' sudo raspi-config nonint do_wifi_country US" (using var host_country_code) to unblock WiFi, if RasPiOS -- this sets SUDO_UID=1000 (otherwise raspi-config often fails on RasPiOS with desktop)] ***
2025-08-30 18:22:53,434 p=48146 u=root n=ansible INFO| fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["sudo", "-u", "#1000", "sudo", "raspi-config", "nonint", "do_wifi_country", "US"], "delta": "0:00:00.192334", "end": "2025-08-30 18:22:53.401908", "msg": "non-zero return code", "rc": 1, "start": "2025-08-30 18:22:53.209574", "stderr": "Authorization required, but no authorization protocol specified", "stderr_lines": ["Authorization required, but no authorization protocol specified"], "stdout": "Cant connect to display: :0", "stdout_lines": ["Cant connect to display: :0"]}

Can you try running the following 3 commands?

```
sudo -i
iiab-network
iiab
```

(Let us know if that helps?
Or if it fails, please reply with a photo or screenshot, Thanks!)

jvonau, 2025-08-31:
FWIW: running `sudo -u '#1000' sudo raspi-config nonint do_wifi_country US` in an ssh session worked on trixie:
>jvonau@pi500:~ $ env
SHELL=/bin/bash
NO_AT_BRIDGE=1
PWD=/home/jvonau
LOGNAME=jvonau
XDG_SESSION_TYPE=tty
MOTD_SHOWN=pam
HOME=/home/jvonau
LANG=en_GB.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.7z=01;31:*.ace=01;31:*.alz=01;31:*.apk=01;31:*.arc=01;31:*.arj=01;31:*.bz=01;31:*.bz2=01;31:*.cab=01;31:*.cpio=01;31:*.crate=01;31:*.deb=01;31:*.drpm=01;31:*.dwm=01;31:*.dz=01;31:*.ear=01;31:*.egg=01;31:*.esd=01;31:*.gz=01;31:*.jar=01;31:*.lha=01;31:*.lrz=01;31:*.lz=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.lzo=01;31:*.pyz=01;31:*.rar=01;31:*.rpm=01;31:*.rz=01;31:*.sar=01;31:*.swm=01;31:*.t7z=01;31:*.tar=01;31:*.taz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tgz=01;31:*.tlz=01;31:*.txz=01;31:*.tz=01;31:*.tzo=01;31:*.tzst=01;31:*.udeb=01;31:*.war=01;31:*.whl=01;31:*.wim=01;31:*.xz=01;31:*.z=01;31:*.zip=01;31:*.zoo=01;31:*.zst=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.jxl=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.crdownload=00;90:*.dpkg-dist=00;90:*.dpkg-new=00;90:*.dpkg-old=00;90:*.dpkg-tmp=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:*.swp=00;90:*.tmp=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:
SSH_CONNECTION=10.0.0.242 47898 10.0.0.34 22
XDG_SESSION_CLASS=user
TERM=xterm-256color
USER=jvonau
SHLVL=1
XDG_SESSION_ID=4
XDG_RUNTIME_DIR=/run/user/1000
SSH_CLIENT=10.0.0.242 47898 22
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
SSH_TTY=/dev/pts/0
TEXTDOMAIN=pam_chksshpwd
_=/usr/bin/env

From d4d32bc3
>=IIAB==========================================================================
COMMAND: /usr/bin/env    # Environment variables
 SHELL=/bin/bash
COLORTERM=truecolor
SUDO_GID=1000
SUDO_COMMAND=/usr/bin/iiab-diagnostics
SUDO_USER=tester
NO_AT_BRIDGE=1
PWD=/home/tester
LOGNAME=root
_=/usr/bin/env
XAUTHORITY=/home/tester/.Xauthority
HOME=/root
LANG=en_GB.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.swp=00;90:*.tmp=00;90:*.dpkg-dist=00;90:*.dpkg-old=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:
XDG_CURRENT_DESKTOP=LXDE
TERM=xterm-256color
USER=root
DISPLAY=:0
SHLVL=1
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
SUDO_UID=1000
MAIL=/var/mail/root

Note the differences between between the two, think there is a subtle difference in using `sudo iiab` and becoming root (`sudo su?`) before running `iiab`.

jvonau, 2025-08-31:
host_ssid: unittest
wifi_up_down: True

`cat /etc/hostapd/hostapd.conf` fixed up to remove #
> Basic configuration
interface=ap0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
ssid=unittest
channel=6
bridge=br0

> Hardware configuration
driver=nl80211
country_code=CA

 > limit emissions to what is legal in country
ieee80211d=1
hw_mode=g`

holta, 2025-08-31:
> Perhaps just running `nmcli radio wifi on` would be enough to overcome unlocking the WiFi, the detection of what country_code to be used for hostapd could be altered a bit to account for what behavior that has been observed.

Might be a very promising approach:

- The current workaround is tested to work well with a standard/vanilla Raspberry Pi OS install.

- But possibly we can make this all more resilient for users who modify the desktop environment, e.g. to LXDE.

holta, 2025-09-25:
@GarlicToast1 can this be closed now?

Thanks to @jvonau's recently merged work below:

- PR #4078 
- PR #4091