iiab/iiab #4099: "Unable to restart service dnsmasq" during "20 min" IIAB "MEDIUM" on Ubuntu 24.04 on x86-64 / test-install

holta, 2025-09-29:
Possible issue to keep an eye on, if it recurs too often?  (Or if this is a very rare timing issue, we can presumably ignore it.)

```
TASK [network : Restarting dnsmasq] ********************************************
Error: : Task failed: Module failed: Unable to restart service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.
See "systemctl status dnsmasq.service" and "journalctl -xeu dnsmasq.service" for details.

Origin: /opt/iiab/iiab/roles/network/tasks/restart.yml:119:3

117 #keep an eye on legacy wifi installs where br0 is present but not 'online' with an ip address
118 #due to hostapd didn't go to a carrier state. All others should get dnsmasq restarted
119 - name: Restarting dnsmasq
      ^ column 3

fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xeu dnsmasq.service\" for details.\n"}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=1000 changed=399  unreachable=0    failed=1    skipped=193  rescued=0    ignored=3
```

Arising from PR #4098 CI/QA workflow (GitHub Actions) here:

https://github.com/iiab/iiab/actions/runs/18108296803/job/51528214042#step:6:4278

holta, 2025-11-07:
> perhaps as much as 33% or 50% of the time??

Or maybe the number is much lower?  (Number might not be representative, but I only see 1 out of 18 commits failing with the dnsmasq restart failure to restart here...)

- https://github.com/iiab/iiab/actions/runs/19173813566/job/54823809257
- https://github.com/iiab/iiab/pull/4120/commits

jvonau, 2025-11-08:
Within #4078 where IIAB-Bridge.netdev was removed the line below is no longer accurate, didn't change the user facing message
>TASK [network : Copy the bridge script - Creates br0]

This VM is using systemd-networkd 
>changed: [127.0.0.1] => (item={'option': 'dhcpcd_result', 'value': 'not-found'})
changed: [127.0.0.1] => (item={'option': 'network_manager_active', 'value': False})
changed: [127.0.0.1] => (item={'option': 'systemd_networkd_active', 'value': True})

Given that layout I would of expected this VM to always fail at that point as br0 was not created but other runs success, perplexing to say the least. 

Side note within 4078 is testing of U25.10 where NetworkManager looks to be using netplan as the single source of truth for the configuration files also noted in https://github.com/iiab/iiab/issues/4058#issuecomment-3446306935. The network code for the most part sidesteps messing with netplan as when first introduced didn't work as expected but now might have to embrace messing with netplan's yaml files.

jvonau, 2025-11-08:
Moving to installing netplan related files for network configuration might be a way forward as NetworkManager, netplan and cloud-init are very closely coupled in the newest OS's that are out and coming out soon. Some of the staging of boot sequencing might need to be tinkered with given the revised layout. The OSs that don't ship cloud-init/netplan are supported with the current layout but I really don't want to support somebodies' one off install of their favorite OS when things are being standardized around netplan given the coupled integration with cloud-init, systemd-networkd and NetworkManager. The admin-console might need some changes given the new reality in the networking layout of the new OS trends. I've already been down the netplan rabbit hole in the [past](https://github.com/iiab/iiab/commits/d44f814e0e7e31e62071da2d114b186214bf56aa/roles/network/templates/network/netplan.j2?browsing_rename_history=true&new_path=roles/network/templates/network/netplan.j2.unused&original_branch=master) 

Task: Now lets start with what files do reside in /etc/netplan in each of the supported OS variants that should be supported.

holta, 2025-11-11:
_Interesting that MEDIUM-sized IIAB installs somehow help create the conditions for dnsmasq to fail[*] to restart... whereas UNITTEST-sized and NONE-sized IIAB installs don't seem to have this problem!_

Here's another example of dnsmasq failing to restart: https://github.com/iiab/iiab/actions/runs/19252256180/job/55039447879?pr=4142 (this just arose during GitHub Actions testing of PR #4142)

```
TASK [network : Restart hostapd if WiFi slave is inactive] *********************
skipping: [127.0.0.1]

TASK [network : Restarting dnsmasq] ********************************************
Error: : Task failed: Module failed: Unable to restart service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.
See "systemctl status dnsmasq.service" and "journalctl -xeu dnsmasq.service" for details.

Origin: /opt/iiab/iiab/roles/network/tasks/restart.yml:119:3

117 #keep an eye on legacy wifi installs where br0 is present but not 'online' with an ip address
118 #due to hostapd didn't go to a carrier state. All others should get dnsmasq restarted
119 - name: Restarting dnsmasq
      ^ column 3

fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xeu dnsmasq.service\" for details.\n"}
```

[*] Intermittently as usual.  Re-running the GitHub Action workflow worked, as can be seen here:

- PR #4142

jvonau, 2025-11-11:
> _Interesting that MEDIUM-sized IIAB installs somehow help create the conditions for dnsmasq to fail[*] to restart... whereas UNITTEST-sized and NONE-sized IIAB installs don't seem to have this problem!_
> 

Unittest has the networking [disabled](https://github.com/iiab/iiab/blob/master/vars/local_vars_unittest.yml#L15-L18) and 
tailscale_install: [True](https://github.com/iiab/iiab/blob/master/vars/local_vars_unittest.yml#L152-L155)

None has the default of networking enabled and tailscale_install: [False](https://github.com/iiab/iiab/blob/master/vars/local_vars_none.yml)

> Here's another example of dnsmasq failing to restart: https://github.com/iiab/iiab/actions/runs/19252256180/job/55039447879?pr=4142 (this just arose during GitHub Actions testing of PR [#4142](https://github.com/iiab/iiab/pull/4142))
> 
> ```
> TASK [network : Restart hostapd if WiFi slave is inactive] *********************
> skipping: [127.0.0.1]
> 
> TASK [network : Restarting dnsmasq] ********************************************
> Error: : Task failed: Module failed: Unable to restart service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.
> See "systemctl status dnsmasq.service" and "journalctl -xeu dnsmasq.service" for details.
> 
> Origin: /opt/iiab/iiab/roles/network/tasks/restart.yml:119:3
> 
> 117 #keep an eye on legacy wifi installs where br0 is present but not 'online' with an ip address
> 118 #due to hostapd didn't go to a carrier state. All others should get dnsmasq restarted
> 119 - name: Restarting dnsmasq
>       ^ column 3
> 
> fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xeu dnsmasq.service\" for details.\n"}
> ```
> 
> [*] Intermittently as usual. Re-running the GitHub Action workflow worked, as can be seen here:
> 
>     * PR [Fix DEPRECATION WARNINGs introduced in ansible-core 2.20 [in roles/kiwix only, for now!] #4142](https://github.com/iiab/iiab/pull/4142)

At the end of the day #4093 will remove this stanza going forward.

jvonau, 2025-11-13:
Don't care

jvonau, 2025-11-26:
Test what you want to grow your understanding, 4093 as the same mods.

jvonau, 2025-11-26:
Simple [addition](https://github.com/iiab/iiab/compare/master...jvonau:iiab:test_cli?expand=1)

https://github.com/jvonau/iiab/actions/runs/19711367991/job/56472611490
```
Run ip a
  ip a
  shell: /usr/bin/bash -e {0}
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:0d:3a:35:a1:40 brd ff:ff:ff:ff:ff:ff
    inet 10.1.0.164/20 metric 100 brd 10.1.15.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::20d:3aff:fe35:a140/64 scope link 
       valid_lft forever preferred_lft forever
3: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 66:16:2c:68:12:40 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
```
```
Run ip r
  ip r
  shell: /usr/bin/bash -e {0}
default via 10.1.0.1 dev eth0 proto dhcp src 10.1.0.164 metric 100 
10.1.0.0/20 dev eth0 proto kernel scope link src 10.1.0.164 metric 100 
10.1.0.1 dev eth0 proto dhcp scope link src 10.1.0.164 metric 100 
168.63.129.16 via 10.1.0.1 dev eth0 proto dhcp src 10.1.0.164 metric 100 
169.254.169.254 via 10.1.0.1 dev eth0 proto dhcp src 10.1.0.164 metric 100 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 
```
```
TASK [network : Add 'detected_network' variable values to /etc/iiab/iiab.ini] ***
changed: [127.0.0.1] => (item={'option': 'has_ifcfg_gw', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'prior_gateway_device', 'value': 'unset'})
changed: [127.0.0.1] => (item={'option': 'dhcpcd_result', 'value': 'not-found'})
changed: [127.0.0.1] => (item={'option': 'network_manager_active', 'value': False})
changed: [127.0.0.1] => (item={'option': 'systemd_networkd_active', 'value': True})
changed: [127.0.0.1] => (item={'option': 'wan_in_interfaces', 'value': False})
changed: [127.0.0.1] => (item={'option': 'wireless_list_1(wifi1)', 'value': 'not found-1'})
changed: [127.0.0.1] => (item={'option': 'wireless_list_2(wifi2)', 'value': 'not found-2'})
changed: [127.0.0.1] => (item={'option': 'num_wifi_interfaces', 'value': 0})
changed: [127.0.0.1] => (item={'option': 'discovered_wireless_iface', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'discovered_wired_iface', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'exclude_devices', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'num_lan_interfaces', 'value': 0})
changed: [127.0.0.1] => (item={'option': 'gui_static_wan', 'value': False})
changed: [127.0.0.1] => (item={'option': 'iiab_lan_iface', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'iiab_wan_iface', 'value': 'eth0'})
changed: [127.0.0.1] => (item={'option': 'can_be_ap', 'value': False})
changed: [127.0.0.1] => (item={'option': 'host_country_code_found', 'value': 'UNSET'})
changed: [127.0.0.1] => (item={'option': 'wifi_firmware_43430', 'value': 'os'})
changed: [127.0.0.1] => (item={'option': 'wifi_firmware_43455', 'value': 19})
```

Well this is different, there is no 'discovered_wired_iface', think the upstream runner image changed for some reason unknown to me

jvonau, 2025-11-28:
The current code doesn't fail when 'appliance' is detected, the restart of dnsmasq is suppressed as there would be no br0 to worry about as br0 is not created at all so no need to provide dhcp server services. 

https://github.com/muthuri-dev/iiab/actions/runs/19764774732/job/56634812499?pr=1#step:6:4575
>TASK [network : Restarting dnsmasq] ********************************************
skipping: [127.0.0.1]

The VM failure occurs when the test VM has a second network interface detected and not having NetworkManager running, the code path would enslave the second interface to br0 when the second device is detected and thus becomes 'gateway' mode. Why a **second interface appears out of the blue randomly** is the primary cause of the failure as I have shown above. I'm at a loss to explain why some runs are not creating the second interface and some are. 

>

muthuri-dev, 2025-11-29:
> The current code doesn't fail when 'appliance' is detected, the restart of dnsmasq is suppressed as there would be no br0 to worry about as br0 is not created at all so no need to provide dhcp [...]

@jvonau - Thanks for the detailed breakdown!

**Appliance mode** ✅
- No br0 created
- dnsmasq restart skipped (correct behavior)
- Works consistently

**Gateway mode with random 2nd interface** ⚡
- When 2nd interface is detected, br0 is created and enslaved
- dnsmasq restart is triggered during br0 initialization
- This is where the race condition occurs

`bind-dynamic` fix makes dnsmasq resilient:
- dnsmasq can bind to br0 even while it's still configuring
- No longer fails if br0 isn't fully "up" yet

AND PR #4093's DNS configuration changes also resolve the race condition:
- Setting `no-resolv` unconditionally
- Adding fallback `server=127.0.0.53` (systemd-resolved)
- Proper DNS sequencing with netplan bridge initialization

which happens I missed this file: https://github.com/iiab/iiab/pull/4093/files#diff-dc687d104cd2fc3518dd4772767dd57c3342b35a8bc9e75d330f6766cfd30654

jvonau, 2025-11-29:
>Gateway mode with random 2nd interface ⚡
    When 2nd interface is detected, br0 is created and enslaved
    dnsmasq restart is triggered during br0 initialization
    This is where the race condition occurs
bind-dynamic fix makes dnsmasq resilient:
    dnsmasq can bind to br0 even while it's still configuring
    No longer fails if br0 isn't fully "up" yet

What is missing from the testing is bluetooth networking that relies on br0 also where br0 is not auto created unlike hostapd, br0 must be preexisting and configured. This functionality was broken by removing 'netdev' in 4078

> @holta - Yes, roles/network/templates/network/dnsmasq-iiab 

This unit file worked more as a traffic cop during boot.